evcc-io/evcc #9672: BMW/MINI Not Authorizing

othellocat, 2023-09-02:
### Describe the bug

While configuring evcc, BMW/MINI give the following error when attempting to add to setup:

Testing the configuration of Mini ...
   Error: cannot create vehicle 'mini': unexpected status: 401 (Unauthorized)

### Steps to reproduce

1.Run _evcc configure_
2.Add MINI or BMW under the vehicle configuration
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 951dbc349e693b35a9bd7c4a42948708e8413918a4bcffba4cbcb86cc617f0c9

interval: 10s # control cycle interval

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 172.x.x.x 
  password: xxxxx 
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 172.x.x.x  
  password: xxxxx 
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 172.x.x.x  
  password: xxxxx 
  name: battery3

site:
  title: Woodrose
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
No log for configurator
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.3

andig, 2023-09-02:
Vmtl falscher user/passwort

closed 2023-09-02 by andig

othellocat, 2023-09-10:
Using bimmerconnected, I've narrowed the login issue to being in North America which uses a different server. Would it be possible to add an optional region setting (north_america) that uses cocoapi.bmwgroup.us for authorizing?

DEBUG:httpcore.connection:connect_tcp.started host='cocoapi.bmwgroup.us' port=443 local_address=None timeout=30.0 socket_options=None

INFO:httpx:HTTP Request: POST https://login.bmwusa.com/gcdm/oauth/authenticate "HTTP/1.1 200 OK"

andig, 2023-09-10:
I see. Could you provide credentials for testing to info@evcc.io or would you want to open a pr?

DanielGlaas, 2023-09-27:
Ich habe den selben Fehler:

`evcc vehicle`:
> .Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': unexpected status: 401 (Unauthorized)
Capacity: 7.0kWh
Features: [Offline]

Mit dem Parameter `-l trace` kommt dann noch diese Ausgabe, die vielleicht helfen könnte:
> {"error": "invalid_client", "error_description": "Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)"}

Die Logindaten haben über ein Jahr lang funktioniert und sind unverändert, ich sehe also keinen Fehler in meiner Konfiguration.

andig, 2023-09-27:
@DanielGlaas jetzt ist die Frage was sich geändert hat. Geht eine ältere Version noch? Falls ja bitte Zugangsdaten für den Test an info@evcc.io.