evcc-io/evcc #9673: EVCC läuft nicht mehr nach Update auf 0.119.3

chadibe0, 2023-09-02:
### Describe the bug

Habe heute auf 0.119.3 upgedatet. Darauf hin lief EVCC nicht mehr. Zumindest war der Server über das GUI nicht erreichbar. Restart des Rechners hat auch nichts gebracht. Daraufhin wieder erfolgreich zurück auf 0.119.2. Alles geht. Und erneut auf 0.119.3. Leider ohne Erfolg. In den Logs sehe ich nicht sehr viel. Vielleicht muss man da noch mehr einschalten. Ich weiss aber nicht wie.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ###

interval: 10s # control cycle interval

sponsortoken: ###

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: fronius.diethelms.home  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: fronius.diethelms.home  
  name: pv2

chargers:
- type: template
  template: easee 
  user: ###  
  password: ###  
  charger: EHZZ6TB3  
  name: wallbox1

vehicles:
- type: template
  template: id 
  title: ID.3  
  user: ###  
  password: ###  
  vin: WVWZZZE1ZMP052579  
  capacity: 55  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  minSoC: 20  
  targetSoC: 80  
  name: ID3

loadpoints:
- title: Garage Rechts
  charger: wallbox1
  vehicle: ID3
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: Diethelms
  meters:
    grid: grid1
    pvs:
    - pv2

mqtt:
  broker: 192.168.178.30:1883
  topic: evcc

tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.3233
    zones:
      - days: Mo-Fr
        hours: 0-7,20-0
        price: 0.2252
      - days: Sa
        hours: 0-7
        price: 0.2252
      - days: Sa
        hours: 13-0
        price: 0.2252
      - days: So
        price: 0.2252
  feedin:
    type: fixed
    price: 0.1716
    zones:
      - days: Mo-Fr
        hours: 0-7,20-0
        price: 0.1315
      - days: Sa
        hours: 0-7
        price: 0.1315
      - days: Sa
        hours: 13-0
        price: 0.1315
      - days: So
        price: 0.1315
  planner:
    type: electricitymaps
    uri: https://api.co2signal.com/
    token: axl7RtcXMy2bvDQ0PB59oPfOFBRz9rVp
    zone: CH
```

### Log details

```text
bernhard@evcc:/etc$ evcc --log debug          
[main  ] INFO 2023/09/02 08:59:18 evcc 0.119.3
[main  ] INFO 2023/09/02 08:59:18 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/02 08:59:18 starting ui and api at :7070
[db    ] INFO 2023/09/02 08:59:28 using sqlite database: /home/bernhard/.evcc/evcc.db
[mqtt  ] INFO 2023/09/02 08:59:28 connecting evcc-171760204 at tcp://192.168.178.30:1883
[mqtt  ] DEBUG 2023/09/02 08:59:28 tcp://192.168.178.30:1883 connected
[lp-1  ] WARN 2023/09/02 08:59:30 locking phase config to 3p for switchable charger
```

### What type of operating system are you running?

Linux

### Version

0.119.3

tzatti1, 2023-09-02:
Selbes Problem hier:
[image]

`# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
  
# sponsors can set telemetry: true to enable anonymous data aggregation  
# see https://github.com/evcc-io/evcc/discussions/4554  
telemetry: false  
  
meters:  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.178.131    
  port: 502    
  usage: grid    
  modbus: tcpip    
  name: grid1  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.178.131    
  port: 502    
  usage: pv    
  modbus: tcpip    
  name: pv2  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.178.131    
  port: 502    
  usage: battery    
  modbus: tcpip    
  name: battery3  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.178.131    
  port: 502    
  usage: battery    
  modbus: tcpip    
  name: battery4  
  
chargers:  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EHZJCFLF    
  name: wallbox6  
  
vehicles:  
- type: template  
  template: id   
  title: VW ID4    
  user: *****  
  password: *****  
  vin: *****  
  capacity: 77    
  name: ev5  
  
loadpoints:  
- title: Garage  
  charger: wallbox6  
  vehicle: ev5  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: false  
  
site:  
  title: My home  
  meters:  
    grid: grid1  
    pv:  
    - pv2  
    battery:  
    - battery3  
    - battery4  
  
tariffs:  
  currency: EUR  
  grid:  
    type: fixed  
    price: 0.3256  
  
  feedin:  
    type: fixed  
    price: 0.082  `

tzatti1, 2023-09-02:
Update: Hat sich bei mir wohl von selbst wieder gelöst nach ca. 30 Minuten

closed 2023-09-02 by andig

chadibe0, 2023-09-03:
Einfach zur Info: Bei mir löst sich das Problem auch nach 12 Stunden und mehr nicht. Gehe jetzt wieder auf die 0.119.2

andig, 2023-09-03:
Es liegt an Easee, nicht an der Version- wird also nix ändern.

chadibe0, 2023-09-03:
Das verstehe ich jetzt leider nicht. Wenn ich zurück auf die evcc 0.119.2 gehe, funktioniert alles und mit der 0.119.3 funktioniert es nicht mehr. D.h. alle, die eine easee haben, können die 0.119.3 nicht (mehr) nutzen?

andig, 2023-09-03:
Dein Issue enthält kein verwertbares Log das den Fehler zeigt. Zeig uns doch bitte mal Logfile für beide Versionen. Ich halte das für ausgeschlossen.

chadibe0, 2023-09-03:
Mache ich gerne. Ich war ja selber auch etwas überrascht, dass man in den Logs nicht mehr sieht. Kann ich da noch mehr einschalten. Aktuell läuft die 119.2. Ich würde dann auf 119.3 updaten die Logs runter ziehen und dann wieder zurück auf 119.2 und noch einmal Logs, ok?

chadibe0, 2023-09-03:
Habe das nun so gemacht. Mehr bekomme ich aber leider auch nicht heraus:

```
Sep 03 11:21:15 evcc systemd[1]: Stopped evcc.service - evcc.
Sep 03 11:22:44 evcc systemd[1]: Started evcc.service - evcc.
Sep 03 11:22:44 evcc evcc[380731]: [main  ] INFO 2023/09/03 11:22:44 evcc 0.119.2
Sep 03 11:22:45 evcc evcc[380731]: [main  ] INFO 2023/09/03 11:22:45 using config file: /etc/evcc.yaml
Sep 03 11:22:45 evcc evcc[380731]: [main  ] INFO 2023/09/03 11:22:45 starting ui and api at :7070
Sep 03 11:22:45 evcc evcc[380731]: [db    ] INFO 2023/09/03 11:22:45 using sqlite database: /var/lib/evcc/evcc.db
Sep 03 11:22:45 evcc evcc[380731]: [mqtt  ] INFO 2023/09/03 11:22:45 connecting evcc-361738409 at tcp://192.168.178.30:1883
Sep 03 11:22:47 evcc evcc[380731]: [lp-1  ] WARN 2023/09/03 11:22:47 locking phase config to 3p for switchable charger
Sep 03 11:23:28 evcc evcc[380731]: [em    ] ERROR 2023/09/03 11:23:28 unexpected status: 404 (Not Found)
Sep 03 11:23:28 evcc evcc[380731]: [main  ] ERROR 2023/09/03 11:23:28 failed configuring planner tariff: cannot create tariff 'electricitymaps': unexpected status: 404 (Not Found)
Sep 03 11:23:28 evcc evcc[380731]: [site  ] WARN 2023/09/03 11:23:28 deprecated: use 'pv' instead of 'pvs'
Sep 03 11:23:28 evcc evcc[380731]: [site  ] INFO 2023/09/03 11:23:28 site config:
Sep 03 11:23:28 evcc evcc[380731]: [site  ] INFO 2023/09/03 11:23:28   meters:      grid ✓ pv ✓ battery ✗
Sep 03 11:23:28 evcc evcc[380731]: [site  ] INFO 2023/09/03 11:23:28     grid:      power ✓ energy ✗ currents ✗
Sep 03 11:23:28 evcc evcc[380731]: [site  ] INFO 2023/09/03 11:23:28     pv 1:      power ✓ energy ✗ currents ✗
Sep 03 11:23:28 evcc evcc[380731]: [site  ] INFO 2023/09/03 11:23:28   vehicles:
Sep 03 11:23:28 evcc evcc[380731]: [site  ] INFO 2023/09/03 11:23:28     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Sep 03 11:23:28 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:28 loadpoint 1:
Sep 03 11:23:28 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:28   mode:        pv
Sep 03 11:23:28 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:28   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Sep 03 11:23:28 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:28   meters:      charge ✓
Sep 03 11:23:28 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:28     charge:    power ✓ energy ✓ currents ✓
Sep 03 11:23:28 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:28 vehicle updated: unknown -> ID.3
Sep 03 11:23:29 evcc evcc[380731]: [site  ] WARN 2023/09/03 11:23:29 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Sep 03 11:23:29 evcc evcc[380731]: [main  ] INFO 2023/09/03 11:23:29 new version available: 0.119.3
Sep 03 11:23:29 evcc evcc[380731]: [mqtt  ] ERROR 2023/09/03 11:23:29 set evcc/loadpoints/1/mode/set: invalid value: NaN
Sep 03 11:23:29 evcc evcc[380731]: [mqtt  ] ERROR 2023/09/03 11:23:29 set evcc/loadpoints/1/phases/set: strconv.Atoi: parsing "NaN": invalid syntax
Sep 03 11:23:30 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:30 car connected
Sep 03 11:23:30 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:30 start charging ->
Sep 03 11:23:49 evcc evcc[380731]: [lp-1  ] INFO 2023/09/03 11:23:49 stop charging <-
Sep 03 11:25:31 evcc systemd[1]: Stopping evcc.service - evcc...
Sep 03 11:25:31 evcc systemd[1]: evcc.service: Deactivated successfully.
Sep 03 11:25:31 evcc systemd[1]: Stopped evcc.service - evcc.
Sep 03 11:25:33 evcc systemd[1]: Started evcc.service - evcc.
Sep 03 11:25:33 evcc evcc[381619]: [main  ] INFO 2023/09/03 11:25:33 evcc 0.119.3
Sep 03 11:25:33 evcc evcc[381619]: [main  ] INFO 2023/09/03 11:25:33 using config file: /etc/evcc.yaml
Sep 03 11:25:33 evcc evcc[381619]: [main  ] INFO 2023/09/03 11:25:33 starting ui and api at :7070
Sep 03 11:25:33 evcc evcc[381619]: [db    ] INFO 2023/09/03 11:25:33 using sqlite database: /var/lib/evcc/evcc.db
Sep 03 11:25:33 evcc evcc[381619]: [mqtt  ] INFO 2023/09/03 11:25:33 connecting evcc-1043879232 at tcp://192.168.178.30:1883
Sep 03 11:25:35 evcc evcc[381619]: [lp-1  ] WARN 2023/09/03 11:25:35 locking phase config to 3p for switchable charger
Sep 03 11:25:41 evcc systemd[1]: Stopping evcc.service - evcc...
Sep 03 11:25:41 evcc systemd[1]: evcc.service: Deactivated successfully.
Sep 03 11:25:41 evcc systemd[1]: Stopped evcc.service - evcc.
Sep 03 11:26:22 evcc systemd[1]: Started evcc.service - evcc.
Sep 03 11:26:22 evcc evcc[381891]: [main  ] INFO 2023/09/03 11:26:22 evcc 0.119.3
Sep 03 11:26:22 evcc evcc[381891]: [main  ] INFO 2023/09/03 11:26:22 using config file: /etc/evcc.yaml
Sep 03 11:26:22 evcc evcc[381891]: [main  ] INFO 2023/09/03 11:26:22 starting ui and api at :7070
Sep 03 11:26:23 evcc evcc[381891]: [db    ] INFO 2023/09/03 11:26:23 using sqlite database: /var/lib/evcc/evcc.db
Sep 03 11:26:23 evcc evcc[381891]: [mqtt  ] INFO 2023/09/03 11:26:23 connecting evcc-313269946 at tcp://192.168.178.30:1883
Sep 03 11:26:25 evcc evcc[381891]: [lp-1  ] WARN 2023/09/03 11:26:25 locking phase config to 3p for switchable charger
```

andig, 2023-09-03:
Außer dass bei Dir komische Werte in die API geschrieben werden:

```
Sep 03 11:23:29 evcc evcc[380731]: [mqtt  ] ERROR 2023/09/03 11:23:29 set evcc/loadpoints/1/mode/set: invalid value: NaN
Sep 03 11:23:29 evcc evcc[380731]: [mqtt  ] ERROR 2023/09/03 11:23:29 set evcc/loadpoints/1/phases/set: strconv.Atoi: parsing "NaN": invalid syntax
```

Ist kein Problem zu erkennen. Hier gibts nix zu tun.

chadibe0, 2023-09-03:
dann bleibe ich halt auf 119.2
Kann ja aber irgendwie nicht sein, dass ich der einzige bin, bei dem die 119.3 nicht funktioniert.

chadibe0, 2023-09-06:
Ich weiss nicht ob du noch an diesem issue interssiert bist. Mittlerweile habe ich herausgefunden woran es liegt. Ich habe die evcc.yaml neu erzeugt und dann lief die 119.4 wie erwartet. Danach habe ich begonnen all das in der evcc.yaml zu ergänzen, was unterschiedlich zu meiner ursprünglichen Konfigdatei war. Nach dem Einfügen dieser Zeilen lief es nicht mehr:

  planner:
    type: electricitymaps
    uri: https://api.co2signal.com/
    token: axl7RtcXMy2bvDQ0PB59oPfOFBRz9rVp
    zone: CH

Liegt also (ausnahmsweise) nicht an Easee

Gruss
Bernhard