evcc-io/evcc #9675: Soc via openWB schlägt fehl (authentication handshake failed)

martin-zh, 2023-09-02:
### Describe the bug

Der Fehler besteht seit Freitag Abend und betrifft diverse Fahrzeuge. Siehe Thread im openWB Forum:
https://openwb.de/forum/viewtopic.php?t=3134&start=690

Fehlermeldung:
2023-09-02 10:37:54: PID: 5034: Lp2: Error from EVCC: 2023/09/02 10:37:54 rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: context deadline exceeded" (LV0)

openWB Version 1.9.303.0 stable; keinerlei Updates in letzter Zeit. Internet Verbindung (openWB Cloud, Tesla SoC) ist ok.

### Steps to reproduce

SoC Abfrage via openWB. Schlägt immer fehl.

### Configuration details

```yaml
-
```

### Log details

```text
siehe openWB Log oben
```

### What type of operating system are you running?

Linux

### Version

-

andig, 2023-09-03:
Sollte nach Serviceneustart wieder funktionieren?

martinwashietl, 2023-09-03:
Hat kurz funktioniert, aber jetzt leider auch nicht mehr

sceph, 2023-09-06:
Funktioniert leider nicht. Die Fehlermeldung aus der openWB hatte sich allerdings zwischenzeitlich etwas verändert:

`2023-09-03 18:01:11: PID: 1996: Lp1: Error from EVCC: 2023/09/03 18:01:11 rpc error: code = Unknown desc = cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights. (LV0)`

Heute allerdings wieder die bisherige Fehlermeldung:

`2023-09-06 09:17:36: PID: 5043: Lp1: Error from EVCC: 2023/09/06 09:17:35 rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: context deadline exceeded" (LV0)`

andig, 2023-09-06:
Sollte jetzt endlich gelöst sein, siehe https://github.com/evcc-io/evcc/issues/9696

sceph, 2023-09-06:
Läuft nicht. Fehlermeldung diesmal:
`2023-09-06 20:36:18: PID: 12382: Lp1: Error from EVCC: 2023/09/06 20:36:18 rpc error: code = Unknown desc = cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights. (LV0)`

andig, 2023-09-06:
Läuft. Der Fehler ist ja ein anderer. Siehe https://github.com/evcc-io/evcc/issues/9641