evcc-io/evcc #9692: Skoda vehicle cannot be created

loeffelpan, 2023-09-02:
### Describe the bug

It seems like no vehicles can be found. Running homeassistant integration and MySkoda app is working fine.

Getting this error.
```
creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'skoda': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
```

### Steps to reproduce

Restarting EVCC.

### Configuration details

```yaml
network:
  schema: http
  host: toothless
  port: 7070

log: debug
levels:
  eebus: info

interval: 10s

meters:
- type: template
  template: senec-home
  usage: grid
  schema: http
  host: 172.20.2.8
  name: grid1
- type: template
  template: senec-home
  usage: pv
  schema: http
  host: 172.20.2.8
  name: pv2
- type: template
  template: senec-home
  usage: battery
  schema: http
  host: 172.20.2.8
  name: battery3

chargers:
- type: template
  template: ellipro
  ski: 5010F9B110FEE892C8BB0C0AB26707E1D8485E6B
  ip: 172.20.2.9
  name: wallbox5

vehicles:
- type: skoda
  title: Superb iV
  user: jan.noormann@gmail.com
  password: ...
  name: ev4
  features: ["coarsecurrent"]
  phases: 1
  capacity: 13
  onIdentify:
    mode: pv

loadpoints:
- title: Carport
  charger: wallbox5
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid1
    pv: pv2
    battery: battery3
  residualPower: 200

tariffs:
  grid:
    type: fixed
    price: 0.2516

  feedin:
    type: fixed
    price: 0.084

  co2:
    type: grünstromindex
    zip: 30926

mqtt:
  broker: 172.20.2.2:1883
  topic: evcc
  user: evcc
  password: ...

eebus:
  certificate:
      private: ...
      public: ...
```
```

### Log details

```text
[main  ] INFO 2023/09/02 23:24:37 evcc 0.119.3
[main  ] INFO 2023/09/02 23:24:37 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/02 23:24:37 starting ui and api at :7070
[db    ] INFO 2023/09/02 23:24:37 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/09/02 23:24:37 connecting evcc-474776930 at tcp://172.20.2.2:1883
[mqtt  ] DEBUG 2023/09/02 23:24:37 tcp://172.20.2.2:1883 connected
[eebus ] INFO 2023/09/02 23:24:37 Local SKI:  9f7c58b60b3d9e0c17600bebf1bac03e7b1b4a16
[main  ] ERROR 2023/09/02 23:24:40 creating vehicle ev4 failed: cannot create vehicle 'skoda': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
[site  ] WARN 2023/09/02 23:24:40 bufferSoc must be larger than prioritySoc
[site  ] INFO 2023/09/02 23:24:40 site config:
[site  ] INFO 2023/09/02 23:24:40   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/02 23:24:40     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/02 23:24:40     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/02 23:24:40     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/02 23:24:40   vehicles:
[site  ] INFO 2023/09/02 23:24:40     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/09/02 23:24:40 loadpoint 1:
[lp-1  ] INFO 2023/09/02 23:24:40   mode:        off
[lp-1  ] INFO 2023/09/02 23:24:40   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/02 23:24:40   meters:      charge ✓
[lp-1  ] INFO 2023/09/02 23:24:40     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/09/02 23:24:40 phase timer inactive
[lp-1  ] DEBUG 2023/09/02 23:24:40 pv timer inactive
[lp-1  ] DEBUG 2023/09/02 23:24:40 guard timer inactive
[site  ] WARN 2023/09/02 23:24:40 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/02 23:24:40 ----
[lp-1  ] DEBUG 2023/09/02 23:24:40 charge power: 0W
[site  ] DEBUG 2023/09/02 23:24:40 pv power: 0W
[site  ] DEBUG 2023/09/02 23:24:40 battery soc: 26%
[site  ] DEBUG 2023/09/02 23:24:40 battery power: 430W
[site  ] DEBUG 2023/09/02 23:24:40 grid power: -10W
[site  ] DEBUG 2023/09/02 23:24:40 site power: 620W
[lp-1  ] DEBUG 2023/09/02 23:24:40 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/02 23:24:40 charger status: A
[lp-1  ] INFO 2023/09/02 23:24:40 car disconnected
[lp-1  ] DEBUG 2023/09/02 23:24:40 set charge mode: off
[lp-1  ] DEBUG 2023/09/02 23:24:40 set min current: 6
[lp-1  ] DEBUG 2023/09/02 23:24:40 set max current: 16
[lp-1  ] DEBUG 2023/09/02 23:24:40 set priority: 0
[site  ] DEBUG 2023/09/02 23:24:50 ----
```
```

### What type of operating system are you running?

Docker container

### Version

0.119.3

SandnerSoft, 2023-09-03:
+1
mit Template: seat

bicaluv, 2023-09-03:
Hallo!  

Ich bekomme seid ca. einer Woche keine Daten mehr von meinem Skoda Citigo. Ein Trace `evcc -c evcc.yaml -l trace vehicle Citigo` ist vorhanden, endet aber mit: 

<details><summary>Trace</summary>
[skoda ] TRACE 2023/09/03 08:30:40 GET https://msg.volkswagen.de/fs-car/usermanagement/users/v1/VW/CZ/vehicles
[skoda ] TRACE 2023/09/03 08:30:40 {"error":{"errorCode":"RS.security.9007","description":"The context has not gained a legitimation due to missing access rights."}}
[main  ] ERROR 2023/09/03 08:30:40 creating vehicle Citigo failed: cannot create vehicle 'template': cannot create vehicle 'skoda': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'skoda': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
Capacity: 36.0kWh
Features: [Offline]
</details>

Bin mir nicht sicher, ob im Trace Tokens/Credentials sind, die ich hier nicht posten sollte.  

<details><summary>Konfiguration (/home/pi/etc/evcc.yaml)</summary>

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  # meter: debug

sponsortoken: *****
  
# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.2.240
  name: sma-grid
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.2.241
[... 95 lines cut ...]
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
    - type: telegram
      token: *****
      chats: *****
      - 1533452901 # list of chat ids
  # - type: email
  #   uri: smtp://<user: *****

```

</details>
Version: `0.119.3 (d2daf613e)`

herby57, 2023-09-03:
Same Problem here. 
Since the Update to 0.119.3 my Cupra(seat) is permanently offline. 
Seems to be the same issue.

pablopalec, 2023-09-03:
Seat MII user here, same problem:
`[main  ] INFO 2023/09/03 09:06:41 evcc 0.119.3
[main  ] INFO 2023/09/03 09:06:41 using config file: /etc/evcc.yaml
[db    ] INFO 2023/09/03 09:06:51 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2023/09/03 09:06:53 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
Capacity: 32.0kWh
Features: [Offline]`

SandnerSoft, 2023-09-03:
im ioBroker.vw-connect wurde der Bug bereits gefixt, vielleicht hilft das weiter.

loeffelpan, 2023-09-03:
Verweise bei sowas wären prima.
Aber ja, den hab ich auch gefunden.
https://github.com/TA2k/ioBroker.vw-connect/issues/255

Soll auf https://github.com/TA2k/ioBroker.vw-connect basieren. Müsste für Audi dieser [Commit](https://github.com/TA2k/ioBroker.vw-connect/commit/efa4091212b05c1c4b678dc953184b642d2809bd) sein.

Da sind auch für Skoda und Seat weitere commits von gestern.

andig, 2023-09-03:
VW stellt wohl wieder grossflächig um. Ich bräuchte Zugangsdaten zum testen, bitte an info@evcc.io

andig, 2023-09-03:
@loeffelpan hier gehts doch gar nicht um Audi?!

loeffelpan, 2023-09-03:
> @loeffelpan hier gehts doch gar nicht um Audi?!

Korrekt war nur der Link, weil es der gleiche Fehler ist, wie bei Audi.
Basiert Skoda und Seat denn auch auf https://github.com/TA2k/ioBroker.vw-connect ?

Da wurde es scheinbar gestern gefixt.

loeffelpan, 2023-09-03:
> VW stellt wohl wieder grossflächig um. Ich bräuchte Zugangsdaten zum testen, bitte an [info@evcc.io](mailto:info@evcc.io)

Haste im Kasten.

andig, 2023-09-03:
@loeffelpan der Ursprungsfehler- RS.security.9007- ist behoben. Bei Dir bekomme ich allerdings einen anderen Fehler in den Tests. Ursache unklar. Da Dein PW relativ komplex ist- könntest Du es bitte mal mit einem einfachen Passwort ohne Sondernzeichen versuchen?

Alle anderen insbesondere mit Seat-Problemen: bitte neues Issue und Zugangsdaten an info@evcc.io. Hier ging es erstmal um Skoda und das Problem ist gelöst.

closed 2023-09-03 by andig

andig, 2023-09-03:
@herby57 für Seat siehe https://github.com/evcc-io/evcc/issues/9606, ich bräuchte noch Zugangsdaten.

loeffelpan, 2023-09-03:
> @loeffelpan der Ursprungsfehler- RS.security.9007- ist behoben. Bei Dir bekomme ich allerdings einen anderen Fehler in den Tests. Ursache unklar. Da Dein PW relativ komplex ist- könntest Du es bitte mal mit einem einfachen Passwort ohne Sondernzeichen versuchen?
> 
> Alle anderen (@herby57) insbesondere mit Seat-Problemen: bitte neues Issue und Zugangsdaten an [info@evcc.io](mailto:info@evcc.io). Hier ging es erstmal um Skoda und das Problem ist gelöst.

Danke. Funktioniert für mich nach eigenem docker-build vom master-repo.
Auch ohne Kennwortänderung (die ich natürlich trotzdem machen werde).

herby57, 2023-09-03:
> @herby57 für Seat siehe #9606, ich bräuchte noch Zugangsdaten.

@andig sind jetzt in deinem Postfach. 

Danke