evcc-io/evcc #9699: Influx: Tag data by month and year

cschlipf, 2023-09-03:
**Is your feature request related to a problem? Please describe.**
The InfluxDB integration is fantastic for creating your own dashboards - even for use cases beyond vehicle charging. For example my dashboards [in this thread](https://github.com/evcc-io/evcc/discussions/4213#discussioncomment-6619940).
However Influx has some grave limitations:
- You cannot easily group data by months or years
- You cannot change the tags later (e.g. I tried to create a script to add missing tags later, but Influx does not support that)

**Describe the solution you'd like**
Tag data that is written by evcc into influx by month and year. E.g. a month tag with values like [1,2,...12] or ["Jan", "Feb",..."Dec"] and a year tag containing the year.

I neither know the EVCC code well nor do I now Go, but from what I see this change could be done in [influxdb.go in Line 72](https://github.com/evcc-io/evcc/blob/5b994e729e4e9e3597758c6e18fa5a629cb49488/server/influxdb.go#L72). The given tags map just needs to be extended by tags for months and year and hopefully this would solve this for all the data written by EVCC into Influx.

**Describe alternatives you've considered**
Tried to create a script that alters the data written by EVCC and adds tags. However Influx does not allow to change or add tags later.

**Additional context**
I know this request is not the core use case of EVCC, but due to the limitations of Influx, I haven't found another alternative yet.

andig, 2023-09-03:
This doesn't seem like a good idea, especially since Influx already has the data. Is there any indication, that doing so is best practice?

cschlipf, 2023-09-03:
@andig See this [thread](https://community.influxdata.com/t/how-to-group-by-time-month/10633), where Anaisdg (Influx Staff) is suggesting exactly this as solution. She suggest however to use the Date proxessor plugin of telegraf, which I am not sure whether it works with EVCC.

In any other DB I would write a script to add this later in a post-processing step. However this again is not possible with influx: https://github.com/influxdata/influxdb/issues/3904

andig, 2023-09-03:
> The second is to use [flux 813](https://github.com/influxdata/flux/blob/master/docs/SPEC.md#aggregateWindow) to query your data. It allows you to query by year and month.

Since flux has been available for a couple of years now this seems to be the way to go?

closed 2023-09-03 by andig

cschlipf, 2023-10-30:
@andig Note that Flux is going out of support:
[image]
https://docs.influxdata.com/flux/v0/stdlib/date/month/

I totally understand that I am trying to solve an issue, which is not in the scope of EVCC. However for EVCC it would be a small thing to add, while writing the data points. For me I do not have a good solution other than reading all the data, annotating it with the month and year tag and then writing the data into a new series (so ending up with duplicate data stores).

Maybe an idea that could please everyone? Would it be a good idea to add a generic tagging to the Influx configuration in EVCC? Something like

```
influx:
  url: http://localhost:8086
  database: evcc
  user: grafana
  password: xxxxxxxx
  tags:
     - key: month
       value: ${month}
     - key: year
       value: ${year}
     - key: evcc-instance
       value: garage-evcc
```

Like you can see this could also be used to tag the values by other dimensions, like an evcc-instance just in case you want to store data from multiple running EVCC instances in the same table.