evcc-io/evcc #9704: Vehicle order not deterministic

naltatis, 2023-09-03:
### Describe the bug

If you've configure multiple vehicles evcc does not expose them in the same order after restarts.

Since the index of the vehicle is used for API calls (assign vehicle to loadpoint) and for storing vehicle data (minsoc, targettime, ...) this is not only an optical problem.

Expected behavior:
Vehicles are always returned in the order they are defined in the yaml file. 

[vehicle sort.webm](https://github.com/evcc-io/docs/assets/152287/9ac7aa5a-854c-44ca-b528-8d36a7869599)

### Steps to reproduce

- run evcc with demo.yaml
- open vehicle dropdown
- stop and start evcc
- vehicle order changes

### Configuration details

```yaml
demo.yaml
```

### Log details

```text
see video
```

### What type of operating system are you running?

Linux

### Version

master

andig, 2023-09-03:
All vehicles or only db-configured?

naltatis, 2023-09-03:
`demo.yaml` > only yaml

closed 2023-09-03 by andig