evcc-io/evcc #9706: Sponsortoken: Token unbekannt, https://sponsor.evcc.io zeigt Fehlermeldung - Backend gestört?

drschliwago, 2023-09-03:
### Describe the bug

Mein Sponsortoken ist seit heute abend unbekant, obwohl ein aktives Sponsoring besteht.

Der Aufruf https://sponsor.evcc.io mit Klick auf Sponsortoken erhalten bringt folgende Fehlermeldung: `code exchange failed: oauth2: "bad_verification_code" "The code passed is incorrect or expired." "https://docs.github.com/apps/managing-oauth-apps/troubleshooting-oauth-app-access-token-request-errors/#bad-verification-code"

Das zusätzliche manuelle Bezahlen von weiteren 2 Euro bringt keine Abhilfe.

Die E-Mail info@evcc.io scheint gestört zu sein: `Action: failed
Final-Recipient: rfc822; [info@evcc.io](mailto:info@evcc.io)
Original-Recipient: rfc822; [info@evcc.io](mailto:info@evcc.io)
Diagnostic-Code: smtp; 552 5.2.3 Message length exceeds limit set by recipient
Status: 5.2.3`

`

`

### Steps to reproduce

1. evcc lokal aufrufen
2. Fehlermeldung: sponsorship required, see https://github.com/evcc-io/evcc#sponsorship, obwohl Sponsoring vorhanden
3. https://sponsor.evcc.io aufrufen: Fehlermeldung `code exchange failed: oauth2: "bad_verification_code" "The code passed is incorrect or expired." "https://docs.github.com/apps/managing-oauth-apps/troubleshooting-oauth-app-access-token-request-errors/#bad-verification-code"
`

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxx

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxxxxxxxx

# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 10.0.0.12  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 10.0.0.12  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 10.0.0.12  
  port: 502  
  name: battery3

chargers:
- type: template
  template: easee 
  user: xxxxxxxxx
  password: xxxxxxxx 
  charger: xxxxxx  
  name: wallbox5

vehicles:
- type: template
  template: vw 
  title: e-Up! Style Plus  
  user:xxxxxxxxxx
  password: xxxxxxxx  
  vin: xxxxxxxxxx  
  capacity: 32.3  
  phases: 2  
  mode: pv  
  minSoC: 30  
  targetSoC: 90  
  minCurrent: 5 
  name: ev4

loadpoints:
- title: Easee
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: HOME
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    
    type: tibber
    token: "xxxxxxxxxxxx" # access token

#  planner:
#    type: gruenstromindex
#    zip: 38126
```

### Log details

```text
[main  ] INFO 2023/09/03 19:40:22 evcc 0.119.4
[main  ] INFO 2023/09/03 19:40:22 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/03 19:40:22 starting ui and api at :7070
[db    ] INFO 2023/09/03 19:40:22 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/09/03 19:40:22 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'easee': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
[main  ] FATAL 2023/09/03 19:40:22 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.119.4

naltatis, 2023-09-03:
@drschliwago die Meldung ist komisch. Kannst du einen Screenshot davon machen?

Zum Sponsoring: Bitte wähle eine der vordefinierten Stufen (tiers) aus. Wir können die Freibetrags-Felder in Github leider nicht deaktivieren. Aber Sponsortoken gibt es nur über die festen Stufen. Bei weiteren Fragen wende dich gerne direkt an info@evcc.io.

closed 2023-09-03 by naltatis

drschliwago, 2023-09-03:
Hey Michael,ich habe ein aktives Sponsoring für 2 Euro je Monat ja laufen. Warum auch immer wie dies nicht mehr erkannt.Warum auch immer ist von jetzt auf gleich mein bestehender Sponsortoken ungültig und ich erhalte auch keinen neuen.Viele Grüße--René SchliwaWolfshagenweg 938126 BraunschweigE-Mail: ***@***.***: (01 63) 7 350 530

naltatis, 2023-09-03:
> Warum auch immer wie dies nicht mehr erkannt.Warum auch immer ist von jetzt auf gleich mein bestehender Sponsortoken ungültig und ich erhalte auch keinen neuen.

Ok, schräg. @andig magst du das mal prüfen?

drschliwago, 2023-09-04:
Hi zusammen,

Ich sende euch zur Sicherheit mal die Sponsorship History… vielleicht ist da was nicht berücksichtigt worden:

￼
Viele Grüße