evcc-io/evcc #9707: Verbindung zu Easee Home

obelast, 2023-09-03:
### Describe the bug

Ich bekomme leider die Verbindung zur Easee Home nicht konfiguriert.
Habe einen Raspi Pi 4 und habe gelesen, dass es in der Vergangenheit schon Probleme mit zu vielen Zugriffen?? auf die Easee-Box gab...
Greift evcc über die Easee-Cloud auf die Box zu oder über die IP-Adresse.
Wenn ich die IP-Adresse anpinge bekomme ich eine Antwort und keine Fehler...
Vielen Dank für alle Hinweise

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
sponsortoken: *****  
  
network:  
  schema: http  
  host: 192.168.178.110  
  port: 7070  
    
tariffs:  
  currency: EUR  
  grid:  
    type: fixed  
    price: 0.4  
  
log: debug  
#levels:  
#  cache: error  
  
interval: 31s # control cycle interval  
  
meters:  
- name: grid  
  type: template  
  template: vzlogger  
  usage: grid  
  host: 192.168.178.110  
  port: 8081  
  uuid: 9ceb43e0-8794-11ed-b6b6-139ce08be86a  
   
chargers:  
- name: my_charger  
  type: template  
  template: easee  
  user: *****  
  password: *****  
  charger: EH_1    
  
loadpoints:  
- title: Wallbox  
  charger: my_charger  
  mode: pv  
  phases: 1  
  mincurrent: 6  
  maxcurrent: 13  
  resetOnDisconnect: true  
  enable:
```

### Log details

```text
[main  ] INFO 2023/09/03 19:58:58 evcc 0.119.3
[main  ] INFO 2023/09/03 19:58:58 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/03 19:58:58 starting ui and api at :7070
[db    ] INFO 2023/09/03 19:58:58 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/09/03 19:58:59 cannot create charger 'my_charger': cannot create charger 'template': cannot create charger 'easee': unexpected status: 403 (Forbidden)
[main  ] FATAL 2023/09/03 19:58:59 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.119.3

MarSlz, 2023-09-03:
Moin Obelast,
Soweit ich weiß bietet die Easee keine lokale Schnittstelle. Die Kommunikation erfolgt immer über die Cloud, wobei die Anzahl der Zugriffe limitiert sind. Da hilft dann mal abwarten.

zugangsdaten stimmen?
Bei Charger muss glaub ich der Name des Chargers in der Cloud angeben werden ( bei mir die Seriennummer)

ich drück die Daumen!

drschliwago, 2023-09-03:
#9706 - gleiche Thematik - dann sind wir bereits 2

naltatis, 2023-09-03:
Ja, wie @MarSlz schreibt geht das aktuell immer über Easee Server. Hier gibts aber aktuell keine bekannten Serverprobleme. Meine Vermutung ist, dass es an den Zugangsdaten liegt. Klassiker sind Sonderzeichen oder führende Nullen im Passwort die in Yaml eine Sonderbedeutung haben. Versuch bitte mal Nutzer und Passwort in Anführungszeichen zu schreiben.

closed 2023-09-03 by naltatis

obelast, 2023-09-04:
Hallo Zusammen,
vielen Dank für die Antworten. Leider funktionert das mit den Anführungszeichen nicht und ich habe nur einen Punkt im Passwort und der ist auch nicht am Anfang.
Was bedeutet denn "Abwarten"? Wie lange sollte ich dann warten?