evcc-io/evcc #9708: API for updating site.voltage

binsentsu, 2023-09-03:
**Is your feature request related to a problem? Please describe.**
The default site voltage is currently a fixed value set through evcc.yaml. In my place voltage can vary during the day between 230 and 240. When calculating the target charger ampere it would be nice if calculation takes the real actual voltage into account iso the fixed one. Dividing the residual power by a too low voltage can now results in a charger target A which is just too high, so the site power becomes a notch positive after charger commits to this value. When this happens there is a constant +1/-1 target A sent to the charger during each cycle interval.

**Describe the solution you'd like**
Create an API so the site.voltage can be updated with the live value.

**Describe alternatives you've considered**
Setting the fixed voltage to the upper value which generally occurs during a day.

**Additional context**
/

andig, 2023-09-04:
Could you use residualPower as workaround? I don't think we'd want flexible voltage. Much more since the controlling algorithm tries to eliminate the delta- so it should level out in the steady state.

binsentsu, 2023-09-04:
Ok. Will look into that. Thanks for insights.

closed 2023-09-04 by binsentsu