evcc-io/evcc #9714: Unterstützung vom Carlo Gavazzi EM540

deantomasevic, 2023-09-04:
**Is your feature request related to a problem? Please describe.**
According to the documentation, the new EM540 energy meter is not currently supported.

https://www.gavazziautomation.com/images/PIM/DATASHEET/DEU/EM540_DS_DEU.pdf

Maybe someone knows if this is already compatible, since it is also connected via ModbusRTURS485.

Thanks!

Best,
Dean

premultiply, 2023-09-04:
Did you already tried to use one of the existing implementations?
https://docs.evcc.io/docs/devices/meters#carlo-gavazzi

I haven't taken a deeper look into it yet.

deantomasevic, 2023-09-04:
I haven't bought the hardware yet, because I wanted to check first if everything is supported.

andig, 2023-09-05:
@deantomasevic please give it a try and lets reopen if there's an issue.

closed 2023-09-05 by andig