evcc-io/evcc #9716: SoC or Temperatur not taken into account when only charger is configured

panzerdev, 2023-09-04:
### Describe the bug

[image]

"Charging" not stopping then `targetSoc` reached or exceeded.

found by @naltatis 
https://github.com/evcc-io/evcc/blob/216dcba3de6a241fcdadd0d0d6adebef424d4943/core/loadpoint.go#L799-L800

### Steps to reproduce

1. Set targetSoc below current Soc

### Configuration details

```yaml
loadpoints:
  - title: "Heizstab"
    charger: ac-thor 
    mode: pv # charge mode (off, now, minpv, pv)
    phases: 1
    minCurrent: 0
    maxCurrent: 13
    meter: ac-thor-meter
    priority: 5

chargers:
  - name: ac-thor
    icon: waterheater
    type: custom
    enable:
      source: http
      method: GET
      uri: https://my-url.home/heizstab/enable?enable=${enable}
    enabled:
      source: http
      method: GET
      uri: https://my-url.home/heizstab/status
      jq: .enabled
    status:
      source: http
      method: GET
      uri: https://my-url.home/heizstab/status
      jq: .status
    maxcurrent:
      source: http
      method: GET
      uri: https://my-url.home/heizstab/power?power=${maxcurrent}
    maxcurrentmillis:
      source: http
      method: GET
      uri: https://my-url.home/heizstab/power?power=${maxcurrentmillis}
    soc:
      source: http
      uri: https://my-url.home/heizstab
      jq: .temp1
      scale: 0.1
    features: [integrateddevice, heating]
```

### Log details

```text
evcc_1  | [site  ] DEBUG 2023/09/04 11:58:54 ----
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:58:54 charge power: 2648W
evcc_1  | [site  ] DEBUG 2023/09/04 11:58:54 pv power: 10925W
evcc_1  | [site  ] DEBUG 2023/09/04 11:58:54 battery soc: 36%
evcc_1  | [site  ] DEBUG 2023/09/04 11:58:54 battery power: 0W
evcc_1  | [site  ] DEBUG 2023/09/04 11:58:54 grid power: -28W
evcc_1  | [site  ] DEBUG 2023/09/04 11:58:54 site power: 72W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:04 ----
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:04 charge power: 2650W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:04 pv power: 10972W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:04 battery soc: 36%
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:04 battery power: 0W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:04 grid power: -19W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:04 site power: 81W
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:06 set target soc: 50
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 ----
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:06 charge power: 2639W
evcc_1  | [main  ] DEBUG 2023/09/04 11:59:06 lp Heizstab at prio 5 gets additional 0W from Carport oben at prio 0, 7461W from Carport unten at prio 0, total 7461W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 pv power: 10937W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 battery soc: 36%
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 battery power: 0W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 grid power: -25W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 giving loadpoint priority for additional: 7461W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:06 site power: -7386W
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:06 charge currents: [12.1 0 0]A
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:06 detected active phases: 1p
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:06 charger status: C
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:07 pv charge current: 45.1A = 13A + 32.1A (-7386W @ 1p)
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 ----
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:14 charge power: 2648W
evcc_1  | [main  ] DEBUG 2023/09/04 11:59:14 lp Heizstab at prio 5 gets additional 0W from Carport oben at prio 0, 7448W from Carport unten at prio 0, total 7448W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 pv power: 10967W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 battery soc: 36%
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 battery power: -29W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 grid power: -53W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 giving loadpoint priority for additional: 7448W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:14 site power: -7430W
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:14 charge currents: [12.1 0 0]A
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:14 detected active phases: 1p
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:14 charger status: C
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:15 pv charge current: 45.3A = 13A + 32.3A (-7430W @ 1p)
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:24 ----
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:24 charge power: 2656W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:24 pv power: 10914W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:24 battery soc: 36%
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:24 battery power: -47W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:24 grid power: 1W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:24 site power: 54W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:34 ----
evcc_1  | [lp-3  ] DEBUG 2023/09/04 11:59:34 charge power: 2641W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:34 pv power: 10918W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:34 battery soc: 36%
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:34 battery power: -52W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:34 grid power: -0W
evcc_1  | [site  ] DEBUG 2023/09/04 11:59:34 site power: 48W
```

### What type of operating system are you running?

Linux

### Version

commit 54d3057042c7c51c0b9a7a810216621d3ad6512f

andig, 2023-09-06:
Spannende Frage: woran erkennen wir, dass wir einen Soc aus anderer Quelle als von einem identifizierten Fahrzeug bekommen? Es reicht eigentlich nicht zu schauen, ob der Charger `api.Battery` unterstützt sondern es muss ja im allgemeinen Fall auch ein passendes Fahrzeug angeschlossen sein das darüber ausgelesen werden kann. 

Das ergäbe folgende Fälle:

- Fahrzeug identifiziert
- Charger unterstützt Soc:
  - Fahrzeug angeschlossen
  - integriertes Gerät: dann sollte das auch immer einen Wert liefern

In jedem Fall ist auch soc=0 ein gültiger Wert, im Zweifel gibts noch keinen und wir verhalten uns daher konservativ.

/cc @premultiply

naltatis, 2023-09-06:
> integriertes Gerät: dann sollte das auch immer einen Wert liefern

Das ist keine korrekte Annahme. Unsere Idee war ja, die [Mini-Loadpoints](https://github.com/evcc-io/evcc/discussions/7235) auch an der Eigenschaft "integratedDevice" festzumachen. Also primär "einfache Geräte" wie Schaltsteckdosen und so wo es kein Fahrzeug gibt. Vielleicht müssen wir das auch noch mal überdenken/hinterfragen.

andig, 2023-09-06:
> Das ist keine korrekte Annahme.

Das passt schon. Das wäre dann ein Charger, aber ohne soc. Da kann also auch nix abgerufen werden. Dein Fall ist außerhalb der Auswahl oben.

andig, 2023-09-06:
> Wäre es evtl. eine Idee den Charger als einzige Quelle für den SoC zu betrachten und dieser Fragt das Fahrzeug ab wenn vorhanden oder nutzt eine andere Quelle? Sobald Wallboxen SoCs abfragen können wird das ja auch relevant sein, oder?

Und dann? Heute kann das keine Wallbox :/

andig, 2023-09-10:
@panzerdev schau mal bitte obs jetzt tut

closed 2023-09-10 by andig

panzerdev, 2023-09-11:
@andig Funktioniert, super Sache! Danke!