evcc-io/evcc #9718: easee:  switch phases unexpected status: 401 (Unauthorized)

horst789github, 2023-09-04:
### Describe the bug

Installierte Version: 0.119.3

switch phases: Get \"https://api.easee.com/api/sites/363750/circuits/347224/settings\": unexpected status: 401 (Unauthorized)
smart charging: Post \"https://api.easee.com/api/chargers/EHD9PXXX/settings\": unexpected status: 401 (Unauthorized)
switch phases: Get \"https://api.easee.com/api/sites/363750/circuits/347224/settings\": unexpected status: 401 (Unauthorized)

der automatische oder manuelle phasen wechsel geht nicht mehr... habe jetzt mal evcc container neugestartet, schon gehts wieder.
nur als info für euch, vielleicht könnt ihr damit was anfangen.

### Steps to reproduce

-

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Version

0.119.3

GrimmiMeloni, 2023-09-04:
Will be addressed by #9717

josefzahner, 2023-09-04:
Endlich wird das Problem angegangen, Danke @GrimmiMeloni !

jheinitz, 2023-09-04:
Das Problem hatte ich heute Mittag auch. Allerdings hat Easee zuvor Wartungsarbeiten am Portal und an der API durchgeführt. Ich nehme an, dass dadurch wieder alle Sessions weggeflogen sind. Nach einem Neustart von **evcc** ging es bei mir direkt wieder.

TTXaser2, 2023-09-04:
Dito, den ganzen Tag über schon. zwei-drei Neustarts hab ich auch schon hinter mir, jetzt geht's schon wieder nicht:

[image]

andig, 2023-09-04:
Siehe #9717

closed 2023-09-04 by andig