evcc-io/evcc #9719: Charger disabled: vehicle not capable of start/stop when nothing connected to chager

Diondk, 2023-09-04:
### Describe the bug

Since the update to v0.119.4 I keep getting an notification saying the charger disabled: vehicle not capable of start/stop. At this moment there is nothing connected to my tesla wall charger.

[image]

### Steps to reproduce

1. update to v0.119.4 
2. open webpage
3. see notifications coming in. 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 10s # control cycle interval

   
    
meters:
- name: pv2
  type: custom
  power:
      source: http
      uri: http://192.168.1.xxx/e
      method: GET # default HTTP method
      headers:
        - content-type: application/json
      insecure: true # set to true to trust self-signed certificates
      jq: .[0].ps0 # parse response json
      scale: 1.0 # floating point factor applied to result, e.g. for kW to W conversion
      timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration
- name: dsmr
  type: custom
  power:
    source: calc
    add:
    - source: http
      uri: http://192.168.1.xxx/api/v2/consumption/electricity-live
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{currently_delivered}| join(" ")
      scale: 1
      timeout: 10s
    - source: http
      uri: http://192.168.1.xxx/api/v2/consumption/electricity-live
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{currently_returned}| join(" ")
      scale: -1
      timeout: 10s
  energy:
    source: calc
    add:
    - source: http
      uri: http://192.168.1.xxx/api/v2/consumption/today
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{electricity_merged}| join(" ")
      scale: 1
      timeout: 10s
    - source: http
      uri: http://192.168.1.xxx/api/v2/consumption/today
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{electricity_returned_merged}| join(" ")
      scale: -1
      timeout: 10s
  currents:
  - source: http
    uri: http://192.168.1.xxx/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: xxx
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l1}| join(" ")
    scale: 1
    timeout: 10s
  - source: http
    uri: http://192.168.1.xxx/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: xxx
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l2}| join(" ")
    scale: 1
    timeout: 10s
  - source: http
    uri: http://192.168.1.xxx/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: xxx
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l3}| join(" ")
    scale: 1
    timeout: 10s

chargers:
- type: template
  template: twc3 
  host: 8.8.8.8
  name: wallbox3

loadpoints:
- title: Oprit
  charger: wallbox3
  mode: off
  phases: 3
  mincurrent: 2
  maxcurrent: 16
  resetOnDisconnect: false

site:
  - title: Huis
    meters:
      grid: dsmr # grid meter reference
      pv: # (pvs = veraltet)
        - pv2  # first pv meter reference
vehicles:
  - name: my_car
    type: template
    template: tesla
    title: Executor # Wird in der BenutzeroberflÃ¤che angezeigt (Optional)
    accessToken: xxxx
    refreshToken: xxx
    vin: xxx
    #onIdentify: # set defaults when vehicle is identified
    mode: pv # enable PV-charging when vehicle is identified
    minSoc: 30 # immediately charge to 0% regardless of mode unless "off" (disabled)
    targetSoc: 90 # limit charge to 90%

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: "xx" 
  # grid:
    #either static grid price (or price zones)
    # type: fixed
    # price: 0.58 # EUR/kWh
    # zones:
      # - days: Mo-Fr
        # hours: 07-21
        # price: 0.73 # EUR/kWh
      # - days: Sa,So
        # price: 0.58 # EUR/kWh
mqtt:
  broker: 8.8.8.8:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: evcc
  password: xxxx
```

### Log details

```text
root@evcc:~# evcc --log debug
[main  ] INFO 2023/09/04 11:32:36 evcc 0.119.4
[main  ] INFO 2023/09/04 11:32:36 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/04 11:32:36 starting ui and api at :7070
[db    ] INFO 2023/09/04 11:32:36 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/09/04 11:32:36 connecting evcc-836473421 at tcp://192.168.1.3                                                                                                                                                             :1883
[mqtt  ] DEBUG 2023/09/04 11:32:36 tcp://192.168.1.3:1883 connected
[site  ] INFO 2023/09/04 11:32:37 site config:
[site  ] INFO 2023/09/04 11:32:37   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/09/04 11:32:37     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/09/04 11:32:37     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/04 11:32:37   vehicles:
[site  ] INFO 2023/09/04 11:32:37     vehicle 1: range ✓ finish ✓ status ✓ clima                                                                                                                                                             te ✗ wakeup ✓
[lp-1  ] INFO 2023/09/04 11:32:37 loadpoint 1:
[lp-1  ] INFO 2023/09/04 11:32:37   mode:        off
[lp-1  ] INFO 2023/09/04 11:32:37   charger:     power ✓ energy ✗ currents ✓ pha                                                                                                                                                             ses ✗ wakeup ✗
[lp-1  ] INFO 2023/09/04 11:32:37   meters:      charge ✓
[lp-1  ] INFO 2023/09/04 11:32:37     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/09/04 11:32:37 phase timer inactive
[lp-1  ] DEBUG 2023/09/04 11:32:37 pv timer inactive
[lp-1  ] DEBUG 2023/09/04 11:32:37 guard timer inactive
[site  ] WARN 2023/09/04 11:32:37 interval <30s can lead to unexpected behavior,                                                                                                                                                              see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/04 11:32:37 ----
[lp-1  ] DEBUG 2023/09/04 11:32:37 charge power: 0W
[site  ] DEBUG 2023/09/04 11:32:37 pv power: 5741W
[site  ] DEBUG 2023/09/04 11:32:37 grid power: -5201W
[site  ] DEBUG 2023/09/04 11:32:37 grid currents: [6 7 7]A
[site  ] DEBUG 2023/09/04 11:32:37 site power: -5201W
[lp-1  ] DEBUG 2023/09/04 11:32:37 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2023/09/04 11:32:37 charge currents: [0.3 0 0.4]A
[lp-1  ] DEBUG 2023/09/04 11:32:37 charger status: A
[lp-1  ] INFO 2023/09/04 11:32:37 car disconnected
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.

andig, 2023-09-04:
Could you please add a log of the actual problem with

    twc: trace
    tesla: trace

will reopen when log available.

closed 2023-09-04 by andig

KristianHeider, 2024-02-06:
Sorry got same problem and even has never worked at all. Do I have to get token each day, each start. Neverless, I got connection with Y and Tesla wall connector. But there was no interaction. 
```
[site  ] DEBUG 2024/02/06 20:29:40 ----
[lp-1  ] DEBUG 2024/02/06 20:29:40 charge power: 1824W
[site  ] DEBUG 2024/02/06 20:29:40 pv power: 0W
[site  ] DEBUG 2024/02/06 20:29:40 site power: 2224W
[lp-1  ] DEBUG 2024/02/06 20:29:40 charge voltages: [225 2.5 0]V
[lp-1  ] DEBUG 2024/02/06 20:29:40 detected connected phases: 1p
[lp-1  ] DEBUG 2024/02/06 20:29:40 charge currents: [8.1 0 0.5]A
[lp-1  ] DEBUG 2024/02/06 20:29:40 detected active phases: 1p
[lp-1  ] DEBUG 2024/02/06 20:29:40 charger status: C
[lp-1  ] DEBUG 2024/02/06 20:29:40 vehicle soc: 68%
[lp-1  ] DEBUG 2024/02/06 20:29:40 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/02/06 20:29:40 vehicle range: 284km
[lp-1  ] DEBUG 2024/02/06 20:29:40 pv charge current: 0A = 0A + -9.67A (2224W @ 1p)
[lp-1  ] DEBUG 2024/02/06 20:29:40 site power 2224W >= 0W disable threshold
[lp-1  ] DEBUG 2024/02/06 20:29:40 pv disable in 0s
[lp-1  ] DEBUG 2024/02/06 20:29:40 pv disable timer elapsed
[lp-1  ] ERROR 2024/02/06 20:29:40 charger disable: vehicle not capable of start/stop
[site  ] DEBUG 2024/02/06 20:29:50 ----
[lp-1  ] DEBUG 2024/02/06 20:29:51 charge power: 1826W
[site  ] DEBUG 2024/02/06 20:29:51 pv power: 0W
[site  ] DEBUG 2024/02/06 20:29:51 site power: 2226W
[lp-1  ] DEBUG 2024/02/06 20:29:51 charge voltages: [225 2.5 0]V
[lp-1  ] DEBUG 2024/02/06 20:29:51 detected connected phases: 1p
[lp-1  ] DEBUG 2024/02/06 20:29:51 charge currents: [8.1 0 0.5]A
[lp-1  ] DEBUG 2024/02/06 20:29:51 detected active phases: 1p
[lp-1  ] DEBUG 2024/02/06 20:29:51 charger status: C
[lp-1  ] DEBUG 2024/02/06 20:29:51 vehicle soc: 68%
[lp-1  ] DEBUG 2024/02/06 20:29:51 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/02/06 20:29:51 vehicle range: 284km
[lp-1  ] DEBUG 2024/02/06 20:29:51 pv charge current: 0A = 0A + -9.68A (2226W @ 1p)
[lp-1  ] DEBUG 2024/02/06 20:29:51 site power 2226W >= 0W disable threshold
[lp-1  ] DEBUG 2024/02/06 20:29:51 pv disable in 0s
[lp-1  ] DEBUG 2024/02/06 20:29:51 pv disable timer elapsed
[lp-1  ] ERROR 2024/02/06 20:29:51 charger disable: vehicle not capable of start/stop