evcc-io/evcc #9739: Easee Home won't charge with less than full power

TheFreaker86, 2023-09-05:
### Describe the bug

Hallo und Moin!
I'm new to Home Assistant and EVCC since I own a PV on the roof for just more than a week. My PV consists of the Sungrow SH10RT with a 9,6 kWh battery and an additional Balkonkraftwerk. I have set up HA and EVCC in the prebuild HA Hyper-V VM. I have built the evcc.yaml using the evcc.exe under Windows.

It all works quite well, but now that I have plugged in an electric car (not my own) into the wallbox for the first time to test the PV regulated charging. EVCC recognizes the car as guest, the Easee and shows all the parameters (battery, PV, charging etc.) during charging correctly. But it charges only with full 22 kW power with pulling energy from the PV, battery and grid. It does not matter what settings are made in the EVCC GUI, it's always full power. I switched on the debugging for the charger and the site to see what's the issue, but I don't see any mistakes on my part. 

The only weird thing I have noticed in the logs is that the reading for  PV 2 power is 0. But this value is readable from the inverter.

please advise. Thank you very much in advance!

P.S.: Is the preferred language here English or German?

### Steps to reproduce

don't know what to tell you here, I just plug in the car, start the charging with the RFID and watch the charging process in Home Assistant/EVCC and the Easee App.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  charger: debug
  site: debug

# unique installation id
plant: XXXX

interval: 10s # control cycle interval

sponsortoken: XXXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
- type: template
  template: sungrow-hybrid 
[... 175 lines cut ...]
[lp-1  ] DEBUG 2023/09/05 15:21:58 charger status: A
[site  ] DEBUG 2023/09/05 15:22:08 ----
[lp-1  ] DEBUG 2023/09/05 15:22:08 charge power: 0W
[site  ] ERROR 2023/09/05 15:22:08 pv 2 power: unexpected status: 401 (Unauthorized)
[site  ] DEBUG 2023/09/05 15:22:08 pv power: 3080W
[site  ] DEBUG 2023/09/05 15:22:08 battery 1 power: 0W
[site  ] DEBUG 2023/09/05 15:22:08 battery 1 soc: 100%
[site  ] DEBUG 2023/09/05 15:22:08 battery 2 power: 0W
[site  ] DEBUG 2023/09/05 15:22:08 battery 2 soc: 100%
[site  ] DEBUG 2023/09/05 15:22:08 battery soc: 110%
[site  ] DEBUG 2023/09/05 15:22:08 battery power: 0W
[site  ] DEBUG 2023/09/05 15:22:08 grid power: -2693W
[site  ] DEBUG 2023/09/05 15:22:08 site power: -2393W
[lp-1  ] DEBUG 2023/09/05 15:22:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/05 15:22:08 charge total import: 102.927kWh
[lp-1  ] DEBUG 2023/09/05 15:22:08 charger status: A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.119.4

andig, 2023-09-05:
Use nightly version.

closed 2023-09-05 by andig

TheFreaker86, 2023-09-06:
Gibt's das noch in Home Assistant? Ich hab was gelesen dass es das nicht mehr gibt