evcc-io/evcc #9747: Kia-API Bug - reporting wrong SoC: Maybe a workaround is possible?

markusjnagel, 2023-09-05:
**Is your feature request related to a problem? Please describe.**
When a charge ends or pauses (e.g. I have it configured to not charge if home battery SoC is below 50%), our Kia SoC reverts to a previous state.

Example from today (logs below): During charging, SoC is reported (API) and estimated (evcc) fine:
[image]

After the charging is paused (battery reached 50% before target-SoC was reached), it reverts to the SoC that the charging session was started with (73%)
[image]

**Describe the solution you'd like**
This is obviously not an evcc bug - evcc reports correctly what :hankey: the API returns. However, since the SoC seems to be correctly reported _while the charging is still active_, maybe the last call to the API could be made _before_ the charging stops, not after. Yes, that's counterintuitive, but please check my logs below.

**Describe alternatives you've considered**
n/a

**Additional context**
While loading in "PV" mode, this is just a nuisance and not really an issue - until we hit the desired SoC. Car max SoC is 100%, it should be controlled via evcc.
However, what happens now is - it charges fine until it reaches the desired SoC, then charging stops. Now the wrong SoC is reported by the API and charging starts again.

**Logs and more context**
Kia eNiro 2021
evcc 0.119.4

According to @andig 's help here (THANK YOU! :+1: ) https://evccgroup.slack.com/archives/C01321PUJAD/p1693674253027289 I have configured DEBUG for the loadpoint and TRACE for the vehicle.
``` 
log: info
levels:
  cache: error
  lp-1: debug
  kia: trace
```

Interestingly enough, **_while_** it is charging, the SoC is reported correctly, the API call to 
`/api/v1/spa/vehicles/8b90ce<redacted>e1b/status/latest` 
and 
`/api/v1/spa/vehicles/8b90ce<redacted>e1b/status` 

both return the correct value in the `evstatus` structure: `"evStatus":{"batteryCharge":true,"batteryStatus":80,.....`

however, after the charging is stopped 
`Sep 05 18:24:09 evcc evcc[604]: [lp-1  ] INFO 2023/09/05 18:24:09 stop charging <- ` 

there is one last call to the `/api/v1/spa/vehicles/8b90ce<redacted>e1b/status/latest` API endpoint, which returns
`"evStatus":{"batteryCharge":false,"batteryStatus":73,....`

Obviously a Kia bug... weirdly, the "range" is reported correctly:
Shortly before the charging stops (with 80% SoC reported): `{"rangeByFuel":{"evModeRange":{"value":319,`
After charging stops, now with 73% SoC reported: `{"rangeByFuel":{"evModeRange":{"value":319,`

Logs:
```
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 charge power: 3365W
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 charge currents: [4.82 4.99 4.82]A
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 detected active phases: 3p
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 charger status: C
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 soc estimated: 80.16% (vehicle: 79.00%)
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 vehicle soc: 80%
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 vehicle soc limit: 100%
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 vehicle range: 314km
Sep 05 18:23:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:09 pv charge current: 1.83A = 6A + -4.17A (2875W @ 3p)
Sep 05 18:23:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:24 charge power: 3340W
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 charge currents: [4.82 4.92 4.78]A
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 detected active phases: 3p
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 charger status: C
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 soc estimated: 80.18% (vehicle: 79.00%)
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 vehicle soc: 80%
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 vehicle soc limit: 100%
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 vehicle range: 314km
Sep 05 18:23:25 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:25 pv charge current: 1.86A = 6A + -4.14A (2858W @ 3p)
Sep 05 18:23:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:23:39 charge power: 3328W
[... 33 lines cut ...]
Sep 05 18:24:10 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:10 vehicle soc: 73%
Sep 05 18:24:10 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:10 vehicle soc limit: 100%
Sep 05 18:24:10 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:10 vehicle range: 319km
Sep 05 18:24:10 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:10 pv charge current: 0.3A = 0A + 0.3A (-207W @ 3p)
Sep 05 18:24:10 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:10 pv enable timer reset
Sep 05 18:24:10 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:10 pv timer inactive
Sep 05 18:24:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:24 charge power: 12W
Sep 05 18:24:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:24 charge currents: [0.05 0 0]A
Sep 05 18:24:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:24 charger status: B
Sep 05 18:24:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:24 pv charge current: 0.345A = 0A + 0.345A (-238W @ 3p)
Sep 05 18:24:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:39 charge power: 12W
Sep 05 18:24:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:39 charge currents: [0.05 0 0]A
Sep 05 18:24:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:39 charger status: B
Sep 05 18:24:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:39 pv charge current: 0.341A = 0A + 0.341A (-235W @ 3p)
Sep 05 18:24:54 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:54 charge power: 12W
Sep 05 18:24:54 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:54 charge currents: [0.05 0 0]A
Sep 05 18:24:54 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:54 charger status: B
Sep 05 18:24:54 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:24:54 pv charge current: 0.348A = 0A + 0.348A (-240W @ 3p)
Sep 05 18:25:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:25:09 charge power: 12W
Sep 05 18:25:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:25:09 charge currents: [0.05 0 0]A
Sep 05 18:25:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:25:09 charger status: B
Sep 05 18:25:09 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:25:09 pv charge current: 0.329A = 0A + 0.329A (-227W @ 3p)
Sep 05 18:25:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/05 18:25:24 charge power: 12W

```

andig, 2023-09-05:
Was wäre denn die Erwartung an den Workaround und wie würde der für andere Fahrzeuge funktionieren? Was spricht dagegen, Kia über den sicher auch in der App sichtbaren Bug zu einer Lösung zu bewegen?

markusjnagel, 2023-09-06:
Hi @andig !
In der Kia App wird es korrekt angezeigt (App und Auto zeigen dasselbe), also nutzt die eine andere API oder dieselbe und macht etwas anders. Aber ich bin bei Dir - besser wäre es, das zugrundeliegende Problem (root cause) anzugehen.

Auf Basis welcher APIDoc habt Ihr das implementiert? Wenn Ihr mir die Quelle gebt und wo im evcc repo das gebaut ist, kümmere ich mich darum (dann brauche ich nicht suchen - bin kein guter Go-Programmierer, aber Code lesen krieg ich hin)
Würde dann ggf den Kontakt (wahrscheinlich sowas wie "thankyouforyouremailbutnobodywillreadit@kia.com" ) anschreiben, in der Hoffnung, dass die den Bug fixen oder sagen, was sie anders machen. Ggf fehlt ein POST auf "/aktualisiere-jetzt" oder sowas.

-------------
Der Vollständigkeit halber noch die Antwort auf Deine erste Frage: 

> Was wäre denn die Erwartung an den Workaround und wie würde der für andere Fahrzeuge funktionieren?

Die Idee wäre, entweder 
1) die letzte erfolgreiche Abfrage während des Ladens zu nehmen oder
2) erst nochmal während des Ladens den SoC abzufragen und erst dann das Laden zu beenden.

Aber das wäre natürlich nur bei der KIA API notwendig. Da Deine Frage endet mit "_und wie würde der für andere Fahrzeuge funktionieren_" nehme ich an, dass die Reihenfolge (was passiert wann) für alle Fahrzeuge gleich ist und die Kia-Template Implementierung nur das Kommando annimmt "jetzt ist das Laden beendet, hole nochmal den letzten Ladestand ab". 
Wenn das so oder so ähnlich ist, geht das natürlich nicht, weil ja nur im Falle Kia die letzte Abfrage einen fehlerhaften Wert zurückliefert - das soll ja keinen Einfluss auf die Funktion für andere Fahrzeuge haben.

andig, 2023-09-06:
> Auf Basis welcher APIDoc habt Ihr das implementiert?

Wie bei (fast) allen Herstellern- es gibt keine, also best effort.

> Wenn das so oder so ähnlich ist, geht das natürlich nicht, weil ja nur im Falle Kia die letzte Abfrage einen fehlerhaften Wert zurückliefert - das soll ja keinen Einfluss auf die Funktion für andere Fahrzeuge haben.

Das fürchte ich auch- wäre in dem Sinne dann wontfix :(

closed 2023-09-06 by markusjnagel

MarkusGH, 2023-09-07:
Die Lösung hier läge doch auf der Hand: 
```
if evStatus.batteryCharge {
   chargePerRange = evStatus.batteryCharge/evStatus.evModeRange
   Soc = evStatus.batteryCharge
} else {
   Soc = chargePerRange * evStatus.batteryCharge
}
```

pjohde, 2023-09-07:
Ich habe genau das selbe Problem. Allerdings mit ELVI EVBox + Hyundai Blue Link App. 

Nutzt Kia die gleiche APP?

Alles ist ok solange er lädt.
Ich habe in der Hyundai app und auch in EVCC 100% als Ladeziel eingestellt.
Nun lädt er...87%...98%...99% 
dann spring er auf 89% mit Ladebereit. Warte auf Fahrzeug … und bleibt da. In der Hyundai App steht der Ladezustand auf 100%. Ich kann das nächste mal gerne mittracen, aber gerade ist er voll geladen :-)

Ich kann leider nicht sagen, ob das schonmal anders war, da ich evcc erst seit ein paar Tagen nutze.

MarkusGH, 2023-09-07:
Andere Alternative: API Antworten bei denen der Soc trotz angestecktem Fahrzeug schnell fällt ignorieren.

markusjnagel, 2023-09-07:
Also - wir sind uns ja einig, dass es ein Defizit der Kia API ist.... interessanterweise hat es gestern funktioniert (also evcc ja sowieso :wink: aber auch die API lieferte nach Ladestopp durch evcc den korrekten Wert).
Bevor sich hier irgendeiner die Mühe macht, um die Defizite herumzuprogrammieren.... ich beobachte das mal weiter und werde hier updaten. Solange lasse ich einfach mal das `Vehicle` auf `TRACE` 

Hier ein Logauszug von gestern: Hausbatterie hatte 50% erreicht, evcc hat erfolgreich angehalten und der Call auf die API lieferte den korrekten Ladestand zurück: `"evStatus":{"batteryCharge":false,"batteryStatus":95,...`

```
ep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 charge power: 3450W
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 charge currents: [4.96 5.05 4.99]A
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 detected active phases: 3p
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 charger status: C
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 soc estimated: 95.46% (vehicle: 95.00%)
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 vehicle soc: 95%
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 vehicle soc limit: 100%
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 vehicle range: 382km
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 pv charge current: 2.83A = 6A + -3.17A (2190W @ 3p)
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 site power 2190W >= 0W disable threshold
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 pv disable in 0s
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 pv disable timer elapsed
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 charger disable
Sep 06 16:02:24 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:24 wake-up timer: stop
Sep 06 16:02:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:39 charge power: 12W
Sep 06 16:02:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:39 charge currents: [0.05 0 0]A
Sep 06 16:02:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:39 charger status: B
Sep 06 16:02:39 evcc evcc[604]: [lp-1  ] INFO 2023/09/06 16:02:39 stop charging <-
Sep 06 16:02:39 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:39 pv timer reset
Sep 06 16:02:40 evcc evcc[604]: [lp-1  ] DEBUG 2023/09/06 16:02:39 pv timer inactive
Sep 06 16:02:40 evcc evcc[604]: [kia   ] TRACE 2023/09/06 16:02:40 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/8b90ce<redacted>e1b/status/latest
Sep 06 16:02:40 evcc evcc[604]: [kia   ] TRACE 2023/09/06 16:02:40 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":50.07475,"lon":8.217389,"alt":-3,"type":0},"head":123,"speed":{"value":0,"unit":1},"accuracy":{"hdop":10,"pdop":10},"time":"20230906150234"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":95,"batteryPlugin":2,"remainTime2":{"
```

@pjohde - Da Kia und Hyundai ja als Konzern irgendwie zusammengehören (die neuen EVs basieren ja auf der gemeinsamen "Konzernplattform", so wie bei VW, Skoda, Seat/Cupra auch) könnte ich mir vorstellen, dass die eine gemeinsame technische Infrastruktur dahinter haben. 
Wenn Du das mal mitschneiden willst - siehe ganz oben wie das Log einzustellen ist. Loadpoint auf Debug und Fahrzeug (das Vehicle-Template das Du nutzt) auf Trace. 
https://docs.evcc.io/docs/reference/configuration/log#levels 
Das müsste bei Dir dann `hyundai` sein: https://docs.evcc.io/docs/devices/vehicles#hyundai-bluelink

pjohde, 2023-09-07:
@markusjnagel Ja, danke, das nächste Mal schau ich mir das ganze Verhalten nochmal genauer in den traces an.
Ich denke auch, dass die beiden Apps zumindest sehr ähnlich sind. Ich glaube mich zu erinnern, als ich mal die blue link Daten in Homeassistant integriert habe, dass die KIA EVO (glaube ich so heißt die) und die BlueLink App gleich über api  anzusprechen waren.
Halte mich gerne hier auf dem Laufenden, falls Du was rausfindest.

andig, 2023-09-08:
> Die Lösung hier läge doch auf der Hand:

Kann ich leider nicht nachvollziehen. Woran machst Du fest dass diese Rechnung richtig ist?

markusjnagel, 2023-09-08:
Nur zur Info - wieder ein Ladevorgang, bei dem die API funktioniert hat...
```
"evStatus":{"batteryCharge":false,"batteryStatus":94,"batteryPlugin":2
```
Heute mal ein Ladevorgang, den evcc nicht selbständig (bei Erreichen einer Grenze), sondern ich über das evcc UI beendet habe (mit Klick auf "off"). 
Aber das "wie" ist für die Fragestellung nicht entscheidend, denke ich - evcc loggt im Loadpoint, das es das Laden beendet, der Auslöser ist vermutlich für das Verhalten des `Vehicle` Code irrelevant.

Scheint so, als ob es jetzt bei Kia (wieder?) funktioniert... einige wenige Ladevorgänge sind natürlich keine Basis für eine echte Aussage, aber ich beobachte mal weiter, mache aber hier nur ein Update, falls das Problem wieder auftauchen sollte...

```
Sep 08 15:46:58 evcc evcc[726]: [lp-1  ] INFO 2023/09/08 15:46:58 stop charging <-
Sep 08 15:46:58 evcc evcc[726]: [kia   ] TRACE 2023/09/08 15:46:58 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/8b90ce<redacted>e1b/status/latest
Sep 08 15:46:58 evcc evcc[726]: [kia   ] TRACE 2023/09/08 15:46:58 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":50.074769,"lon":8.217392,"alt":4,"type":0},"head":124,"speed":{"value":0,"unit":1},"accuracy":{"hdop":10,"pdop":10},"time":"20230908093552"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":false,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":94,"batteryPlugin":2
```

markusjnagel, 2023-09-10:
Sooooooo, sehr spannend. Heute hat sich Kia :hankey: entschlossen, wieder falsche Werte durchzugeben...

Situation: Lädt, korrekte  Werte, evcc :heart: pausiert wie gewollt, kurz vorher noch den richtigen Wert abgeholt:
Timestamp: `2023/09/10 14:29:39`
`{"batteryCharge":true,"batteryStatus":91`. 
Nach Stop (Timestamp: `2023/09/10 14:29:54`) kommt wieder falscher Wert zurück. 
Timestamp `2023/09/10 14:29:54` 
`"batteryCharge":false,"batteryStatus":87`

Da kann das beste System nix machen. Wie man in der IT sagt: Das Scheiße-rein-Scheiße-raus-Prinzip :wink: 

```
Sep 10 14:29:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:29:39 charge power: 3436W
Sep 10 14:29:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:29:39 charge currents: [5.03 5.03 4.88]A
Sep 10 14:29:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:29:39 detected active phases: 3p
Sep 10 14:29:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:29:39 charger status: C
Sep 10 14:29:39 evcc evcc[1119]: [kia   ] TRACE 2023/09/10 14:29:39 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/8b90ce<redacted>e1b/status/latest
Sep 10 14:29:39 evcc evcc[1119]: [kia   ] TRACE 2023/09/10 14:29:39 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":50.074761,"lon":8.217417,"alt":0,"type":0},"head":118,"speed":{"value":0,"unit":1},"accuracy":{"hdop":10,"pdop":10},"time":"20230910094140"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":true,"batteryStatus":90,"batteryPlugin":2,"remainTime2":{"etc1":{"value":35,"unit":1},"etc2":{"value":215,"unit":1},"etc3":{"value":55,"unit":1},"atc":{"value":145,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":352,"unit":1},"totalAvailableRange":{"value":352,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[1],"time":{"time":"0600","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"0CH","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"0CH","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":9,"time":{"time":"1200","timeSection":0}},"end":{"day":9,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":390,"unit":1},"totalAvailableRange":{"value":390,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":390,"unit":1},"totalAvailableRange":{"value":390,"unit":1}},"type":2},"plugType":1}]}},"ign3":true,"hoodOpen":false,"transCond":true,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":73,"batState":0},"sleepModeCheck":false,"time":"20230910131429","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"03:36:07"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":35424.8,"unit":1}}},"msgId":"9a8aafac-2760-4dd4-aabd-2b5dc2b2f5e7"}
Sep 10 14:29:39 evcc evcc[1119]: [kia   ] TRACE 2023/09/10 14:29:39 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/8b90ce<redacted>e1b/status
Sep 10 14:29:44 evcc evcc[1119]: [kia   ] TRACE 2023/09/10 14:29:44 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":true,"batteryStatus":91,"batteryPlugin":2,"remainTime2":{"etc1":{"value":32,"unit":1},"etc2":{"value":190,"unit":1},"etc3":{"value":50,"unit":1},"atc":{"value":130,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":356,"unit":1},"totalAvailableRange":{"value":356,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[1],"time":{"time":"0600","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"0CH","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"0CH","unit":0,"hvacTempType":1},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":9,"time":{"time":"1200","timeSection":0}},"end":{"day":9,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":390,"unit":1},"totalAvailableRange":{"value":390,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value" [... cut ...]
Sep 10 14:30:24 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:30:24 pv charge current: 3.11A = 0A + 3.11A (-2149W @ 3p)
Sep 10 14:30:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:30:39 charge power: 18W
Sep 10 14:30:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:30:39 charge currents: [0.05 0.03 0]A
Sep 10 14:30:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:30:39 charger status: B
Sep 10 14:30:39 evcc evcc[1119]: [lp-1  ] DEBUG 2023/09/10 14:30:39 pv charge current: 3.04A = 0A + 3.04A (-2097W @ 3p)

```