evcc-io/evcc #9748: Error: Connection to charger not valid

stadlaa, 2023-09-05:
### Describe the bug

Suddenly there was an Error, saying: connection to charger not valid! It came every 2-4sec and was not always exactly the same message but with simmilar meaning (Ithink it was sth. like: charge meter not valid!).

The Charger (Fronius WattPilot) was perfectly accessable through the Fronius App. After a restart of evcc everything worked perfectly fine.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
[... 230 lines cut ...]
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
no error anymore
```

### What type of operating system are you running?

Windows

### Version

evcc version 0.119.0

TheFreaker86, 2023-09-05:
You published your username, Password and VINs for your Renault account. You should change your password **right now (!!!)** whereever this password is used and cleanse this post here for any private data

andig, 2023-09-05:
/cc @mabunixda

klobin925, 2023-09-05:
Ich denke gleiches wie #7601

stadlaa, 2023-09-06:
> You published your username, Password and VINs for your Renault account. You should change your password **right now (!!!)** whereever this password is used and cleanse your post here for any private data immediately!

Thank you very much! How stupid of me!

andig, 2023-09-09:
Duplikat zu https://github.com/evcc-io/evcc/issues/9806

closed 2023-09-09 by andig