evcc-io/evcc #9750: Daten zum aktuellen Ladevorgang sind sehr versteckt

klobin925, 2023-09-06:
Hi,
Mir ist gerade zum ersten mal aufgefallen, dass man auf %Sonne anzeige tippen kann, um weitere Werte anzuzeigen. Ich habe mir sowas vorgestellt um ein Hinweis auf die weiteren Werte zu geben.

Aktuell
[image]
Mein Vorschlag
[image]

naltatis, 2023-09-06:
Ja, guter Punkt. Mir ist bislang noch kein schöner dezenter Hinweis eingefallen. Die Idee mit den Punkte finde ich gut, beißt sicher aber leider mit dem Loadpoint-Wechsel auf kleinen Screen:

[image]

Matze03, 2023-09-06:
Das finde ich nicht, im Gegenteil, die Punkte kennt jeder und wären für jeden das Kennzeichen, "hier gibt's noch was"...

klobin925, 2023-09-06:
Vielleicht bin ich ein solches UI nicht gewohnt, in dem man überall draufdrücken kann, um weiter Dinge zu sehen und einzustellen. 

Auch die Baterieeinstellungen hinter der Prozentzahl und das ändern der Schwellwerte wieder hinter den Prozenten finde ich schwierig (ich weiß hier soll der Unterstrich darauf hinweisen. Hat bei mir aber nicht gut funktioniert)
Sowas hätte ich eher in den Einstellungen erwartet.

wiggal, 2023-09-06:
Unterstreichen der "99% Sonnenenergie" ?

naltatis, 2023-09-06:
> Sowas hätte ich eher in den Einstellungen erwartet.

Ja, das kann durchaus auch noch kommen. Am Einstellungsbereich wird gearbeitet.

> Das finde ich nicht, im Gegenteil, die Punkte kennt jede

Ok, das meinte ich nicht. Ich halt die Idee für gut, wir bekommen nur technische Problemen, wenn ich in einem Bereich, der wischbar ist (Ladepunkte) noch einen Bereich einführe, der wischbar ist (Werte im Ladepunkt). Und den Werte-Bereich nicht wischbar machen, sondern beim Klicken zu bleiben, bricht mit der Punktemetapher, da sie dann nicht überall das Gleiche bedeutet.

maf-soft, 2023-09-06:
Habe ich ebenfalls nicht gewusst (obwohl ich das am PC nutze und durch den Mauspfeil sogar schon mehr Info habe als am Touchscreen) und es wäre gut wenn man es irgendwie erkennen könnte. Aber eine gute Idee habe ich auch nicht. Fachmann fragen? :) Bin gespannt

maf-soft, 2023-09-06:
Da fällt mir auf, über dem schon unterstrichenen 100% rechts unten ist der Mauspfeil falsch.

Wenn gerade geladen wird, kann man die bisherige Ladezeit nicht anzeigen, sondern nur die Restzeit.

Gehört hier zwar nicht hin, aber naja...

marioschaper, 2023-09-07:
[image]
wie wäre es mit einem "Touch/Tap" Symbol unterhalb der tapbaren anzeige (wegen mir auch farblich nicht so offensichtlich)

Matze03, 2023-09-07:
> wie wäre es mit einem "Touch/Tap" Symbol unterhalb der tapbaren anzeige (wegen mir auch farblich nicht so offensichtlich)

Farbgebung in schwarz / weiß, ich finde ich den Vorschlag gut!

PR #9783 (2023-09-09): Improve session info discoverability
fixes #9750

- add underline to indicate interaction
- open dropdown by clicking on the label
  - show all options, better overview
  - same pattern as in session table (table headers)
- clicking on the value still toggles to next option (quick change)
  - disable unwanted browser text select when clicking faster
- add underline to footer links
- "duration" and "remaining" now don't exclude each other
- some fields are only shown if data exists (tariff.grid, tariff.co2, ...)

[image]
[image]

video of toggling by click or select

https://github.com/evcc-io/ [... cut]
closed 2023-09-09 by andig