evcc-io/evcc #9752: Senec Home.V3 Hybrid Duo - connection refused

TecTim18, 2023-09-06:
### Describe the bug

Hallo zusammen,

seit der Umstellung auf Version 0.119.3 und nun auch 0.119.4. kann der Senec Hybrid Inverter samt seines Meters und Batterie nicht mehr abgefragt werden. Mit der Version 0.119.2 besteht dieses Problem nicht.
Nun erhalte ich einen "Connection refused".

[image]

### Steps to reproduce

Update auf 0.119.2 oder höher

### Configuration details

```yaml
network:
  host: evcc.local
  port: 7070
interval: 3s # control cycle interval

sponsortoken: XXX
telemetry: true
plant: XXX
log: debug

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: "XXX"

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

meters:
  - name: grid_meter
    type: template
    template: senec-home
    usage: grid
    host: 192.168.20.12
  - name: senec_pv_meter
    type: template
    template: senec-home
    usage: pv
    host: 192.168.20.12
  - name: battery
    type: template
    template: senec-home
    usage: battery
    host: 192.168.20.12
  - name: pv-garage
    type: template
    template: shelly-1pm
    usage: pv
    host: 192.168.20.35 # IP-Adresse oder Hostname
    channel: 0 # Optional

chargers:
  - name: go-e_V3
    type: template
    template: go-e-v3
    host: 192.168.20.13
  - name: go-e_gemini
    type: template
    template: go-e-v3
    host: 192.168.20.14

loadpoints:
  - title: Garage
    charger: go-e_V3
    mode: pv # Standard Modus des Ladepunktes
    phases: 0 # Phasenumschaltung mit 0 aktiviert
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true # Rueksetzung auf Standard-Modus des Ladepunktes nach beenden des Ladevorgangs
    enable:
      delay: 2m # Dauer ueber die ausreichend Ueberschuss vorhanden sein muss um Ladung zu starten
    disable:
      delay: 4m # Dauer uber die auch ohne genuegend Ueberschuss weiter geladen wird#    priority: 2

  - title: Carport
    charger: go-e_gemini
    mode: pv # Standard Modus des Ladepunktes
    phases: 0
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true # Rueksetzung auf Standard-Modus des Ladepunktes nach beenden des Ladevorgangs
    enable:
      delay: 2m # Dauer ueber die ausreichend Ueberschuss vorhanden sein muss um Ladung zu starten
    disable:
      delay: 4m # Dauer ueber die auch ohne genuegend Ueberschuss weiter geladen wird
    priority: 1

vehicles:
  - name: Twingo
    type: template
    template: renault
    title: Renault Twingo
    user: XXX
    password: XXX
    capacity: 21.4 # Akku-Kapazität in kWh
    mincurrent: 8
    targetSoc: 80

  - name: Hyundai Kona
    type: template
    template: offline
    title: Hyundai Kona # Wird in der Benutzeroberfläche angezeigt # Optional
    capacity: 64 # Akkukapazität in kWh # Optional

site:
  title: Energie Management Fremuth
  meters:
    grid: grid_meter
    pv:
      - senec_pv_meter
      - pv-garage
    batteries: battery
  smartCostLimit: 0.16 # Wechselt im PV Modus auf schnelles laden sollange Preis unterschritten ist

influx:
  url: http://192.168.20.10:8086
  database: evcc
  user: XXX
  password: XXX
```

### Log details

```text
`[site  ] DEBUG 2023/09/06 09:43:23 ----
[lp-1  ] DEBUG 2023/09/06 09:43:23 charge power: 0W
[lp-2  ] DEBUG 2023/09/06 09:43:23 charge power: 0W
[site  ] ERROR 2023/09/06 09:43:24 pv 1 power: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] DEBUG 2023/09/06 09:43:24 pv power: 68W
[site  ] ERROR 2023/09/06 09:43:24 battery 1 power: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] ERROR 2023/09/06 09:43:24 battery 1 soc: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] DEBUG 2023/09/06 09:43:24 battery soc: 0%
[site  ] DEBUG 2023/09/06 09:43:24 battery power: 0W
[site  ] ERROR 2023/09/06 09:43:25 grid meter: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] ERROR 2023/09/06 09:43:25 grid meter: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] DEBUG 2023/09/06 09:43:26 ----
[lp-1  ] DEBUG 2023/09/06 09:43:26 charge power: 0W
[lp-2  ] DEBUG 2023/09/06 09:43:26 charge power: 0W
[site  ] ERROR 2023/09/06 09:43:27 pv 1 power: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] DEBUG 2023/09/06 09:43:27 pv power: 75W
[site  ] ERROR 2023/09/06 09:43:27 battery 1 power: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] ERROR 2023/09/06 09:43:27 battery 1 soc: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] DEBUG 2023/09/06 09:43:27 battery soc: 0%
[site  ] DEBUG 2023/09/06 09:43:27 battery power: 0W
[site  ] ERROR 2023/09/06 09:43:28 grid meter: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused
[site  ] ERROR 2023/09/06 09:43:28 grid meter: Post "https://192.168.20.12/lala.cgi": dial tcp 192.168.20.12:443: connect: connection refused`
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.119.4

TecTim18, 2023-09-06:
Habe soeben die Version 0.119.2 wiederhergestellt. Damit läuft nun wieder alles reibungslos:

[image]

`[site  ] DEBUG 2023/09/06 10:05:05 ----
[lp-1  ] DEBUG 2023/09/06 10:05:05 charge power: 0W
[lp-2  ] DEBUG 2023/09/06 10:05:06 charge power: 0W
[site  ] DEBUG 2023/09/06 10:05:06 pv power: 3939W
[site  ] DEBUG 2023/09/06 10:05:06 battery soc: 48%
[site  ] DEBUG 2023/09/06 10:05:06 battery power: -2486W
[site  ] DEBUG 2023/09/06 10:05:06 grid power: -350W
[site  ] DEBUG 2023/09/06 10:05:06 giving priority to battery charging at soc: 48%
[site  ] DEBUG 2023/09/06 10:05:06 site power: -350W
[lp-2  ] DEBUG 2023/09/06 10:05:06 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/09/06 10:05:06 charge total import: 3.231kWh
[lp-2  ] DEBUG 2023/09/06 10:05:06 charger status: A
[site  ] DEBUG 2023/09/06 10:05:08 ----
[lp-1  ] DEBUG 2023/09/06 10:05:08 charge power: 0W
[lp-2  ] DEBUG 2023/09/06 10:05:08 charge power: 0W
[site  ] DEBUG 2023/09/06 10:05:09 pv power: 3949W
[site  ] DEBUG 2023/09/06 10:05:09 battery soc: 48%
[site  ] DEBUG 2023/09/06 10:05:09 battery power: -2484W
[site  ] DEBUG 2023/09/06 10:05:09 grid power: -356W
[site  ] DEBUG 2023/09/06 10:05:09 giving priority to battery charging at soc: 48%
[site  ] DEBUG 2023/09/06 10:05:09 site power: -356W
[lp-1  ] DEBUG 2023/09/06 10:05:09 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/06 10:05:09 charge total import: 3862.665kWh
[lp-1  ] DEBUG 2023/09/06 10:05:09 charger status: A`

klobin925, 2023-09-06:
Die SENEC Speicher haben eine neue Software erhalten und die Kommunikation wurde auf https geändert. Es scheint also, dass du das Update noch nicht erhalten hast und deshalb die neuen Versionen von evcc nicht funktionieren.

TecTim18, 2023-09-06:
Noch besser 👍 
Hatte die Option nur für die globale Anwendung auf dem Schirm. 

Funktioniert ✅  Danke