evcc-io/evcc #9763: Feature Request: Max Grid Use option

dpaeme, 2023-09-06:
Hi EVCC team,

**Currently, EVCC has four options in the charging menu:**
- "**Off**" - no charging whatsoever
- "**Solar**" - only use excess solar energy
- "**Min+Solar**" - use excess solar energy, but keep charging at the minimum of the vehicle's capacity if the amount of solar energy dips below the minimum rate
- "**Fast**" - charge at the fastest energy rate the vehicle can ingest

**Can I propose a fifth charging option? Let's call it "Max Grid Use".** Or **MaxGrid** for short and for the sake of the descriptions in this feature request. This should be **a configurable energy value which should be the maximum rate of Grid energy use allowed while charging**. 

**The use case:** 
In Belgium, and certainly in other in other countries too, we aren't only charged for the energy we consume (on-peak or off-peak), but we are also _taxed_ a rate based on the peak energy use in a certain month. In practice, this has an ugly impact on your energy bill during winter. 

For example: if you charge at 11kW for a concurrent 15mins, the system will place your _peak_ for that month on 11kW, and you will be taxed an extra percentage on top for *all* energy consumed during that month, not only the energy consumed during the peak. It's also a progressive tax, the higher you peak, the more you're taxed. And we're talking several hundreds of euros over the course of a year.

I think you can understand why it would be interesting to be able to avoid excess peak energy use when charging using EVCC.

Some practical examples of this setting:
One:
- MaxGrid is set at 6kW
- House energy use is 0,3W
- No Solar coming in
--> Car charges at 5,7kW (total staying at the configured allowed 6kW grid peak)

Two:
- Maxgrid is set at 6kW
- House energy use is 0,3kW
- Solar is on it's peak, delivering 10kW
--> Car charges at 9,7kW (all excess solar can be dumped into the car), there is no grid use.

Three
- Maxgrid is set at 6kW
- House energy use is 2,3kW (the heatpump started working)
- Solar is only delivering 1kW (it's winter)
--> Car charges at 4,7kW

I'm sure that this would help a lot in keeping charging costs down during winter, when there's not too much Solar coming in.

Would this be something that can be considered by the EVCC team?

_(and as icing on the proverbial cake: having this MaxGrid setting be be configurable to only charge during off-peak times would be so so perfect!)_

Thanks!!
Much appreciated,

David

andig, 2023-09-06:
/cc @premultiply

VolkerK62, 2023-09-06:
Due to the described use cases, it should work

https://github.com/evcc-io/evcc/pull/8427#issue-1752373393
use cases
Lastbegrenzung: bestimmte Leistung in einem Stromkreis nicht überschreiten
zB alle Wallboxen zusammen max 11kW
**zB max Leistung am Netzpunkt <= 5kW**

andig, 2023-09-06:
Aber wie würdest du das steuern? Als Sofortladen?

andig, 2023-09-06:
No, it's not but will be at some point. Seems for now, we'd refer to that and not add an additional mode.  Another option is also adjusting max current using UI or API and switching to immediate charging.

Given workarounds exist I'm closing as wontfix.

closed 2023-09-06 by andig

dpaeme, 2023-09-15:
Hi @andig, it sounds like a good solution, but it looks like I ran into an issue:

I've been experimenting with the maxCurrent setting, through the API, and for some reason evcc isn't accepting/using the value.

If I set `% curl -X POST https://evccserver/api/loadpoints/1/maxcurrent/9`
The result looks good: `{"result":9}`

I then set the mode to "now": `% curl -X POST https://evccserver/api/loadpoints/1/mode/now`
Which also looks good: `{"result":"now"}`

But whatever happens, my charger keeps going at full tilt. 
If I query the state: ` % curl -X GET https://evccserver/api/state`

I always get the same result:
```
...
"enabled": true,
"guardAction": "inactive",
"guardRemaining": 0,
"maxCurrent": 32,
"minCurrent": 2.2,
"minSoc": 0,
"mode": "now",
"phaseAction": "inactive",
"phaseRemaining":0,
"phasesActive": 3,
...
```

The same when I try to set the maxcurrent with the mode setting on `pv`, it keeps reverting to 32A after a short while.

(ps: I'm on v0.119.5, the standard evcc docker image)

andig, 2023-09-15:
Wrong order of calls?

dpaeme, 2023-09-15:
I don’t think so, even calling maxcurrent when not changing the charging mode makes it revert to the max setting after a couple of seconds.

andig, 2023-09-15:
Yeah, thats expected. I think you could obly limit the max current of the loadpoint for your case.

dpaeme, 2023-09-15:
Strange, it’s probably me, but I don’t understand the app’s behaviour here. 

Wouldn’t it be expected that it accepts a value and not revert to anything else unless you make another call or state change?

(And limiting the setting in the config doesn’t make sense in the makeshift load management case: you’d want your non-pv charging to have a lower maxcurrent than the pv charging maxcurrent, so you’d still need a stable but dynamic system)

andig, 2023-09-15:
Actually: not sure. But thats a new issue!

dpaeme, 2023-09-15:
Thanks for looking into it :-)

If you need any help or logging or whatnot, don’t hesitat to give me a shout!

andig, 2023-09-15:
Please start a new issue 👍🏻

dpaeme, 2023-09-17:
Done: https://github.com/evcc-io/evcc/issues/9942