evcc-io/evcc #9776: Minimal charge feature treats unknown SOC as 0% after evcc start

maf-soft, 2023-09-07:
### Describe the bug

I restarted evcc and it started charging with status minimal charge 5%.

I used the new setting for the first time.

After a restart my Tesla Model 3 never shows the SOC and I guess this is normal? It is shown when charging starts or the car is plugged in. So it seems the new feature thinks it is 0% and starts charging 3p. Then the SOC becomes known and it is stopped again after the delay.

### Steps to reproduce

Plan / Ankunft / Min. Ladung 5%
Kein Überschuss
evcc neustarten

### Configuration details

```yaml
Tesla Model 3, Easee, Discovergy
```

### Log details

```text
Ich nehme an, der Fall ist auch so klar, sonst liefere ich es nach
```

### What type of operating system are you running?

Windows

### Version

0.119.4

StefanSchoof, 2023-09-07:
Andere Lösung: In Doku deutlich machen das man besten evcc nicht neu starten, wenn gerade kein Auto angeschlossen ist. Ein Neustart macht soviel Egde Cases, die eine sonder Behandlung benötigen.

andig, 2023-09-07:
Sehe ich auch so- und es bleibt immer der Fall, dass das Fahrzeug nicht erreichbar ist. Dann muss "im Zweifel für den Angeklagten" dennoch geladen werden!

closed 2023-09-07 by andig

maf-soft, 2023-09-07:
Das erschließt sich mir überhaupt nicht. Dass es jetzt wichtigeres gibt, ja, natürlich. Ich wollte diesen Bug nur dokumentieren.

Aber Neustarts kommen immer wieder mal vor, z.B. bei einem Update, mache haben das auch automatisiert, oder wenn man gerade wegen irgendwelcher Einstellungen experimentiert - und dann jedes Mal 2 Minuten mit 3p zu laden, nur weil das Auto ja vielleicht wirklich plötzlich leer sein könnte, ist doch kein Argument? Im Normalfall läuft evcc ja gerade, wenn man sein leeres Auto ansteckt, und dann wird der SOC korrekt ermittelt und das neue Feature würde seinen Dienst tun. Einen nicht ermittelten SOC sicherheitshalber wie 0% zu behandeln, finde ich einfach falsch, es ist so extrem unwahrscheinlich, dass das jemals so besser wäre.