evcc-io/evcc #9787: Lademenge startet nicht bei 0 kWh

DocLifeguard, 2023-09-08:
### Describe the bug

Ich benutze evcc mit einer EON vBox (Vestel) und einer Solaredge PV-Anlage. Rein technisch funktioniert das Laden/Überschlussladen wunderbar. Jedoch beginnt die Anzeige der geladenen Energie nicht bei 0 sondern immer bei einem Wert zwischen 1930 und 1937 kWh. Die Anzeige springt sofort auf den hohen Wert, sobald der Ladevorgang beginnt. Vorher steht er sauber bei 0. Die Menge (also die Differenz) stimmt allerdings.

[image]

### Steps to reproduce

1. Fahrzeug an Wallbox anschließen
2. Authentifzierung an Wallbox durchführen
3. Ladevorgang beginnt, geladene Energie springt auf 1930+ kWh
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 7e83383a3ffe6b6161326bc6455deb5714271032939064649c28a8fc7ed5155f

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3ODU3NzY4MTgsImlh>

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

chargers:
- type: template
  template: vestel
  id: 255
  host: 192.168.178.83
  port: 502
  modbus: tcpip
  name: wallbox1

vehicles:
- type: template
  template: audi
  title: Q4 etron
  user: XXX
  password: XXX
  vin: XXX
  capacity: 82
  name: ev1

loadpoints:
- title: Garage
  charger: wallbox1
  vehicle: ev1
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

meters:
- name: my_pv
  type: template
  template: solaredge-hybrid
  usage: pv
  modbus: tcpip
  id: 1
  host: 192.168.178.116
  port: 1502
- name: my_grid
  type: template
  template: solaredge-hybrid
  usage: grid
  modbus: tcpip
  host: 192.168.178.116
  id: 1
  port: 1502

- name: my_battery
  type: template
  template: solaredge-hybrid
  usage: battery
  modbus: tcpip
  host: 192.168.178.116
  id: 1
  port: 1502
site:
  title: Zuhause
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.4

  feedin:
    type: fixed
    price: 0.08
```

### Log details

```text
[main  ] INFO 2023/09/08 08:26:30 evcc 0.119.2
[main  ] INFO 2023/09/08 08:26:30 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/08 08:26:30 starting ui and api at :7070
[db    ] INFO 2023/09/08 08:26:30 using sqlite database: /home/andreas/.evcc/evcc.db
[site  ] WARN 2023/09/08 08:26:33 battery configured but residualPower is missing (add residualPower: 100 to site)
[site  ] INFO 2023/09/08 08:26:33 site config:
[site  ] INFO 2023/09/08 08:26:33   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/08 08:26:33     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/08 08:26:33     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/08 08:26:33     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/08 08:26:33   vehicles:
[site  ] INFO 2023/09/08 08:26:33     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/09/08 08:26:33 loadpoint 1:
[lp-1  ] INFO 2023/09/08 08:26:33   mode:        minpv
[lp-1  ] INFO 2023/09/08 08:26:33   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/08 08:26:33   meters:      charge ✓
[lp-1  ] INFO 2023/09/08 08:26:33     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/09/08 08:26:33 phase timer inactive
[lp-1  ] DEBUG 2023/09/08 08:26:33 pv timer inactive
[lp-1  ] DEBUG 2023/09/08 08:26:33 guard timer inactive
[lp-1  ] INFO 2023/09/08 08:26:33 vehicle updated: unknown -> Q4 etron
[lp-1  ] DEBUG 2023/09/08 08:26:33 max charge current: 6A
[site  ] WARN 2023/09/08 08:26:33 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/08 08:26:33 ----
[lp-1  ] DEBUG 2023/09/08 08:26:34 charge power: 3467W
[site  ] DEBUG 2023/09/08 08:26:34 pv power: 2274W
[main  ] INFO 2023/09/08 08:26:34 new version available: 0.119.4
[site  ] DEBUG 2023/09/08 08:26:34 battery soc: 4%
[site  ] DEBUG 2023/09/08 08:26:34 battery power: 2348W
[site  ] DEBUG 2023/09/08 08:26:34 grid power: -21W
[site  ] DEBUG 2023/09/08 08:26:34 site power: 2327W
[lp-1  ] DEBUG 2023/09/08 08:26:37 vehicle odometer: 21620km
[lp-1  ] DEBUG 2023/09/08 08:26:37 charge voltages: [226 232 231]V
[lp-1  ] DEBUG 2023/09/08 08:26:37 detected connected phases: 3p
[lp-1  ] DEBUG 2023/09/08 08:26:37 charge currents: [13.2 0 0]A
[lp-1  ] DEBUG 2023/09/08 08:26:38 charge total import: 29.500kWh
[lp-1  ] DEBUG 2023/09/08 08:26:38 charger status: C
[lp-1  ] INFO 2023/09/08 08:26:38 car connected
[lp-1  ] DEBUG 2023/09/08 08:26:38 pv timer elapse
[lp-1  ] DEBUG 2023/09/08 08:26:38 pv timer inactive
[lp-1  ] DEBUG 2023/09/08 08:26:38 charger: guard elapse
[lp-1  ] DEBUG 2023/09/08 08:26:38 guard timer inactive
[lp-1  ] DEBUG 2023/09/08 08:26:38 charge total import: 29.500kWh
[lp-1  ] INFO 2023/09/08 08:26:38 start charging ->
[lp-1  ] DEBUG 2023/09/08 08:26:38 wake-up timer: stop
[lp-1  ] DEBUG 2023/09/08 08:26:38 vehicle status: C
[lp-1  ] DEBUG 2023/09/08 08:26:38 vehicle soc: 61%
[lp-1  ] DEBUG 2023/09/08 08:26:38 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/08 08:26:38 vehicle range: 285km
[lp-1  ] DEBUG 2023/09/08 08:26:38 pv charge current: 2.63A = 6A + -3.37A (2327W @ 3p)
[site  ] DEBUG 2023/09/08 08:26:43 ----
[lp-1  ] DEBUG 2023/09/08 08:26:44 charge power: 1368W
[site  ] DEBUG 2023/09/08 08:26:44 pv power: 2309W
[site  ] DEBUG 2023/09/08 08:26:44 battery soc: 4%
[site  ] DEBUG 2023/09/08 08:26:44 battery power: 161W
[site  ] DEBUG 2023/09/08 08:26:44 grid power: 2W
[site  ] DEBUG 2023/09/08 08:26:44 site power: 163W
[lp-1  ] DEBUG 2023/09/08 08:26:44 vehicle odometer: 21620km
[lp-1  ] DEBUG 2023/09/08 08:26:44 charge voltages: [228 231 230]V
[lp-1  ] DEBUG 2023/09/08 08:26:44 detected connected phases: 3p
[lp-1  ] DEBUG 2023/09/08 08:26:44 charge currents: [6.06 0 0]A
[lp-1  ] DEBUG 2023/09/08 08:26:44 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/08 08:26:44 charge total import: 29.500kWh
[lp-1  ] DEBUG 2023/09/08 08:26:44 charger status: C
[lp-1  ] DEBUG 2023/09/08 08:26:44 soc estimated: 61.01% (vehicle: 61.00%)
[lp-1  ] DEBUG 2023/09/08 08:26:44 vehicle soc: 61%
[lp-1  ] DEBUG 2023/09/08 08:26:44 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/08 08:26:44 vehicle range: 285km
[lp-1  ] DEBUG 2023/09/08 08:26:44 pv charge current: 5.29A = 6A + -0.709A (163W @ 1p)
[site  ] DEBUG 2023/09/08 08:26:53 ----
[lp-1  ] DEBUG 2023/09/08 08:26:54 charge power: 1367W
[site  ] DEBUG 2023/09/08 08:26:54 pv power: 2301W
[site  ] DEBUG 2023/09/08 08:26:54 battery soc: 4%
[site  ] DEBUG 2023/09/08 08:26:54 battery power: 152W
[site  ] DEBUG 2023/09/08 08:26:54 grid power: 8W
[site  ] DEBUG 2023/09/08 08:26:54 site power: 160W
[lp-1  ] DEBUG 2023/09/08 08:26:54 charge voltages: [228 231 230]V
[lp-1  ] DEBUG 2023/09/08 08:26:54 detected connected phases: 3p
[lp-1  ] DEBUG 2023/09/08 08:26:54 charge currents: [6.05 0 0]A
[lp-1  ] DEBUG 2023/09/08 08:26:54 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/08 08:26:54 charge total import: 29.500kWh
[lp-1  ] DEBUG 2023/09/08 08:26:54 charger status: C
[lp-1  ] DEBUG 2023/09/08 08:26:54 soc estimated: 61.01% (vehicle: 61.00%)
[lp-1  ] DEBUG 2023/09/08 08:26:54 vehicle soc: 61%
[lp-1  ] DEBUG 2023/09/08 08:26:54 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/08 08:26:54 vehicle range: 285km
[lp-1  ] DEBUG 2023/09/08 08:26:54 pv charge current: 5.3A = 6A + -0.696A (160W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.119.2

premultiply, 2023-09-08:
Was sagt denn `evcc charger --log trace`?

P.S.: In deiner Konfig fehlt `residualPower`.

andig, 2023-09-09:
> Was sagt denn evcc charger --log trace?

Benötigen wir immer noch um den Fehler zu sehen. Falls nicht reproduzierbar würde ich hier schließen.

DocLifeguard, 2023-09-09:
Hier die Ausgabe des tracelogs (sobald ich den Service bei laufendem Ladevorgang neu starte, verhält sich das System wie es soll)

[main  ] INFO 2023/09/09 09:58:29 evcc 0.119.5
[main  ] INFO 2023/09/09 09:58:29 using config file: /etc/evcc.yaml
[db    ] INFO 2023/09/09 09:58:29 using sqlite database: /home/andreas/.evcc/evcc.db
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT       fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT       fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%      device_id       %") -1 <nil>
[db    ] TRACE 2023/09/09 09:58:29 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[vestel] TRACE 2023/09/09 09:58:29 modbus: send 00 01 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/09/09 09:58:39 modbus: send 00 02 00 00 00 06 ff 04 04 0c 00 02
[vestel] ERROR 2023/09/09 09:58:59 heartbeat: dial tcp 192.168.178.83:502: i/o timeout
[vestel] ERROR 2023/09/09 09:59:29 heartbeat: dial tcp 192.168.178.83:502: i/o timeout
[vestel] TRACE 2023/09/09 09:59:39 modbus: closing connection due to idle timeout: 1m0.009569139s
[vestel] ERROR 2023/09/09 09:59:59 heartbeat: dial tcp 192.168.178.83:502: i/o timeout
Power:          read tcp 192.168.178.114:49144->192.168.178.83:502: i/o timeout
Energy:         read tcp 192.168.178.114:35640->192.168.178.83:502: i/o timeout
Current L1..L3: dial tcp 192.168.178.83:502: i/o timeout
Voltage L1..L3: dial tcp 192.168.178.83:502: i/o timeout
Charge status:  dial tcp 192.168.178.83:502: i/o timeout
Enabled:        dial tcp 192.168.178.83:502: i/o timeout
Charged:        dial tcp 192.168.178.83:502: i/o timeout
Duration:       dial tcp 192.168.178.83:502: i/o timeout

andig, 2023-09-09:
*ERST* evcc stoppen, dann ausführen.

DocLifeguard, 2023-09-09:
Sorry (und danke für die Geduld...)
[main  ] INFO 2023/09/09 11:13:58 evcc 0.119.5
[main  ] INFO 2023/09/09 11:13:58 using config file: /etc/evcc.yaml
[db    ] INFO 2023/09/09 11:13:59 using sqlite database: /home/andreas/.evcc/evcc.db
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT       fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT       fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%  device_id       %") -1 <nil>
[db    ] TRACE 2023/09/09 11:13:59 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 01 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 0d 7c
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 02 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 01 2a
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 03 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 03 00 00 00 05 ff 04 02 3c 5a
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 04 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 04 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 05 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 05 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 06 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 06 00 00 00 05 ff 04 02 00 e0
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 07 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 07 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 08 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e5
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 09 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 09 00 00 00 05 ff 04 02 00 03
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 0a 00 00 00 06 ff 04 03 e9 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 0b 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 0b 00 00 00 05 ff 03 02 00 20
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 0c 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 0c 00 00 00 07 ff 04 04 00 1d 9b 4d
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 0d 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 0d 00 00 00 07 ff 04 04 00 00 1a 49
Power:          3452W
Energy:         29.8kWh
Current L1..L3: 15.4A 0A 0A
Voltage L1..L3: 224V 231V 229V
Charge status:  C
Enabled:        true
Charged:        1940.3kWh
Duration:       1h52m9s

andig, 2023-09-09:
Für mich sieht das aus, als wären die Register vertauscht:

```
vestelRegTotalEnergy     = 1036
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 02 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 01 2a

vestelRegSessionEnergy   = 1502
[vestel] TRACE 2023/09/09 11:13:59 modbus: send 00 0c 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/09/09 11:13:59 modbus: recv 00 0c 00 00 00 07 ff 04 04 00 1d 9b 4d
```
Was wird bei einem neuen Ladevorgang ausgegeben?

DocLifeguard, 2023-09-10:
Hi, 
nachfolgend das Tracelog bei einem neuen Ladevorgang:

[main  ] INFO 2023/09/10 10:06:51 evcc 0.119.5
[main  ] INFO 2023/09/10 10:06:51 using config file: /etc/evcc.yaml
[db    ] INFO 2023/09/10 10:06:51 using sqlite database: /home/andreas/.evcc/evcc.db
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%       device_id       %") -1 <nil>
[db    ] TRACE 2023/09/10 10:06:51 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 01 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 0d 69
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 02 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 01 2a
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 03 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 03 00 00 00 05 ff 04 02 3c b4
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 04 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 04 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 05 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 05 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 06 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 06 00 00 00 05 ff 04 02 00 de
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 07 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 07 00 00 00 05 ff 04 02 00 e6
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 08 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 09 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 09 00 00 00 05 ff 04 02 00 03
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 0a 00 00 00 06 ff 04 03 e9 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 0b 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 0b 00 00 00 05 ff 03 02 00 20
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 0c 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 0c 00 00 00 07 ff 04 04 00 1d 95 48
[vestel] TRACE 2023/09/10 10:06:51 modbus: send 00 0d 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/09/10 10:06:51 modbus: recv 00 0d 00 00 00 07 ff 04 04 00 00 0a f7
Power:          3433W
Energy:         29.8kWh
Current L1..L3: 15.5A 0A 0A
Voltage L1..L3: 222V 230V 231V
Charge status:  C
Enabled:        true
Charged:        1938.8kWh
Duration:       46m47s

DocLifeguard, 2023-09-10:
Nachdem ich das System einmal neu gestartet habe, ist die Anzeige korrekt:
[image]

premultiply, 2023-09-10:
Welches System?

DocLifeguard, 2023-09-10:
Mit System meine den evcc Service (also mit systemctl stop evcc gestoppt und anschließend wieder gestartet. Dann ist die Anzeige korrrekt).
evcc charger:
[main  ] INFO 2023/09/10 12:00:01 evcc 0.119.5
[main  ] INFO 2023/09/10 12:00:01 using config file: /etc/evcc.yaml
[db    ] INFO 2023/09/10 12:00:01 using sqlite database: /home/andreas/.evcc/evcc.db
Power:          3185W
Energy:         29.9kWh
Current L1..L3: 14.5A 0A 0A
Voltage L1..L3: 222V 230V 230V
Charge status:  C
Enabled:        true
Charged:        1939.4kWh
Duration:       43m8s

andig, 2023-09-10:
Ich verstehe nicht, was Du uns hier sagen willst. Beide Ausgaben- vorher und hinter- zeigen unplausible, identische Werte:

```
Energy: 29.9kWh
Charged: 1939.4kWh
```

Bitte Postings formatieren, das ganze db Log braucht es nicht!

DocLifeguard, 2023-09-10:
Hey, die Ausgaben werden von evcc erzeugt, nicht von mir. Wenn die unplausibel sind, liegt das (hoffentlich) nicht an mir. Ich habe extra die Screenshots angefügt, um zu zeigen, dass ein Restart von evcc bei laufendem Ladevorgang dazu führt, dass die Anzeige nachher korrekt ist. Also scheint evcc ja irgendwoher die korrekten Daten zu bekommen bzw. Irgendwas scheint ja der Restart zu bewirken.

andig, 2023-09-10:
Ich frag jetzt nochmal ganz schwarz/weiss: lässt sich der Fehler irgendwie stand-alone mit 

    evcc charger --log trace

beobachten? Unter welcher Bedingung?

DocLifeguard, 2023-09-10:
Nachfolgend ein Tracelog von evcc charger, wenn kein Auto angeschlossen ist. Das UI zeigt dazu Leistung: 0,0 kW und Geladen: 11,7 kWh an:
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 01 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 02 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 01 2c
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 03 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 03 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 04 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 04 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 05 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 05 00 00 00 05 ff 04 02 00 00
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 06 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 06 00 00 00 05 ff 04 02 00 e3
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 07 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 07 00 00 00 05 ff 04 02 00 e1
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 08 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e4
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 09 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 09 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 0a 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 0a 00 00 00 05 ff 03 02 00 00
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 0b 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 0b 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/09/10 18:57:27 modbus: send 00 0c 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/09/10 18:57:27 modbus: recv 00 0c 00 00 00 07 ff 04 04 00 00 4e 96
Power:          0W
Energy:         30.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 227V 225V 228V
Charge status:  A
Enabled:        false
Charged:        0.0kWh
Duration:       5h35m18s

Sobald ein Auto am Kabel hängt und der Ladevorgang beginnt, zeigt das UI Geladen: 193x,x kWh an (x zwischen 0 und 9). Das Tracelog habe ich ja bereits weiter oben angehängt. 
Wenn ich dann evcc stoppe und sofort wieder starte, springt "Geladen" im UI auf 0,0 kWh, das Tracelog bleibt aber gleich (wie beschrieben). Obwohl dann Charged im Log weiterhin einen viel zu hohen Wert anzeigt, ist die Anzeige im UI korrekt.

premultiply, 2023-09-10:
@andig Auch hier ChargedEnergy() raus oder haben wir noch irgendwo eine potenzielle Fehlermöglichkeit in der Vestel-Implementierung?

andig, 2023-09-10:
Das wäre auch mein Impuls. Passt aber nicht zu den Erklärungen oben. Überhaupt nicht :(

andig, 2023-09-12:
I denke ohne eigenen Zugang zu der Box kommen wir hier nicht weiter. Bei Bedarf bitte Portweiterleitung und Mal an info@evcc.io.

closed 2023-09-12 by andig