evcc-io/evcc #9788: Laden starten nicht mehr bei PV-Überschuss

maatinh, 2023-09-08:
### Describe the bug

Seit einiger Zeit verhält sich evcc nicht mehr so zuverlässig wie in der Vergangenheit. Ich habe eine Zoe (für evcc keine Möglichkeit SOC abzufragen) mit einem NRGKick+NRGKick-Connect (NRGKick weiß nicht, ob Auto angeschlossen ist oder nicht!).
Sobald ich den Ladestecker anstecke, fängt das Auto erstmal mit Laden an. evcc hat dann immer erkannt, wenn kein Überschuss da war und das Laden abgebrochen. Vor ein paar Wochen hat das dann nicht mehr zuverlässig funktioniert, scheint nun aber wieder zu gehen (Nightly / https://github.com/evcc-io/evcc/commit/d7ca1fc3). Dafür hat evcc heute morgen nicht angefangen das Auto zu laden als genug Überschuss da war.
Das Auto wurde gestern Abend angeschlossen, hat kurz geladen und evcc hat den Ladevorgang gestoppt. Mitternächlich wird evcc gestoppt. Heute morgen dann wie gesagt der Fehler. Nachdem ich manuell die LadeSteuerung auf "Aus" und dann wieder auf "PV" geschalten habe, ist der Ladevorgang dann gestartet.

Logs:

Auto gestern Abend angesteckt:
```
[lp-1  ] DEBUG 2023/09/07 20:53:39 charge power: 0W
[lp-1  ] DEBUG 2023/09/07 20:53:40 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/07 20:53:40 charge total import: 3240.310kWh
[lp-1  ] DEBUG 2023/09/07 20:53:40 charger status: B
[lp-1  ] DEBUG 2023/09/07 20:53:40 pv charge current: 0A = 0A + -1.58A (363W @ 1p)
[lp-1  ] DEBUG 2023/09/07 20:54:09 charge power: 0W
[lp-1  ] DEBUG 2023/09/07 20:54:10 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/07 20:54:10 charge total import: 3240.310kWh
[lp-1  ] DEBUG 2023/09/07 20:54:10 charger status: B
[lp-1  ] DEBUG 2023/09/07 20:54:10 pv charge current: 0A = 0A + -1.79A (411W @ 1p)
[lp-1  ] DEBUG 2023/09/07 20:54:39 charge power: 1760W
[lp-1  ] DEBUG 2023/09/07 20:55:00 charge currents: [9.41 0 0]A
[lp-1  ] DEBUG 2023/09/07 20:55:00 charge total import: 3240.327kWh
[lp-1  ] DEBUG 2023/09/07 20:55:00 charger status: C
[lp-1  ] DEBUG 2023/09/07 20:55:00 charge start energy: 3.24e+03kWh
[lp-1  ] INFO 2023/09/07 20:55:00 start charging ->
[lp-1  ] DEBUG 2023/09/07 20:55:00 wake-up timer: stop
[lp-1  ] WARN 2023/09/07 20:55:01 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/09/07 20:55:01 charger: guard elapse
[lp-1  ] DEBUG 2023/09/07 20:55:01 guard timer inactive
[lp-1  ] DEBUG 2023/09/07 20:55:01 pv timer elapse
[lp-1  ] DEBUG 2023/09/07 20:55:01 pv timer inactive
[lp-1  ] DEBUG 2023/09/07 20:55:01 pv charge current: 0.239A = 10A + -9.76A (2245W @ 1p)
[lp-1  ] DEBUG 2023/09/07 20:55:01 site power 2245W >= 1W disable threshold
[lp-1  ] DEBUG 2023/09/07 20:55:01 pv disable in 0s
[lp-1  ] DEBUG 2023/09/07 20:55:01 pv disable timer elapsed
[lp-1  ] DEBUG 2023/09/07 20:55:01 charger disable
[lp-1  ] DEBUG 2023/09/07 20:55:01 wake-up timer: stop
[lp-1  ] DEBUG 2023/09/07 20:55:09 charge power: 0W
[lp-1  ] DEBUG 2023/09/07 20:55:10 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/07 20:55:10 charge total import: 3240.328kWh
[lp-1  ] DEBUG 2023/09/07 20:55:10 charger status: B
[lp-1  ] DEBUG 2023/09/07 20:55:10 final charge energy: 0.003kWh
[lp-1  ] INFO 2023/09/07 20:55:10 stop charging <-
[lp-1  ] DEBUG 2023/09/07 20:55:10 charge total import: 3240.328kWh
[lp-1  ] DEBUG 2023/09/07 20:55:10 pv charge current: 0A = 0A + -1.83A (422W @ 1p)
[lp-1  ] DEBUG 2023/09/07 20:55:10 pv enable timer reset
[lp-1  ] DEBUG 2023/09/07 20:55:10 pv timer inactive
[lp-1  ] DEBUG 2023/09/07 20:55:39 charge power: 0W
[lp-1  ] DEBUG 2023/09/07 20:55:40 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/07 20:55:40 charge total import: 3240.328kWh
[lp-1  ] DEBUG 2023/09/07 20:55:40 charger status: B
[lp-1  ] DEBUG 2023/09/07 20:55:40 pv charge current: 0A = 0A + -2.01A (463W @ 1p)
```

Mitternachts-Restart:
```
[lp-1  ] DEBUG 2023/09/07 23:59:39 charge power: 0W
[lp-1  ] DEBUG 2023/09/07 23:59:40 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/07 23:59:40 charge total import: 3240.328kWh
[lp-1  ] DEBUG 2023/09/07 23:59:40 charger status: B
[lp-1  ] DEBUG 2023/09/07 23:59:40 pv charge current: 0A = 0A + -0.943A (217W @ 1p)
[lp-1  ] DEBUG 2023/09/08 00:00:03 charge total import: 3240.328kWh
[main  ] INFO 2023/09/08 00:00:08 evcc 0.119.4 (d7ca1fc3)
[main  ] INFO 2023/09/08 00:00:08 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/08 00:00:08 starting ui and api at :7070
[lp-1  ] WARN 2023/09/08 00:00:08 poll mode '{always 1m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2023/09/08 00:00:08 site config:
[site  ] INFO 2023/09/08 00:00:08   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/09/08 00:00:08     grid:      power ✓ energy ✓ currents ✗
[... 270 lines cut ...]

Woran kann das liegen?

Gruß,
Martin

### Steps to reproduce

siehe oben

### Configuration details

```yaml
siehe oben
```

### Log details

```text
siehe oben
```

### What type of operating system are you running?

Docker container

andig, 2023-09-08:
> Hier erkennt evcc dass es Laden könnte, es wird aber nicht richtig gestartet.

Es wird gestartet, aber nichts passiert:

```
[lp-1  ] DEBUG 2023/09/08 08:41:50 charger enable
[lp-1  ] DEBUG 2023/09/08 08:42:09 charge currents: [0 0 0]A
```

Warum lässt sich nicht sagen. Evcc scheint das Richtige zu tun. Probier mal bitte den Charger auf `trace` zu setzen damit wir mehr sehen.

maatinh, 2023-09-08:
Kann ich den loglevel im laufenden Betrieb ändern? Oder stoppen, config ändern, starten?

andig, 2023-09-12:
Keine neuen Erkenntnisse.

closed 2023-09-12 by andig

maatinh, 2023-09-13:
Hi,

sorry - bisher konnte ich das noch nicht wieder reproduzieren, da das Auto nicht geladen wurde bzw. nicht immer angesteckt war.
Heute Vormittag hat das Auto erst geladen. Dann wurde um ca. 10:14 das Laden aufgrund von zu wenig Überschuss gestoppt.
Gegen 10:45 hätte es wieder gestartet werden müssen. Das ist aber nicht erfolgt:

```

[nrgconn] TRACE 2023/09/13 10:11:38 GET http://192.168.XXX.XXX/api/measurements/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:11:39 {"ChargingCurrentPhase":[15.6,0.0,0.0],"ChargingEnergy":0.074,"ChargingEnergyOverAll":3264.646,"ChargingEnergyPhase":[0.074,0.0,0.0],"ChargingPower":3.41,"ChargingPowerPhase":[3.41,0.0,0.0],"Frequency":49.99,"TemperatureMainUnit":40.0,"VoltagePhase":[235.9,128.0,0.0],"MacAddress":"00:11:22:33:44:55","Online":true,"Timestamp":1694592699}
[lp-1  ] DEBUG 2023/09/13 10:11:39 charge power: 3410W
[lp-1  ] DEBUG 2023/09/13 10:11:39 charge currents: [15.6 0 0]A
[lp-1  ] DEBUG 2023/09/13 10:11:39 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/13 10:11:39 charge total import: 3264.646kWh
[nrgconn] TRACE 2023/09/13 10:11:39 GET http://192.168.XXX.XXX/api/settings/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:11:39 {"Info":{"ErrorCodes":["0"],"Connected":true},"Values":{"ChargingStatus":{"Charging":true},"ChargingEnergy":{"Value":199.98,"Limited":false,"Min":0.5,"Max":200.0},"ChargingCurrent":{"Value":16.0,"Min":6.0,"Max":16.0},"Settings":{"Average_kWh_consumption":{"Value":20.0,"Min":0.1,"Max":99.9},"CostsPer_kWh":{"Value":0.28,"Min":0.01,"Max":2.49},"Efficacy":{"Value":95.0,"Min":10.0,"Max":100.0},"BLETransmissionPower":{"Value":4,"Min":1,"Max":4}},"DeviceMetadata":{"Name":"NRGkick_Garage_2EBD","Password":null,"PasswordMatch":true}},"ProductType":"NRGkick","HardwareVersion":"1","FirmwareVersion":"1","MacAddress":"00:11:22:33:44:55","Online":true,"Timestamp":1694592699}
[lp-1  ] DEBUG 2023/09/13 10:11:39 charger status: C
[lp-1  ] DEBUG 2023/09/13 10:11:39 pv charge current: 7.9A = 16A + -8.1A (1862W @ 1p)
[nrgconn] TRACE 2023/09/13 10:11:39 PUT http://192.168.XXX.XXX/api/settings/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:11:39 {"Values":{"ChargingStatus":{"Charging":true},"ChargingCurrent":{"Value":7},"DeviceMetadata":{"Password":"****"}}}
[lp-1  ] DEBUG 2023/09/13 10:11:39 max charge current: 7A
[nrgconn] TRACE 2023/09/13 10:12:08 GET http://192.168.XXX.XXX/api/measurements/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:12:09 {"ChargingCurrentPhase":[6.28,0.0,0.0],"ChargingEnergy":0.084,"ChargingEnergyOverAll":3264.656,"ChargingEnergyPhase":[0.084,0.0,0.0],"ChargingPower":0.69,"ChargingPowerPhase":[0.69,0.0,0.0],"Frequency":50.0,"TemperatureMainUnit":40.0,"VoltagePhase":[237.7,125.2,0.0],"MacAddress":"00:11:22:33:44:55","Online":true,"Timestamp":1694592728}
[lp-1  ] DEBUG 2023/09/13 10:12:09 charge power: 690W
[nrgconn] TRACE 2023/09/13 10:12:19 GET http://192.168.XXX.XXX/api/measurements/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:12:21 {"ChargingCurrentPhase":[6.26,0.0,0.0],"ChargingEnergy":0.086,"ChargingEnergyOverAll":3264.658,"ChargingEnergyPhase":[0.086,0.0,0.0],"ChargingPower":0.69,"ChargingPowerPhase":[0.69,0.0,0.0],"Frequency":50.0,"TemperatureMainUnit":40.0,"VoltagePhase":[236.9,124.8,0.0],"MacAddress":"00:11:22:33:44:55","Online":true,"Timestamp":1694592739}
[lp-1  ] DEBUG 2023/09/13 10:12:21 charge currents: [6.26 0 0]A
[lp-1  ] DEBUG 2023/09/13 10:12:21 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/13 10:12:21 charge total import: 3264.658kWh
[nrgconn] TRACE 2023/09/13 10:12:21 GET http://192.168.XXX.XXX/api/settings/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:12:21 {"Info":{"ErrorCodes":["0"],"Connected":true},"Values":{"ChargingStatus":{"Charging":true},"ChargingEnergy":{"Value":199.98,"Limited":false,"Min":0.5,"Max":200.0},"ChargingCurrent":{"Value":7.0,"Min":6.0,"Max":16.0},"Settings":{"Average_kWh_consumption":{"Value":20.0,"Min":0.1,"Max":99.9},"CostsPer_kWh":{"Value":0.28,"Min":0.01,"Max":2.49},"Efficacy":{"Value":95.0,"Min":10.0,"Max":100.0},"BLETransmissionPower":{"Value":4,"Min":1,"Max":4}},"DeviceMetadata":{"Name":"NRGkick_Garage_2EBD","Password":null,"PasswordMatch":true}},"ProductType":"NRGkick","HardwareVersion":"1","FirmwareVersion":"1","MacAddress":"00:11:22:33:44:55","Online":true,"Timestamp":1694592741}
[lp-1  ] DEBUG 2023/09/13 10:12:21 charger status: C
[lp-1  ] DEBUG 2023/09/13 10:12:21 pv charge current: 10.7A = 7A + 3.72A (-855W @ 1p)
[nrgconn] TRACE 2023/09/13 10:12:21 PUT http://192.168.XXX.XXX/api/settings/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:12:21 {"Values":{"ChargingStatus":{"Charging":true},"ChargingCurrent":{"Value":10},"DeviceMetadata":{"Password":"****"}}}
[lp-1  ] DEBUG 2023/09/13 10:12:21 max charge current: 10A
[nrgconn] TRACE 2023/09/13 10:12:38 GET http://192.168.XXX.XXX/api/measurements/00:11:22:33:44:55
[nrgconn] TRACE 2023/09/13 10:12:39 {"ChargingCurrentPhase":[8.92,0.0,0.0],"ChargingEnergy":0.093,"ChargingEnergyOverAll":3264.666,"ChargingEnergyPhase":[0.093,0.0,0.0],"ChargingPower":1.61,"ChargingPowerPhase":[1.65,0.0,0.0],"Frequency":50.0,"TemperatureMainUnit":40.0,"VoltagePhase":[236.1,126.9,0.0],"MacAddress":"00:11:22:33:44:55","Online":true,"Timestamp":1694592758}
[lp-1  ] DEBUG 2023/09/13 10:12:39 charge power: 1610W
[lp-1  ] DEBUG 2023/09/13 10:12:39 charge currents: [8.92 0 0]A
[lp-1  ] DEBUG 2023/09/13 10:12:39 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/13 10:12:39 charge total import: 3264.666kWh
[... cut ...]
[nrgconn] TRACE 2023/09/13 11:03:19 {"Info":{"ErrorCodes":["0"],"Connected":true},"Values":{"ChargingStatus":{"Charging":false},"ChargingEnergy":{"Value":199.97,"Limited":false,"Min":0.5,"Max":200.0},"ChargingCurrent":{"Value":15.0,"Min":6.0,"Max":16.0},"Settings":{"Average_kWh_consumption":{"Value":20.0,"Min":0.1,"Max":99.9},"CostsPer_kWh":{"Value":0.28,"Min":0.01,"Max":2.49},"Efficacy":{"Value":95.0,"Min":10.0,"Max":100.0},"BLETransmissionPower":{"Value":4,"Min":1,"Max":4}},"DeviceMetadata":{"Name":"NRGkick_G

maatinh, 2023-09-18:
@andig Hilft das (s.o.) weiter?

andig, 2023-09-18:
So wie ich das sehe schreiben wir fleissig Ladefreigabe und Strom, es passiert aber nichts:

    {"Values":{"ChargingStatus":{"Charging":true},"ChargingCurrent":{"Value":7},"DeviceMetadata":{"Password":"****"}}}

Da bliebe eigentlich nur mal den Support zu fragen, warum das nicht verarbeitet wird. Vielleicht ließe sich auch irgendwie rausfinden, ob die Freigabe im Charger erfolgreich verarbeitet wurde?

maatinh, 2023-09-18:
@andig : Ich finde gar kein PUT Request um das Laden zu stoppen?

[image]

Für mich sieht es so aus, als würde das Laden über GET gestoppt und über PUT gestartet. Ist das korrekt?
Wäre es möglich für den NRGKick das Laden zu stoppen/starten indem nur der Charging-Status geändert wird? Ohne gleichzeitige Angabe des Charging-Currents-Parameters?

andig, 2023-09-18:
Dein Punkt war doch:

> Laden starten nicht mehr bei PV-Überschuss

Im Log (bzw. im Screenshot- deshalb kann ich auch nichts raus kopieren) sehe ich "disable in 1min". Vmtl. war dein Auto vorher schon voll- jedenfalls fehlt der Teil vom Log wo abgeschaltet worden wäre...

maatinh, 2023-09-18:
Der Teil vom Screenshot steht auch oben genauso im vorherigen Post. Das Auto war zu dem Zeitpunk nicht voll.

andig, 2023-09-21:
Es ändert doch nichts. Im Logfile siehst Du, dass evcc kein Ladeende sendet. Dennoch stoppt das Auto anscheinend. Warum  kann ich Dir nicht sagen. Außerdem gibts doch hier um den Ladestart???

andig, 2023-09-26:
Kannst du das bitte als „code“ formatieren?

maatinh, 2023-09-26:
Habs geändert. Im Code kann man die relevanten Stellen nicht hervorheben, darum hatte ich es so gemacht.

andig, 2023-09-27:
Im Log fehlt `site` daher ist der Kontext unklar.