evcc-io/evcc #9790: VW Wallbox startet mit einer Phase

Marsx79, 2023-09-08:
### Describe the bug

Seit dem Update auf Version 0.119.x startet der Ladevorgang bei "Schnell" nur auf einer Phase mit 3,8kW, und schaltet nicht auf drei Phasen.

Workaround: Laden auf "Aus" dann wieder auf "Schnell". Danach lädt er mit drei Phasen und 11kW.

Mit Version 0.118.x bestand das Problem nicht.

VW ID.4 mit VW Charger Pro.

### Steps to reproduce

1.  leer
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2023/09/08 10:56:29 evcc 0.119.4
[main  ] INFO 2023/09/08 10:56:29 using config file: /etc/evcc.yaml
[db    ] INFO 2023/09/08 10:56:29 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2023/09/08 10:56:29 Local SKI:  0e683ff183aa462a66bee0026961a66e2925b3cc
[eebus ] DEBUG 2023/09/08 10:56:29 mdns: announce
[eebus ] DEBUG 2023/09/08 10:56:29 mdns: using zeroconf
[eebus ] DEBUG 2023/09/08 10:56:29 mdns: start search
[eebus ] DEBUG 2023/09/08 10:56:29 starting websocket server on :4712
[eebus ] DEBUG 2023/09/08 10:56:29 ski: c68e16b1456eb147a9585b926e913c74b4d230a0 name: Elli-Wallbox-2218A2DK0X brand:>
[eebus ] DEBUG 2023/09/08 10:56:29 delaying connection to c68e16b1456eb147a9585b926e913c74b4d230a0 by 254ms to minimi>
[eebus ] DEBUG 2023/09/08 10:56:29 trying to connect to c68e16b1456eb147a9585b926e913c74b4d230a0 at 192.168.178.141
[eebus ] DEBUG 2023/09/08 10:56:29 initiating connection to c68e16b1456eb147a9585b926e913c74b4d230a0 at 192.168.178.1>
[eebus ] DEBUG 2023/09/08 10:56:30 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/09/08 10:56:30 Recv: d:_i:47859_Elli-Wallbox-2218A2DK0X: to NodeManagement read 302230 NodeManage>
[eebus ] DEBUG 2023/09/08 10:56:30 Send: d:_i:47859_Elli-Wallbox-2218A2DK0X reply 2 302230 NodeManagementDetailedDisc>
[eebus ] DEBUG 2023/09/08 10:56:30 Send: d:_i:47859_Elli-Wallbox-2218A2DK0X result 3 302230 ResultData 0
[eebus ] DEBUG 2023/09/08 10:56:30 Recv: d:_i:47859_Elli-Wallbox-2218A2DK0X: to NodeManagement reply 302232 1 NodeMan>
[eebus ] DEBUG 2023/09/08 10:56:30 Send:  call 4 NodeManagementSubscriptionRequestCall
```

### Log details

```text
[lp-1  ] DEBUG 2023/09/08 10:51:52 charger status: A
[site  ] DEBUG 2023/09/08 10:52:02 ----
[lp-1  ] DEBUG 2023/09/08 10:52:02 charge power: 0W
[site  ] DEBUG 2023/09/08 10:52:04 grid power: 284W
[site  ] DEBUG 2023/09/08 10:52:04 pv power: 0W
[site  ] DEBUG 2023/09/08 10:52:04 site power: 284W
[lp-1  ] DEBUG 2023/09/08 10:52:04 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Docker container

### Version

v0.119.4

Marsx79, 2023-09-08:
Noch als Ergänzung. Es gibt bereits ein Issue: Lademenge startet nicht bei 0 kWh #9787
Bei mir startet das Laden auch mit so einem hohen Wert, der angeblich von der Solaranlage kommt. Allerdings ist meine Solaranlage noch gar nicht angeschlossen. Vielleicht hängt das ganze mit dem Problem zusammen?

StefanSchoof, 2023-09-08:
Sprichst du von dem dem Elli ID. Charger der mit dem BW Branding verkauft wird? Der hat keine automatische Phasen umschaltung

Marsx79, 2023-09-08:
Ja genau. Ach so. Ich dachte das hängt mit der Phasenumschaltung zusammen das nur mit 1/3 geladen wird.  Dann ist das wohl ein anders Problem. Fakt ist, er startet nur mit ca. 3,8 kw und nicht mit der vollen Leistung.

andig, 2023-09-08:
Mach bitte ein Logfile das das Problem zeigt ;)

closed 2023-09-08 by andig

Marsx79, 2023-09-14:
> Mach bitte ein Logfile das das Problem zeigt ;)

Hab ich gemacht.