evcc-io/evcc #9804: Anteil Sonnenenergie falsch bei Varta Speicher ohne PV Sensor

ilion1, 2023-09-09:
### Describe the bug

EVCC läuft korrekt und lädt im PV-Modus mit Überschuss. Dennoch werden Anteile Sonnenenergie von deutlich unter 50% angezeigt. Die Ursache ist wahrscheinlich der fehlende [PV-Sensor](https://github.com/evcc-io/evcc/discussions/9781)
[image]
Wie man sieht, ist der höchste Anteil zu einem Zeitpunkt als es fast schon dunkel ist nach 20:15 Uhr. Somit vermute ich, dass nur Strom aus Speicher in die Berechnung eingeht, wie es auch in der [Doku](https://docs.evcc.io/docs/guides/setup/#ersparnisberechnung) erläutert wird.
Kann man die Logik für Modus PV-Laden auch für die Berechnung Sonnenenergie anwenden? Der Speicher liefert alle Werte für Netzbezug, Verbrauch und  Speicherentnahme und EVCC kennt die Ladeleistung. Das reicht aus meiner Sicht für eine korrekte Berechnung auch ohne PV-Sensor aus.

### Steps to reproduce

1. EVCC lädt im PV-Modus
2. Angezeigter Anteil Sonnenenergie ist sehr niedrig
3. Weitere Details in der [Diskussion](https://github.com/evcc-io/evcc/discussions/9781)

### Configuration details

```yaml
meters:
- type: template
  template: varta
  usage: grid
  host: 192.168.178.55
  port: 502
  name: grid1
- type: template
  template: varta
  usage: pv
  host: 192.168.178.55
  port: 502
  name: pv2
- type: template
  template: varta
  usage: battery
  host: 192.168.178.55
  port: 502
  name: battery3

chargers:
- type: template
  template: abl
  id: 1
  device: /dev/ttyACM0
  baudrate: 38400
  comset: 8E1
  modbus: rs485serial
  name: wallbox5
```

### Log details

```text
siehe Logs im verlinkten Diskussionsbeitrag
```

### What type of operating system are you running?

Linux

### Version

0,118.x und 0.119.x

andig, 2023-09-09:
> Kann man die Logik für Modus PV-Laden auch für die Berechnung Sonnenenergie anwenden? Der Speicher liefert alle Werte für Netzbezug, Verbrauch und Speicherentnahme und EVCC kennt die Ladeleistung. Das reicht aus meiner Sicht für eine korrekte Berechnung auch ohne PV-Sensor aus.

Du möchtest also einen virtuellen PV Sensor?

/cc @premultiply

ilion1, 2023-09-09:
Danke für die schnelle Antwort. Wenn EVCC mit Überschuss lädt bzw. Netzbezug kleiner oder gleich 0 ist, sollte das dann nicht zu 100% Sonnenenergie führen? Wie ist die Berechnungslogik genau über die Zeit? Wahrscheinlich wird EVCC die jeweiligen Werte über alle Aktualisierungsintervalle summieren, um daraus den Wert zu ermitteln.

Folgende Beispiele mit Werten, die von Wallbox zusammen mit Speicher geliefert werden können. Bitte korrigieren, falls ich das falsch verstehe:
Virtuelle PV-Leistung = Verbrauch + Ladepunkt + Einspeisung, maximaler Wert für Sonnenenergie ist 100%, minimal 0%
Zeitpunkt 1: Verbrauch 500 W, Ladepunkt 5000 W, Einspeisung 1000 W (--> virtuell 6500 W PV-Leistung, 100% Sonne)
Zeitpunkt 2: Verbrauch 2000 W, Ladepunkt 4000 W, Einspeisung -1500 W bzw. Netzbezug 1500 W (--> virtuell 4500 W PV-Leistung, 4500/6000 = 75% Sonne)
Zeitpunkt 3: Verbrauch 0 W, Ladepunkt 5000 W, Einspeisung -1000 W (--> virtuell 3000 W PV-Leistung, 3000/6000 = 50% Sonne)
Zeitpunkt 4: Verbrauch 1000 W, Ladepunkt 4000 W, Einspeisung -5000 W (--> virtuell 0 W PV-Leistung, 0% Sonne)
Gesamt= (5000 * 1,0 + 4000 * 0,75 + 6000 * 0,5 + 4000 * 0) / (5000 + 4000 + 6000 + 4000) = 0,579 --> 57,9 % Sonnenenergie

GrimmiMeloni, 2023-09-10:
> Zeitpunkt 3: Verbrauch 0 W, Ladepunkt 5000 W, Einspeisung -1000 W (--> virtuell 3000 W PV-Leistung, 3000/6000 = 50% Sonne)

Hier kann ich nicht folgen. Wieso nicht 4000W PV-Leistung? Und Gesamtleistung 5000W, und daher 4000/5000 80% Sonne ?

premultiply, 2023-09-10:
> Virtuelle PV-Leistung = Verbrauch + Ladepunkt + Einspeisung

"Verbrauch" ist selbst nur eine virtuelle Größe.

ilion1, 2023-09-10:
> > Zeitpunkt 3: Verbrauch 0 W, Ladepunkt 5000 W, Einspeisung -1000 W (--> virtuell 3000 W PV-Leistung, 3000/6000 = 50% Sonne)
> 
> Hier kann ich nicht folgen. Wieso nicht 4000W PV-Leistung? Und Gesamtleistung 5000W, und daher 4000/5000 80% Sonne ?

Vollkommen richtig. Das kommt davon, wenn man zu viel editiert. Es muss 4000/5000 mit 80% Sonne sein. Ich wollte unterschiedliche Szenarien darstellen. Eigentlich fehlt noch der Parameter Batteriebezug (positiv) bzw. Batterie laden (negativ) in der Berechnung.

ilion1, 2023-09-10:
> > Virtuelle PV-Leistung = Verbrauch + Ladepunkt + Einspeisung
> 
> "Verbrauch" ist selbst nur eine virtuelle Größe.
> 
> > ```
> > - type: template
> >   template: varta
> >   usage: pv
> >   host: 192.168.178.55
> >   port: 502
> >   name: pv2
> > ```
> 
> Hier ist ein PV-Meter definiert?
> 
> Was sagt `evcc meter`?

Das ist in der [Log-Datei](https://github.com/evcc-io/evcc/files/12552677/evcc.log.txt) zu finden, die im Discussion Thread angehangen ist:
```
[site  ] INFO 2023/09/07 21:09:43 site config:
[site  ] INFO 2023/09/07 21:09:43   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/07 21:09:43     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/07 21:09:43     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/07 21:09:43     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/07 21:09:43   vehicles:
[site  ] INFO 2023/09/07 21:09:43     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/09/07 21:09:43 loadpoint 1:
[lp-1  ] INFO 2023/09/07 21:09:43   mode:        minpv
[lp-1  ] INFO 2023/09/07 21:09:43   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/09/07 21:09:43   meters:      charge ✗
[lp-1  ] DEBUG 2023/09/07 21:09:43 phase timer inactive
[lp-1  ] DEBUG 2023/09/07 21:09:43 pv timer inactive
[lp-1  ] DEBUG 2023/09/07 21:09:43 guard timer inactive
[lp-1  ] INFO 2023/09/07 21:09:43 vehicle updated: unknown -> Opel Mokka
[lp-1  ] DEBUG 2023/09/07 21:09:43 set min current: 6
[lp-1  ] DEBUG 2023/09/07 21:09:43 set target soc: 80
[lp-1  ] DEBUG 2023/09/07 21:09:43 set max current: 16
[site  ] WARN 2023/09/07 21:09:43 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/07 21:09:43 ----
[lp-1  ] DEBUG 2023/09/07 21:09:43 charge power: 0W
[site  ] DEBUG 2023/09/07 21:09:43 pv power: 0W
[site  ] DEBUG 2023/09/07 21:09:43 battery soc: 80%
[site  ] DEBUG 2023/09/07 21:09:43 battery power: 395W
[site  ] DEBUG 2023/09/07 21:09:43 grid power: 19W
[site  ] DEBUG 2023/09/07 21:09:43 site power: 514W
[lp-1  ] DEBUG 2023/09/07 21:09:44 vehicle odometer: 7776km
[lp-1  ] DEBUG 2023/09/07 21:09:44 charger status: B

```
Vielen Dank für den Hinweis. Durch EVCC configure ist das PV-Meter in die Konfiguration gekommen. Vielleicht reicht es ja schon aus, dieses aus der Konfiguration zu entfernen, da dort immer nur 0 W kommen kann, so lange der entsprechende Sensor fehlt. Ich habe es jetzt geändert. Interessanterweise wird jetzt eine PV-Leistung angezeigt, die meiner Meinung nach aber zu gering ist:

```
[main  ] INFO 2023/09/10 13:24:04 evcc 0.119.5
[main  ] INFO 2023/09/10 13:24:04 using config file: /home/pi/evcc/evcc.yaml
[db    ] INFO 2023/09/10 13:24:05 using sqlite database: /root/.evcc/evcc.db
grid1
-----
Power: -3304W

pv2
---
Power: 0W

battery3
--------
Power: -2479W
Soc:   76%
```

nach Entfernen des PV-Sensors aus der Konfiguration:
```
[main  ] INFO 2023/09/10 13:25:21 evcc 0.119.5
[main  ] INFO 2023/09/10 13:25:21 using config file: /home/pi/evcc/evcc.yaml
[db    ] INFO 2023/09/10 13:25:21 using sqlite database: /root/.evcc/evcc.db
grid1
-----
Power: -3287W

battery3
--------
Power: -2496W
Soc:   77%
```

Jetzt scheint mir aber der ermittelte Wert für PV-Leistung zu gering zu sein (In ist kleiner als Out):
```
[lp-1  ] DEBUG 2023/09/10 13:27:39 charge power: 0W
[site  ] DEBUG 2023/09/10 13:27:39 battery soc: 78%
[site  ] DEBUG 2023/09/10 13:27:39 battery power: -2474W
[site  ] DEBUG 2023/09/10 13:27:39 grid power: -3269W
[site  ] DEBUG 2023/09/10 13:27:39 pv power: 3369W
[site  ] DEBUG 2023/09/10 13:27:39 site power: -5643W
[lp-1  ] DEBUG 2023/09/10 13:27:39 charger status: A
```
[image]

andig, 2023-09-10:
Ich kann bisher keinen Fehler feststellen- Thema nach Discussion verschoben.