evcc-io/evcc #9806: Crashes with Fronius Wattpilot since last Update to 0.119.4

savus4, 2023-09-09:
### Describe the bug

Hallo zusammen,

erstmal danke für super Software! Funktioniert echt gut und ist auch noch schön anzuschauen!

Seit dem neusten Update auf die [0.119.5](https://github.com/evcc-io/evcc/releases/tag/0.119.5) funktioniert evcc mit meinem Wattpilot nicht mehr. Hat vermutlich etwas mit der [Versionsänderung der wattpilot Library](https://github.com/evcc-io/evcc/commit/478d586eb3073878eb62814b894c084d075a09af) zu tun könnte ich mir vorstellen?

Downgrade auf die letzte Version lässt es erstmal wieder laufen.
Danke fürs Anschauen bei Gelegenheit!

### Steps to reproduce

1. Update to version 0.119.5 from 0.119.4
2. evcc will always crash now and restart

### Configuration details

```yaml
<details><summary>Konfiguration (/etc/evcc.yaml)</summary>

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 15s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1
  usage: grid
  host: 192.168.53.121
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.53.121
  name: pv2
- type: template
  template: fronius-solarapi-v1
  usage: battery
  host: 192.168.53.121
  name: battery3

chargers:
- type: template
  template: fronius-wattpilot
  host: 192.168.53.23
  password: *****
  name: wallbox5

vehicles:
- type: template
  template: renault
  title: Twingo
  user: *****
  password: *****
  vin: *****
  capacity: 22
  phases: 3
  icon: car
[... 146 lines cut ...]
Sep 09 11:19:08 riepls-pi evcc[4530]: [lp-1  ] DEBUG 2023/09/09 11:19:08 charger: guard elapse
Sep 09 11:19:08 riepls-pi evcc[4530]: [lp-1  ] DEBUG 2023/09/09 11:19:08 guard timer inactive
Sep 09 11:19:08 riepls-pi evcc[4530]: panic: interface conversion: interface {} is nil, not string
Sep 09 11:19:08 riepls-pi evcc[4530]: goroutine 437 [running]:
Sep 09 11:19:08 riepls-pi evcc[4530]: github.com/mabunixda/wattpilot.(*Wattpilot).GetRFID(...)
Sep 09 11:19:08 riepls-pi evcc[4530]:         github.com/mabunixda/wattpilot@v1.6.0/wattpilot.go:712
Sep 09 11:19:08 riepls-pi evcc[4530]: github.com/evcc-io/evcc/charger.(*Wattpilot).Identify(0x4000740ba0?)
Sep 09 11:19:08 riepls-pi evcc[4530]:         github.com/evcc-io/evcc/charger/fronius-wattpilot.go:135 +0x80
Sep 09 11:19:08 riepls-pi evcc[4530]: github.com/evcc-io/evcc/core.(*Loadpoint).createSession(0x40008d0000)
Sep 09 11:19:08 riepls-pi evcc[4530]:         github.com/evcc-io/evcc/core/loadpoint_session.go:40 +0xe4
Sep 09 11:19:08 riepls-pi evcc[4530]: github.com/evcc-io/evcc/core.(*Loadpoint).evVehicleConnectHandler(0x40008d0000)
Sep 09 11:19:08 riepls-pi evcc[4530]:         github.com/evcc-io/evcc/core/loadpoint.go:470 +0x204
Sep 09 11:19:08 riepls-pi evcc[4530]: reflect.Value.call({0x1bef780?, 0x4000e51cc0?, 0x4000ae3f70?}, {0x211aab0, 0x4}, {0x47e33c0, 0x0, 0x0?})
Sep 09 11:19:08 riepls-pi evcc[4530]:         reflect/value.go:596 +0x994
Sep 09 11:19:08 riepls-pi evcc[4530]: reflect.Value.Call({0x1bef780?, 0x4000e51cc0?, 0x0?}, {0x47e33c0?, 0x10?, 0x4000ae3f70?})
Sep 09 11:
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.5

StefanSchoof, 2023-09-09:
Siehe auch https://github.com/evcc-io/evcc/discussions/9779

savus4, 2023-09-09:
@StefanSchoof Ah sorry, hab nicht weit genug nach unten in den Discussions geschaut. 

Dann hier zu machen am besten?

StefanSchoof, 2023-09-09:
Da es ein bestätigen Problem ist, ist es denke in Ordnung dafür ein Issue zu haben. Macht es leichter andern die drauf stoßen den Report zu finden

andig, 2023-09-09:
/cc @mabunixda

mabunixda, 2023-09-09:
PR ist im Teststadium bei mir lokal, morgen oder Montag kommt der PR zum mergen

neujbit, 2023-09-09:
Bei mir crasht evcc auch mit dem Wattpiloten

Pixinger, 2023-09-09:
Eine kleine Erweiterung zum Issue.
Seit dem Update auf 0.119.5 kann ich EVCC auch nicht mehr starten. Ich vermute das Problem wird ähnlich sein, hat allerdings eine etwas andere Fehlermeldung. Bin mir nicht sicher ob dafür ein eigenes Ticket erstellt werden sollte. Vielleicht hilft die Information ja weiter.

```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/09/09 19:53:37 evcc 0.119.5
[main  ] INFO 2023/09/09 19:53:37 using config file: /config/evcc.yaml
[main  ] INFO 2023/09/09 19:53:37 starting ui and api at :7070
[lp-1  ] WARN 2023/09/09 19:53:40 locking phase config to 1p for switchable charger
[site  ] INFO 2023/09/09 19:53:40 site config:
[site  ] INFO 2023/09/09 19:53:40   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/09 19:53:40     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/09 19:53:40     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/09 19:53:40     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/09 19:53:40   vehicles:
[site  ] INFO 2023/09/09 19:53:40     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/09/09 19:53:40 loadpoint 1:
[lp-1  ] INFO 2023/09/09 19:53:40   mode:        off
[lp-1  ] INFO 2023/09/09 19:53:40   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/09/09 19:53:40   meters:      charge ✓
[lp-1  ] INFO 2023/09/09 19:53:40     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2023/09/09 19:53:40 vehicle updated: unknown -> CAR-XYZ
[lp-1  ] INFO 2023/09/09 19:53:40 car connected
panic: interface conversion: interface {} is float64, not string
goroutine 418 [running]:
github.com/mabunixda/wattpilot.(*Wattpilot).GetRFID(...)
	github.com/mabunixda/wattpilot@v1.6.0/wattpilot.go:712
github.com/evcc-io/evcc/charger.(*Wattpilot).Identify(0x400091a440?)
	github.com/evcc-io/evcc/charger/fronius-wattpilot.go:135 +0x80
github.com/evcc-io/evcc/core.(*Loadpoint).createSession(0x4000884c00)
	github.com/evcc-io/evcc/core/loadpoint_session.go:40 +0xe4
github.com/evcc-io/evcc/core.(*Loadpoint).evVehicleConnectHandler(0x4000884c00)
	github.com/evcc-io/evcc/core/loadpoint.go:470 +0x204
reflect.Value.call({0x1bef7a0?, 0x4000fd78c0?, 0x4001099910?}, {0x211aaf0, 0x4}, {0x47e3460, 0x0, 0x0?})
	reflect/value.go:596 +0x994
reflect.Value.Call({0x1bef7a0?, 0x4000fd78c0?, 0x0?}, {0x47e3460?, 0x10?, 0x4001099910?})
	reflect/value.go:380 +0x94
github.com/asaskevich/EventBus.(*EventBus).doPublish(0x1cedaa0?, 0x40011051d0, {0x212722f?, 0x4000fd78d0?}, {0x0?, 0x1a4b708?, 0x4000da9968?})
	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:160 +0x60
github.com/asaskevich/EventBus.(*EventBus).Publish(0x400078b740, {0x212722f, 0x7}, {0x0, 0x0, 0x0})
	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:144 +0x5ac
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0x4000884c00)
	github.com/evcc-io/evcc/core/loadpoint.go:886 +0x44c
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x4000884c00, 0x3ff0000000000000?, 0x0, 0x0?, 0x0?, 0x3ff0000000000000, 0x4000bd9b80, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1498 +0x1b0
github.com/evcc-io/evcc/core.(*Site).update(0x4000eda800, {0x2b5af18?, 0x4000884c00})
	github.com/evcc-io/evcc/core/site.go:755 +0x400
github.com/evcc-io/evcc/core.(*Site).Run(0x4000eda800, 0x4000e991a0, 0x6fc23ac00)
	github.com/evcc-io/evcc/core/site.go:850 +0x1a4
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:282 +0x2c
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:281 +0x16ac
```

Installiert als Homeassistant Addon (https://github.com/evcc-io/hassio-addon) auf einem Raspi.

andig, 2023-09-10:
Depends on https://github.com/mabunixda/wattpilot/issues/23

closed 2023-09-10 by andig

Xerenas, 2024-07-09:
Ich habe aktuell aus das Problem: c**harge rater: connection is not valid**, **charge currents: connection is not valid** und **charge power: connection is not valid**. Habe zwei Wattpiloten im Einsatz. Gibt es hierfür schon eine Lösung?

maijo366, 2024-07-18:
> Ich habe aktuell aus das Problem: c**harge rater: connection is not valid**, **charge currents: connection is not valid** und **charge power: connection is not valid**. Habe zwei Wattpiloten im Einsatz. Gibt es hierfür schon eine Lösung?

Leider auch immer wieder das selbe Problem. Alles upgedatet /EVCC - Wattpilot. Problem tritt jede Minute auf. Nach Neustart ist kurz der Fehler wieder weg.

andig, 2024-07-18:
Das Problem hier ist seit fast einem Jahr behoben.