evcc-io/evcc #9817: VW Tiguan: vehicle soc: missing fuel status - no SOC displayed

markuspulch, 2023-09-10:
### Describe the bug

Hi,

I get the following error message with v0.119.5 with my Tiguan
vehicle soc: missing fuel status

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 5s # control cycle interval

sponsortoken: xxxx

# log settings
log: warn
levels:
  site: warn
  lp-1: warn
  lp-2: trace
  Tiguan: trace
  Zoe: warn

site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter
    pv:
      - pv # list of pv inverters/ meters
    battery: battery # battery meter
  residualPower: 300
  prioritySoC: 40 # give home battery priority up to this soc (empty to disable)
  bufferSoC: 90 # ignore home battery discharge above soc (empty to disable)

meters:
  - name: grid
    type: custom
    power:
      source: mqtt
      broker: 192.168.178.70:1883
      topic: W/c0847dba48e0/customParams/stats
      jq: .KostalGridPoint # parse response json https://stedolan.github.io/jq/tutorial/
      scale: -1
  - name: pv
    type: custom
    power:
      source: mqtt
      broker: 192.168.178.70:1883
      topic: W/c0847dba48e0/customParams/stats
      jq: .KostalPvGeneration # parse response json
      scale: 1
  - name: battery
    type: custom
    power:
      source: mqtt
      broker: 192.168.178.70:1883
      topic: W/c0847dba48e0/customParams/stats
      jq: .VictronGridPoint # parse response json
      scale: -1
    soc:
      source: mqtt
      broker: 192.168.178.70:1883
      topic: W/c0847dba48e0/customParams/stats  
      jq: .BatterySoc # parse response json
      scale: 1

chargers:
  - name: Wallbox1
    type: template
    template: smartwb
    host: 192.168.178.48 # IP-Adresse oder Hostname 
  - name: Wallbox2
    type: template
    template: smartwb
    host: 192.168.178.49  # IP-Adresse oder Hostname 

vehicles:
  - name: Zoe
    type: template
    template: renault
    title: Zoe # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 50 # Akku-Kapazität in kWh # Optional 
    phases: 3
    mode: pv # enable PV-charging when vehicle is identified
    minSoC: 40 # charge to at least 20% independent of charge mode
    targetSoC: 100 # limit charge to 90%
    minCurrent: 8
    maxCurrent: 16
  - name: Tiguan
    type: template
    template: id
    title: Tiguan # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: 'xxxxx' # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 13 # Akku-Kapazität in kWh # Optional 
    phases: 1
    mode: pv # enable PV-charging when vehicle is identified
    minSoC: 0 # charge to at least 20% independent of charge mode
    targetSoC: 100 # limit charge to 90%
    minCurrent: 8
    maxCurrent: 16

loadpoints:
  - title: Garage 1 # display name for UI
    charger: Wallbox1 # charger
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: Zoe # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      poll:
        mode: always
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 1 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)
  - title: Garage 2 # display name for UI
    charger: Wallbox2 # charger
    mode: "minpv" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: Tiguan # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      poll:
        mode: always
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    minCurrent: 8 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: "xxxxx" # access token
  feedin:
    type: fixed
    price: 0.1 # EUR/kWh

# mqtt message broker
mqtt:
  broker: 192.168.178.51:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: mqtt
  password: xxxxx

# influx database
influx:
  url: http://192.168.178.51:8086
  database: evcc
  user: evcc
  password: xxxx
```

### Log details

```text
[lp-2  ] DEBUG 2023/09/10 08:52:35 charge start energy: 1.84e+03kWh
[lp-2  ] INFO 2023/09/10 08:52:35 car connected
[lp-2  ] DEBUG 2023/09/10 08:52:35 pv timer elapse
[lp-2  ] DEBUG 2023/09/10 08:52:35 pv timer inactive
[lp-2  ] DEBUG 2023/09/10 08:52:35 charger: guard elapse
[lp-2  ] DEBUG 2023/09/10 08:52:35 guard timer inactive
[lp-2  ] DEBUG 2023/09/10 08:52:35 charge total import: 1839.280kWh
[lp-2  ] ERROR 2023/09/10 08:52:35 vehicle soc: missing fuel status
[lp-2  ] DEBUG 2023/09/10 08:52:35 pv charge current: 6.28A = 0A + 6.28A (-1446W @ 1p)
[lp-2  ] DEBUG 2023/09/10 08:52:35 pv enable timer reset
[lp-2  ] DEBUG 2023/09/10 08:52:35 pv timer inactive
[lp-2  ] DEBUG 2023/09/10 08:52:37 charge power: 0W
[lp-2  ] DEBUG 2023/09/10 08:52:37 vehicle odometer: 25570km
[lp-2  ] DEBUG 2023/09/10 08:52:37 charge voltages: [226 226 226]V
[lp-2  ] DEBUG 2023/09/10 08:52:37 detected connected phases: 3p
[lp-2  ] DEBUG 2023/09/10 08:52:37 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/09/10 08:52:37 charge total import: 1839.280kWh
[lp-2  ] DEBUG 2023/09/10 08:52:37 charger status: B
[lp-2  ] DEBUG 2023/09/10 08:52:37 next soc poll remaining time: 59m58s
```

### What type of operating system are you running?

Docker container

### Version

0.119.5

StefanSchoof, 2023-09-10:
My ID.3 is getting the soc. Is the Info in the App? Is your car in the offline mode?

andig, 2023-09-10:
> I get the following error message with v0.119.5 with my Tiguan

Does that imply earlier versions work? Will need trace logs of old and new version. Please run

    evcc vehicle --log trace

for both. If vehicle doesn't send the data there's nothing we can do.

andig, 2023-09-12:
Keine neuen Erkenntnisse.

closed 2023-09-12 by andig

flarno11, 2023-11-13:
Maybe same issue here for an VW e-Golf since November 8th:

```
[id    ] TRACE 2023/11/13 15:01:14 GET https://emea.bff.cariad.digital/vehicle/v1/vehicles
[id    ] TRACE 2023/11/13 15:01:16 {"data":[
    {"vin":"<VIN Redacted ID.4>","role":"PRIMARY_USER","userRoleStatus":"ENABLED","enrollmentStatus":"COMPLETED","brandCode":"V","model":"ID.4","nickname":"ID.4 1st","capabilities":[{"id":"webApp","status":[2003],"expirationDate":"2082-12-25T23:59:59Z","userDisablingAllowed":false},{"id":"automation","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"batteryChargingCare","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"batteryColdWarning","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"batterySupport","userDisablingAllowed":false},{"id":"charging","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"chargingProfiles","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":false},{"id":"chargingStations","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"climatisation","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"climatisationTimers","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":false},{"id":"dealerAppointment","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"destinations","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"fuelStatus","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":false},{"id":"ignition","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":false},{"id":"mapUpdate","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"measurements","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":false},{"id":"onlineSpeech","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"parkingBrake","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":false},{"id":"parkingInformation","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"parkingPosition","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"personalizationOnline","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"plugAndCharge","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"poiSearch","expirationDate":"2025-12-30T23:59:59Z","userDisablingAllowed":true},{"id":"readiness","userDisablingAllowed":false},{"id":"routing","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"state","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"trafficInformation","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true},{"id":"vehicleHealth","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":true},{"id":"vehicleHealthInspection","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":false},{"id":"vehicleHealthWarnings","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":false},{"id":"vehicleLights","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":false},{"id":"vehicleWakeUpTrigger","expirationDate":"2053-01-09T23:59:59Z","userDisablingAllowed":false},{"id":"webRadio","expirationDate":"2024-04-07T00:00:00Z","userDisablingAllowed":true}],"images":{},"coUsers":[],"devicePlatform":"WCAR","tags":[]},
    {"vin":"<VIN Redacted e-Golf>","role":"PRIMARY_USER","userRoleStatus":"ENABLED","enrollmentStatus":"COMPLETED","brandCode":"V","model":"e-Golf","nickname":"e-Golf","capabilities":[],"images":{},"coUsers":[],"devicePlatform":"MBB","tags":[]}]}
[id    ] TRACE 2023/11/13 15:01:16 GET https://emea.bff.cariad.digital/vehicle/v1/vehicles/***/selectivestatus?jobs=all
[id    ] TRACE 2023/11/13 15:01:17 {"userCapabilities":{"capabilitiesStatus":{"error":{"message":"Bad Gateway","errorTimeStamp":"2023-11-13T14:01:17Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4105,"group":2,"retry":true}}}}
.Soc:          missing fuel status
Capacity:      0.0kWh
Charge status: missing charging status
```

flarno11, 2023-11-14:
Seems like it was a temporary issue on VW side, the error disappeared.