evcc-io/evcc #9818: Tooltip für Werte einzelner Ladepunkte

StevieC121176, 2023-09-10:
Könnte man evtl. analog der PV Erzeugung auch den entsprechenden Tooltip für die Werte der einzelnen Ladepunkte anzeigen?

[image]

[image]

_Originally posted by @StevieC121176 in https://github.com/evcc-io/evcc/issues/9730#issuecomment-1712770967_

naltatis, 2023-09-10:
Ich bin mit den Mutli-Tooltips sowieso unglücklich. @panzerdev hatte mal vorgeschlagen diese Punkte ausklappbar zu machen sodass es dann wirklich eine Zeile pro Untereintrag gibt. Das fänd ich deutlich eleganter (wenn auch aufwänderig).

naltatis, 2024-02-13:
Blocked by https://github.com/evcc-io/evcc/issues/6781

\\cc @andig

andig, 2025-03-08:
Depends on https://github.com/evcc-io/evcc/pull/19416

closed 2025-04-05 by andig