evcc-io/evcc #9822: Easee: unexpected status: 400 (Bad Request)

TheFreaker86, 2023-09-10:
### Describe the bug

EVCC won't start. I did not change the evcc.yaml since I got EVCC to start about 2 weeks ago. The PV guided charging did not work (It charged with full power of 22 kW no matter what) in 0.119.3 but at least the GUI of EVCC worked. In the meantime I have installed Home Assistant with the two new releases that were published in the last few days. Currently I am running HA version 2023.9.1. I also tried to revert to previous releases of EVCC (0.119.4, 0.119.3, and 0.119.2) to no avail, as expected. The logfile shows the same as with 0.119.5 which I am running now. I have also increased the timeout in the charger config from 10s to 30s and after that to 60s like it was recommended in #9815. As expected, this did nothing as well. The error message seemed unrelated to that to begin with. But as stated in #9815 increasing the timeout anyway is a good idea for this charger.

I have read about the known issue that mincurrent has to be 8A. Is this still a thing and maybe causing the issue? Seems unlikely to me, but I can only take an educated guess.

It's a little bit frustrating since my sponsor token is about to expire and I haven't been able so far to test PV guided charging :-(

please advise. Thank you very much for your effort!

### Steps to reproduce

Don't know what to tell you here.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  charger: debug
  site: debug
  main: debug

# unique installation id
plant: XXXXX

interval: 10s # control cycle interval

sponsortoken: XXXXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: Sungrow-MODBUS.fritz.box  
  port: 502  
  usage: battery  
  modbus: tcpip  
  capacity: 9.6  
  name: battery4

- type: template
  template: solarman
  usage: pv
  host: mikroinverter.fritz.box
  user: !secret DeyeInvUser
  password: !secret DeyeInvPW
  name: pv1

chargers:
- type: template
  template: easee 
  user: !secret EaseeUser  
  password: !secret EaseePW  
  charger: !secret EaseeChargerID 
  timeout: 60s  
  name: wallbox5

loadpoints:
- title: Wall-E
  charger: wallbox5
  mode: minpv
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    - pv1
    battery:
    - battery3
    - battery4
  residualPower: 300
  prioritySoc: 30
  bufferSoc: 80
  bufferStartSoc: 30

vehicles:
  type: template
  template: opel
  title: Corsa-e
  icon: car
  user: !secret OpelUser
  password: !secret OpelPW
  vin: !secret OpelCorsaFIN
  capacity: 46
  phases: 3
  cache: 5m
  mode: pv
  minSoc: 30
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 16
  priority: 0
  name: ev1
    
# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.33 # EUR/kWh
    zones:
      - days: Mon-Fri
        hours: 2-5
        price: 0.2 # EUR/kWh
      - days: Sat,Sun
        price: 0.15 # EUR/kWh

    # or variable tariffs
    # type: tibber
    # token: "476c477d8a039529478ebd690d35ddd80e3308ffc49b59c65b142321aee963a4" # access token
    # homeid: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c" # optional if multiple homes associated to account
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/09/10 12:56:27 evcc 0.119.5
[main  ] INFO 2023/09/10 12:56:27 using config file: /config/evcc.yaml
[main  ] INFO 2023/09/10 12:56:27 starting ui and api at :7070
[db    ] INFO 2023/09/10 12:56:28 using sqlite database: /data/evcc.db
[main  ] FATAL 2023/09/10 12:56:28 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'easee': unexpected status: 400 (Bad Request)
[main  ] FATAL 2023/09/10 12:56:28 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.119.4 and 0.119.5

andig, 2023-09-10:
Bitte trace log

TheFreaker86, 2023-09-10:
Wow,
danke vielmals für die schnelle Antwort! Hab das Loglevel von 'charger' auf trace gesetzt. Und ich habe mal weil im Log steht unten `Username or password is invalid` jeweils mal das `!secret XYZ` durch die eigentlichen Zugangsdaten ersetzt und die Zugangsdaten anhand eines Logins auf https://easee.cloud getestet. Die Logindaten sind korrekt. Hat aber alles nix genützt, EVCC startet nicht. Hab übrigens noch battery3 aus der evcc.yaml gelöscht, das war redundant.

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/09/10 14:16:06 evcc 0.119.5
[main  ] INFO 2023/09/10 14:16:06 using config file: /config/evcc.yaml
[main  ] INFO 2023/09/10 14:16:06 starting ui and api at :7070
[db    ] INFO 2023/09/10 14:16:06 using sqlite database: /data/evcc.db
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2023/09/10 14:16:06 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[easee ] TRACE 2023/09/10 14:16:06 POST https://api.easee.com/api/accounts/login
[easee ] TRACE 2023/09/10 14:16:06 {"userName":"***","password":"***"}
--
{"errorCode":100,"errorCodeName":"InvalidUserPassword","type":null,"title":"Username or password is invalid","status":400,"detail":"[Empty in production]","instance":null,"extensions":{}}
[main  ] FATAL 2023/09/10 14:16:06 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'easee': unexpected status: 400 (Bad Request)
[main  ] FATAL 2023/09/10 14:16:06 will attempt restart in: 5m0s

andig, 2023-09-10:
Na dann isses klar- u/p kommen nicht korrekt an. Vmtl in der yaml fehlerhaft angegeben. Ohne die echten Daten hier nicht zu beantworten. Wenn Du Hilfe brauchst bitte Zugangsdaten an info@evcc.io.

closed 2023-09-10 by andig

TheFreaker86, 2023-09-10:
Seltsamerweise funktioniert es jetzt was die Wallbox angeht. Dabei habe ich wieder !secret in der evcc.yaml stehen.
/ Dafür funktioniert der MODBUS-Zugriff nicht mehr *seufz*