evcc-io/evcc #9825: Wallbox Huawei  Smart Charger  SCharger-7KS-S0 / SCharger-22KT-S0

annotationshacker, 2023-09-10:
Hi Guys,
could I ask here for a future support of the relative new charger Smart Charger SCharger-7KS-S0 / SCharger-22KT-S0 from Huawei?

Thanks!
Donald

andig, 2023-09-10:
Is there any documentation? Does it support Sunspec?

annotationshacker, 2023-09-10:
information is requested at huawei support ...

andig, 2023-09-12:
Please reopen when more details are available.

closed 2023-09-12 by andig

annotationshacker, 2023-09-14:
Response from the Huawei Support:
For the moment it's not possible to integrate the charger for the API. In future it might be possible, as for the other devices.

unfortunately  no possibilities at the moment

premultiply, 2023-09-14:
In general as good advice:
Do not buy chargers offered by inverter vendors. They are just sold to build a vendor lock-in. Very limited features, high price but beautiful branded. And nobody can help you with all the built-in software issues.

andig, 2023-10-16:
See https://github.com/evcc-io/evcc/discussions/10262