evcc-io/evcc #9826: Charger starts with max power and stays there

kscholty, 2023-09-10:
### Describe the bug

I've got a wallbox based on an Phoenix  EM-CP-PP-ETH controller.
The system always starts full power and remains that way. In PV only mode it will be switched on and off continously.

### Steps to reproduce

1) Power on controller (or set it to 32A power manually)
2) Connact car while insufficient PV
3) Switch to min+PV

Result: (See logfile):
 Charger will be enabled but max power will not be set to 6A. It remains at 32A
 From the debug log output it looks as if  `charger.setLimit()`  was called (message is `charger enable`) but the block that actually sets the limit is not executed (no message `max charge current:`).
 If I am right that can only mean that even though the charger is disabled `chargeCurrent == lp.chargeCurrent` holds true all the time, since the limit is never set as can be seen in the logfile.
It seems that right at the beginning the charge limit is set to 6A `(in prepare() ?? )` but somehow not propagated to the charger, which probably causes this behaviour.

**In the logfile I removed all the lines for lp2 to better show the events for lp1**

### Configuration details

```yaml
network:
  schema: http
  host: evcc.home.lan # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 30s # control cycle interval
 
meters:
- type: template
  template: e3dc 
  usage: grid  
  host: s10x.home.lan  
  port: 502  
  name: grid1
- type: template
  template: e3dc 
  usage: battery  
  capacity: 12
  host: s10x.home.lan  
  port: 502  
  name: battery3
- type: template
  template: victron-energy
  usage: battery
  capacity: 20
  host: nanopi
  port: 502
  name: battery1
- type: template
  template: victron-energy
  usage: pv
  host: nanopi
  port: 502
  name: pv1
- type: template
  template: shelly-3em
  host: 192.168.100.12
  usage: charge
  name: EM3

chargers:
- type: template
  template: phoenix-em-eth 
  host: 192.168.100.8
[... 274 lines cut ...]
[site  ] DEBUG 2023/09/10 12:31:14 battery power: 6955W
[site  ] DEBUG 2023/09/10 12:31:14 grid power: 457W
[site  ] DEBUG 2023/09/10 12:31:14 site power: 7532W
[lp-1  ] DEBUG 2023/09/10 12:31:14 charge voltages: [238 232 234]V
[lp-1  ] DEBUG 2023/09/10 12:31:14 detected connected phases: 3p
[lp-1  ] DEBUG 2023/09/10 12:31:14 charge currents: [15.9 15.8 16]A
[lp-1  ] DEBUG 2023/09/10 12:31:14 detected active phases: 3p
[lp-1  ] DEBUG 2023/09/10 12:31:14 charge total import: 1.170kWh
[lp-1  ] DEBUG 2023/09/10 12:31:15 charger status: C
[lp-1  ] DEBUG 2023/09/10 12:31:15 soc estimated: 29.29% (vehicle: 29.00%)
[lp-1  ] DEBUG 2023/09/10 12:31:15 vehicle soc: 29%
[lp-1  ] DEBUG 2023/09/10 12:31:15 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/09/10 12:31:15 vehicle range: 127km
[lp-1  ] DEBUG 2023/09/10 12:31:15 pv charge current: 0A = 6A + -10.9A (7532W @ 3p)

// This continues ...
```

### What type of operating system are you running?

Linux

### Version

0.119.4

andig, 2023-09-10:
@premultiply Strom nachlesen?

andig, 2023-09-10:
> In PV only mode it will be switched on and off continuously.

Das spricht dagegen. Da stimmt mit der Steuerung etwas nicht. Im PV Mode müsste diese passen!

kscholty, 2023-09-10:
Das wird deshalb dauern an- und ausgeschaltet, da bei der Berechnung des verfügbaren Stroms ein sehr hoher Wert berechnet wird. 
Also Verfügbar: 4500, ladestrom 11.000, ergo viel zu wenig Überschuss ==> Abschaltung
Beim nächsten Update: Oh, genug Überschuss, also wieder einschalten. Und das geht ewig so weiter, da nie in der Wallbox der MaxCurrent gesetzt wird.
Irgendwie stimmt `lp.ChargeCurrent` nicht mit dem tatsächlich gesetzten Wert überein.
Ich habe mal einen PR gemacht, der das `CurrentGetter` Interface implementiert. Dann müsste eigentlich `syncCharger()` das ganze korrigieren.

premultiply, 2023-09-10:
https://github.com/evcc-io/evcc/pull/9729

kscholty, 2023-09-10:
Ich habe aber einen EM-ETH, keinen EV. 
Habe jetzt in dem PR nichts gesehen, was den verändert, außer dem Template. Oder habe ich etwas übersehen?

premultiply, 2023-09-10:
Da hast du recht. Hab ich übersehen.
Bei den ganzen kryptischen Bezeichnungen kommt man aber auch wirklich schnell durcheinander.

Da muss ich die EM-Implementierung wohl auch nochmal kritisch durchschauen.
Hab schon gedacht die benutzt ohnehin niemand mehr...

Aber die aktuelle Firmware hast du trotzdem auf deinem Controller, oder?

closed 2023-09-10 by andig

kscholty, 2023-09-10:
Ich habe die Firmware nie angefasst, daher kann ich dazu nichts sagen. Wusste noch nicht einmal, dass man die aktualisieren kann. Werde ich mal schauen, ob ich was aktuelleres finde.

Nun ja, wenn du in das Logfile schaust, siehst du, dass der loadpoint gar nicht erst versucht, etwas zu setzen. Ich habe die Webseite des Controllers  verfolgt und sehe dass, wenn `maxCurrent()` aufgerufen wird, der Wert tatsächlich auch gesetzt wird. 
Ich gehe davon aus, dass der loadpoint irgendwie glaubt, es seien 6A gesetzt und daher so lange nicht `charger.maxCurrent` aufruft, wie 6A gewünscht sind.  Der Controller steht aber tatsächlich auf 32A.
`syncCharger()` hilft auch nicht, da das nötige Interface nicht implementiert ist. 
Dazu habe ich vorhin einen pull-request eröffnet, den Andi auch schon eingespielt hat. 

Das Problem ist m.E. also, dass der Loadpoint nur dann einen neuen Wert setzt, wenn der sich verändert hat. Das geht dann schief, wenn die Info, welcher Wert den nun gesetzt ist, falsch ist.

premultiply, 2023-09-11:
Das kann eigentlich(tm) nur beim Start mit einem angeschlossenen/ladenden Fahrzeug auftreten.
Bei neueren Implementierungen lesen wir dazu bei der Initialisierung teilweise die aktuelle Stromvorgabe einmalig explizit ein.

kscholty, 2023-09-11:
Habe es heute mit dem nightly getestet. Konnte das Problem nicht mehr reproduzieren.
Auch manuelle Änderungen des Ladestroms über die Webseite des Controllers werden jetzt zuverlässig erkannt und korrigiert.
Damit funktioniert das für mich gut.

PR #9828 (2023-09-10): Phoenix EM-ETH: add api.CurrentGetter
This implements the `api.currentGetter` interface from Phoenix EM-ETH controllers.
I hope it will fix #9826