evcc-io/evcc #9832: Energy price calculation off when charging battery from grid

Magnus8513, 2023-09-10:
### Describe the bug

when charging a battery from grid the charged energy is somehow taken as having a reducing impact on the resulting price for charging or house loads. See two examples below:

1. here I would expect the resulting price to be exactly the grid price as no PV yield present:
[image]

2. Here I expect a price of 13,54:
[image]

First added this as a question but now quite confident it is indeed a bug:
https://github.com/evcc-io/evcc/discussions/9821 

### Steps to reproduce

1. Evcc setup with a battery
2. Charge battery from grid
3. Check resulting price which is too low

### Configuration details

```yaml
Can add later if really relevant.
```

### Log details

```text
Can add if relevant.
```

### What type of operating system are you running?

Linux

### Version

0.119.5

andig, 2023-09-10:
I think we didn‘t even consider charging battery from grid yet and always treat that as PV-powered.

premultiply, 2023-09-10:
Exactly as @andig said.

naltatis, 2023-09-10:
> evcc geht davon aus, dass der Hausspeicher primär zur Speicherung von überschüssigem, selbstproduzierten Sonnenstrom verwendet wird. Erfüllt der Speicher auch netzdienliche Leistungen (bspw. Sonnen) oder führt eine Kalibrierung mit Netzstrom durch, ist diese Annahme nicht immer korrekt.

https://docs.evcc.io/docs/guides/setup/#ersparnisberechnung

Ja, würden wir aktuell als Sonderfall sehen den wir noch explizit behandeln wollen.

andig, 2023-09-10:
@naltatis klingt nach Backlog?

naltatis, 2023-09-10:
Vielleicht übersehe ich da das aber mMn steckt da ne Menge Komplexität drin. Wir müssen den Durchschnittsenergiewert des Speichers über Zeit (und restart) tracken und Verluste abschätzen damit das sinnvolle Werte ergibt. Ich bin mir nicht sicher, ob wir das wirklich in Angriff nehmen wollen.

Wie verbreitet ist es denn den Akku bewusst aus dem Netz zu laden? Wenn wir hier nur über Kalibrierung und manuelle Aktionen sprechen würd ich das erstmal ignorieren.

naltatis, 2023-09-11:
@Magnus8513 Sorry, ich hab mir gerade noch mal deinen ersten Screenshot in Ruhe angeschaut. Wenn ich das richtig deute, ist dein Punkt gar nicht, dass wir Hausakkustrom immer als "selbstproduziert" betrachten sondern, dass in dieser Situation auch der Hausverbrauchspreis komisch berechnet wird (16,9ct anstatt erwarteten 31,5ct).

Für das Thema Preisberechnung haben wir gerade einen Umbau in Arbeit (https://github.com/evcc-io/evcc/pull/9137). Ich werde deinen Fall da mal als weiteren Testcase mit aufnehmen. Vielleicht ist das durch die Umstellung bereits behoben. Falls nicht, sollte das eigentlich kein großes Thema sein. Ich mach diesen Issue daher wieder auf.

Magnus8513, 2023-09-11:
Richtig, primär gemeint mit dem bug war etwas was sich hoffentlich leicht fixen lässt. Nämlich, dass das "laden" von Speicher schon als "kostenlose" Produktion gewertet wird. Ihr scheint hier einfach das Vorzeichen der Leistung am Speicher zu ignorieren.

Hier genügt aus meiner sicher im ersten Schritt eine Unterscheidung ob es gerade rein oder raus geht aus dem Speicher.

Das größere Thema dahinter ist/wäre dann noch den tatsächlichen Preis von Energie im Speicher zu berechnen abhängig davon wann der womit und zu welchem Preise beladen wurde - das war hier aber gar nicht gemeint.

Zur Frage ob das ein "häufiger" use case ist:
Relevant wird/ist dad m.E. in Verbindung mit dynamischen Strom Preisen via tibber & co und genau damit experimentiere ich hier gerade und bin zuversichtlich, dass ich trotz Verlusten im Speicher bei entsprechend niedrigen Speicher kosten hier Wirtschaft zumindest nicht unsinnvoll und netzdienlich Hebeln kann. Genaue Auswertungen dazu fahre ich gerade selbst noch. 

Viele Grüße 
Magnus

PR #9137 (2023-09-13): Separate price/co2 calculate for home and loadpoints
With this change home energy use will be treated preferentially. Solar share, price and co2 for evcc controlled loadpoints are based on the remaining energy mix.

Note: This change aligns the actual calculation with bar visualization (black, green, yellow) in the UI. Due to his change your solar share per charging session will be lower than before.

fixes #9832 
docs https://github.com/evcc-io/docs/pull/441

Bezug nehmend auf https://github.com/evcc-io/evcc/issues/9097 und https://github.com/evcc-io/evcc/discussions/9111 hier der Pull Reques [... cut]
closed 2023-09-13 by naltatis