evcc-io/evcc #9835: panic: interface conversion: interface {} is nil, not string

dave-pi, 2023-09-10:
### Describe the bug

After updating evcc from 0.119.3 to 0.119.5 add-on doesn't start in HA, with the following log:

```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/09/10 21:08:08 evcc 0.119.5
[main  ] INFO 2023/09/10 21:08:08 using config file: /config/evcc.yaml
[main  ] INFO 2023/09/10 21:08:08 starting ui and api at :7070
[lp-1  ] WARN 2023/09/10 21:08:13 locking phase config to 1p for switchable charger
[site  ] INFO 2023/09/10 21:08:13 site config:
[site  ] INFO 2023/09/10 21:08:13   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/09/10 21:08:13     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/10 21:08:13     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/10 21:08:13   vehicles:
[site  ] INFO 2023/09/10 21:08:13     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/09/10 21:08:13 loadpoint 1:
[lp-1  ] INFO 2023/09/10 21:08:13   mode:        pv
[lp-1  ] INFO 2023/09/10 21:08:13   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/09/10 21:08:13   meters:      charge ✓
[lp-1  ] INFO 2023/09/10 21:08:13     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/09/10 21:08:13 phase timer inactive
[lp-1  ] DEBUG 2023/09/10 21:08:13 pv timer inactive
[lp-1  ] DEBUG 2023/09/10 21:08:13 guard timer inactive
[lp-1  ] INFO 2023/09/10 21:08:13 vehicle updated: unknown -> ID.4
[lp-1  ] DEBUG 2023/09/10 21:08:13 max charge current: 6A
[site  ] DEBUG 2023/09/10 21:08:13 ----
[lp-1  ] DEBUG 2023/09/10 21:08:13 charge power: 1383W
[site  ] DEBUG 2023/09/10 21:08:13 pv power: 0W
[site  ] DEBUG 2023/09/10 21:08:13 grid power: 1755W
[site  ] DEBUG 2023/09/10 21:08:13 site power: 1755W
[lp-1  ] DEBUG 2023/09/10 21:08:14 vehicle odometer: 10966km
[lp-1  ] DEBUG 2023/09/10 21:08:14 charge currents: [6 0 0]A
[lp-1  ] DEBUG 2023/09/10 21:08:14 charger status: C
[lp-1  ] INFO 2023/09/10 21:08:14 car connected
[lp-1  ] DEBUG 2023/09/10 21:08:14 pv timer elapse
[lp-1  ] DEBUG 2023/09/10 21:08:14 pv timer inactive
[lp-1  ] DEBUG 2023/09/10 21:08:14 charger: guard elapse
[lp-1  ] DEBUG 2023/09/10 21:08:14 guard timer inactive
panic: interface conversion: interface {} is nil, not string
goroutine 478 [running]:
github.com/mabunixda/wattpilot.(*Wattpilot).GetRFID(...)
	github.com/mabunixda/wattpilot@v1.6.0/wattpilot.go:712
github.com/evcc-io/evcc/charger.(*Wattpilot).Identify(0xc000b13c40?)
	github.com/evcc-io/evcc/charger/fronius-wattpilot.go:135 +0x65
github.com/evcc-io/evcc/core.(*Loadpoint).createSession(0xc0005f6000)
	github.com/evcc-io/evcc/core/loadpoint_session.go:40 +0xff
github.com/evcc-io/evcc/core.(*Loadpoint).evVehicleConnectHandler(0xc0005f6000)
	github.com/evcc-io/evcc/core/loadpoint.go:470 +0x265
reflect.Value.call({0x21f9840?, 0xc000a23c70?, 0x10?}, {0x2724e59, 0x4}, {0x4e2f220, 0x0, 0x7f82e1d175b8?})
	reflect/value.go:596 +0xce7
reflect.Value.Call({0x21f9840?, 0xc000a23c70?, 0x0?}, {0x4e2f220?, 0xc000c650c8?, 0xc00033e6a0?})
	reflect/value.go:380 +0xb9
github.com/asaskevich/EventBus.(*EventBus).doPublish(0x22f7d20?, 0xc000ea7a40, {0x2731659?, 0xc000a23c80?}, {0x0?, 0xc00033e670?, 0x10?})
	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:160 +0x4d
github.com/asaskevich/EventBus.(*EventBus).Publish(0xc000b12d60, {0x2731659, 0x7}, {0x0, 0x0, 0x0})
	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:144 +0x5bf
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0xc0005f6000)
	github.com/evcc-io/evcc/core/loadpoint.go:886 +0x46b
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc0005f6000, 0x0?, 0x0, 0x0?, 0x0?, 0x0, 0xc000980090, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1498 +0x212
github.com/evcc-io/evcc/core.(*Site).update(0xc00145b800, {0x31667d8?, 0xc0005f6000})
	github.com/evcc-io/evcc/core/site.go:755 +0x5ba
github.com/evcc-io/evcc/core.(*Site).Run(0xc00145b800, 0xc000d34960, 0x6fc23ac00)
	github.com/evcc-io/evcc/core/site.go:850 +0x206
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:282 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:281 +0x1a13
[... 359 lines cut ...]
github.com/asaskevich/EventBus.(*EventBus).doPublish(0x22f7d20?, 0xc000f50360, {0x2731659?, 0xc00081acd0?}, {0x0?, 0xc000b2b040?, 0x10?})
        github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:160 +0x4d
github.com/asaskevich/EventBus.(*EventBus).Publish(0xc000becec0, {0x2731659, 0x7}, {0x0, 0x0, 0x0})
        github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:144 +0x5bf
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0xc0007cc800)
        github.com/evcc-io/evcc/core/loadpoint.go:886 +0x46b
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc0007cc800, 0x0?, 0x0, 0x0?, 0x0?, 0x0, 0xc0009e6320, 0x0)
        github.com/evcc-io/evcc/core/loadpoint.go:1498 +0x212
github.com/evcc-io/evcc/core.(*Site).update(0xc000949000, {0x31667d8?, 0xc0007cc800})
        github.com/evcc-io/evcc/core/site.go:755 +0x5ba
github.com/evcc-io/evcc/core.(*Site).Run(0xc000949000, 0xc000f0eba0, 0x6fc23ac00)
        github.com/evcc-io/evcc/core/site.go:850 +0x206
github.com/evcc-io/evcc/cmd.runRoot.func4()
        github.com/evcc-io/evcc/cmd/root.go:282 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
        github.com/evcc-io/evcc/cmd/root.go:281 +0x1a13
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

evcc version 0.119.5

andig, 2023-09-10:
Already fixed

closed 2023-09-10 by andig

dave-pi, 2023-09-10:
OK sorry for the noise, I think I searched only in open issues and did not notice previous ones already closed.
Btw, just a suggestion, in new bug template it might be better to ask people to output `evcc discuss` than to ask to 'Show evcc configuration file evcc.yaml', as it could prevent accidental leak of passwords.

andig, 2023-09-11:
Good suggestion, thank you!