evcc-io/evcc #9836: Kia EV6: Login Error

snowball77, 2023-09-10:
### Describe the bug

I configured our Kia EV6, but when I start EVCC I get an error:
[main  ] ERROR 2023/09/10 22:37:49 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)

### Steps to reproduce

1. Konfigured Kia EV6 as outlined below, then start evcc

### Configuration details

```yaml
Config is (with user, password, etc. obfuscated)

vehicles:
- type: template
  template: kia
  title: EV6
  user: xxxxxxxxxxx
  password: 'xxxxxxxxx'
  vin: xxxxxxxxxxxx
  capacity: 77
  phases: 3
  icon: car
  cache: 15m
  language: de
  mode: off
  minSoc: 20
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 16
  name: ev3
```

### Log details

```text
[..]
[main  ] ERROR 2023/09/10 22:37:49 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[..]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.5

andig, 2023-09-10:
Missing full logfile

closed 2023-09-10 by andig

snowball77, 2023-09-11:
I don't know whether I can reopen. I wanted to be helpful and filtered the only message related to this. All other messages deal with the other parts of the program. I also looked into the code, but the Kia/Hyundai part is complex and partially decomposition could be improved.

❯ evcc --log debug
[main  ] INFO 2023/09/10 22:37:36 evcc 0.119.5
[main  ] INFO 2023/09/10 22:37:36 using config file: /Users/snowball/evcc.yaml
[main  ] INFO 2023/09/10 22:37:36 starting ui and api at :7070
[db    ] INFO 2023/09/10 22:37:36 using sqlite database: /Users/snowball/.evcc/evcc.db
[main  ] ERROR 2023/09/10 22:37:49 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[site  ] INFO 2023/09/10 22:37:49 site config:
[site  ] INFO 2023/09/10 22:37:49   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/09/10 22:37:49     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/10 22:37:49     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/10 22:37:49   vehicles:
[site  ] INFO 2023/09/10 22:37:49     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/09/10 22:37:49 loadpoint 1:
[lp-1  ] INFO 2023/09/10 22:37:49   mode:        off
[lp-1  ] INFO 2023/09/10 22:37:49   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/10 22:37:49   meters:      charge ✓
[lp-1  ] INFO 2023/09/10 22:37:49     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/09/10 22:37:49 phase timer inactive
[lp-1  ] DEBUG 2023/09/10 22:37:49 pv timer inactive
[lp-1  ] DEBUG 2023/09/10 22:37:49 guard timer inactive
[lp-1  ] INFO 2023/09/10 22:37:49 vehicle updated: unknown -> EV6 (offline)
[site  ] WARN 2023/09/10 22:37:49 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/10 22:37:49 ----
[lp-1  ] DEBUG 2023/09/10 22:37:49 charge power: 0W
[site  ] DEBUG 2023/09/10 22:37:49 pv power: 0W
[site  ] DEBUG 2023/09/10 22:37:49 grid power: 206W
[site  ] DEBUG 2023/09/10 22:37:49 site power: 206W
[lp-1  ] DEBUG 2023/09/10 22:37:49 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/10 22:37:49 charge total import: 302.708kWh
[lp-1  ] DEBUG 2023/09/10 22:37:49 charger status: A
[lp-1  ] INFO 2023/09/10 22:37:49 car disconnected
[site  ] DEBUG 2023/09/10 22:37:59 ----
[lp-1  ] DEBUG 2023/09/10 22:37:59 charge power: 0W
[site  ] DEBUG 2023/09/10 22:37:59 pv power: 0W
[site  ] DEBUG 2023/09/10 22:37:59 grid power: 202W
[site  ] DEBUG 2023/09/10 22:37:59 site power: 202W
[lp-1  ] DEBUG 2023/09/10 22:37:59 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/10 22:37:59 charge total import: 302.708kWh
[lp-1  ] DEBUG 2023/09/10 22:37:59 charger status: A
[site  ] DEBUG 2023/09/10 22:38:09 ----
[lp-1  ] DEBUG 2023/09/10 22:38:09 charge power: 0W
[site  ] DEBUG 2023/09/10 22:38:09 pv power: 0W
[site  ] DEBUG 2023/09/10 22:38:09 grid power: 204W
[site  ] DEBUG 2023/09/10 22:38:09 site power: 204W
[lp-1  ] DEBUG 2023/09/10 22:38:09 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/10 22:38:09 charge total import: 302.708kWh
[lp-1  ] DEBUG 2023/09/10 22:38:09 charger status: A
[site  ] DEBUG 2023/09/10 22:38:19 ----

andig, 2023-09-12:
This needs a trace log for 

    evcc vehicle

snowball77, 2023-09-13:
Here we go:

```
evcc vehicle --log trace
[main  ] INFO 2023/09/13 02:46:47 evcc 0.119.5
[main  ] INFO 2023/09/13 02:46:47 using config file: /Users/snowball/evcc.yaml
[db    ] INFO 2023/09/13 02:46:48 using sqlite database: /Users/snowball/.evcc/evcc.db
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2023/09/13 02:46:48 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[kia   ] TRACE 2023/09/13 02:46:48 POST https://prd.eu-ccapi.kia.com:8080/api/v1/spa/notifications/register
[kia   ] TRACE 2023/09/13 02:46:49 {"pushRegId":"70EDFF5C06A3D12983625B2F2D45CA1EA72F5F634D349799F3ACD097BD9C3BB2","pushType":"APNS","uuid":"c9e4dfd8-4bee-47f4-8f43-c8e2ff019658"}
--
{"retCode":"S","resCode":"0000","resMsg":{"deviceId":"8ac46353-4bf5-4131-aa49-11f839333825"},"msgId":"abdd19cc-46cf-4572-bb98-5cf915027c20"}
[kia   ] TRACE 2023/09/13 02:46:49 GET https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=fdc85c00-0a2f-4c64-bcb4-2cfb1500730a&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
[kia   ] TRACE 2023/09/13 02:46:49 <a href="https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset">Found</a>.
[kia   ] TRACE 2023/09/13 02:46:49 GET https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset
[kia   ] TRACE 2023/09/13 02:46:49 <!doctype html><html lang="en"><head><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,shrink-to-fit=no,maximum-scale=1,user-scalable=0"><meta name="theme-color" content="#ffffff"><link href="https://fonts.googleapis.com/css?family=Nanum+Gothic:400,700,800&amp;subset=korean" rel="stylesheet"><link href="https://fonts.googleapis.com/icon?family=Material+Icons" rel="stylesheet"><link rel="manifest" href="/web/v1/user/manifest.json"><link rel="icon" href="/web/v1/user/favicon_kia.ico" type="image/x-icon"><!--[if lt IE 10]><script src="/web/v1/user/fetch.js"></script><![endif]--><title></title><link href="/web/v1/user/static/css/main.9c0a4b8e.chunk.css" rel="stylesheet"></head><body><noscript>You need to enable JavaScript to run this app.</noscript><div id="root"></div><script>!function(u){function e(e){for(var o,r,t=e[0],a=e[1],n=e[2],c=0,l=[];c<t.length;c++)r=t[c],i[r]&&l.push(i[r][0]),i[r]=0;for(o in a)Object.prototype.hasOwnProperty.call(a,o)&&(u[o]=a[o]);for(p&&p(e);l.length;)l.shift()();return d.push.apply(d,n||[]),f()}function f(){for(var e,o=0;o<d.length;o++){for(var r=d[o],t=!0,a=1;a<r.length;a++){var n=r[a];0!==i[n]&&(t=!1)}t&&(d.splice(o--,1),e=s(s.s=r[0]))}return e}var r={},i={34:0},d=[];function s(e){if(r[e])return r[e].exports;var o=r[e]={i:e,l:!1,exports:{}};return u[e].call(o.exports,o,o.exports,s),o.l=!0,o.exports}s.e=function(n){var e=[],r=i[n];if(0!==r)if(r)e.push(r[2]);else{var o=new Promise(function(e,o){r=i[n]=[e,o]});e.push(r[2]=o);var t,c=document.createElement("scrip [... cut ...]
{"errId":"764c961a-b4b3-482d-a9e5-d01638264868","errCode":"4003","errMsg":"Invalid values"}
[main  ] ERROR 2023/09/13 02:46:50 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 77.0kWh
Features: [Offline]
```

snowball77, 2023-09-13:
I tried bluelinky and actually I get the same result there:
```
> bluelinky@8.2.1 debug
> cross-env LOG_LEVEL=debug ts-node debug.ts

? What Region are you in? EU
? Which brand are you using? kia
{ region: 'EU', brand: 'kia' }
Logging in...
[2023-09-13 04:48:34] debug: EU Controller created
[2023-09-13 04:48:34] debug: Bluelinky is logging in automatically, to disable use autoLogin: false
[2023-09-13 04:48:34] debug: @EuropeController.login: Trying to sign in with EuropeanBrandAuthStrategy
[2023-09-13 04:48:34] error: @EuropeController.login: sign in with EuropeanBrandAuthStrategy failed with error Error: @EuropeanBrandAuthStrategy.login: Authentication failed, cannot retrieve error message
[2023-09-13 04:48:34] debug: @EuropeController.login: Trying to sign in with EuropeanLegacyAuthStrategy

/Users/snowball/git/bluelinky/src/tools/common.tools.ts:19
    return new ManagedBluelinkyError(
           ^
ManagedBluelinkyError: @EuropeController.login: [400] Bad Request on [POST] https://prd.eu-ccapi.kia.com:8080/api/v1/user/signin - {"errId":"98e762c2-0a39-4ecf-ab85-a2012379f24a","errCode":"4003","errMsg":"Invalid values"}
    at Object.exports.manageBluelinkyError (/Users/snowball/git/bluelinky/src/tools/common.tools.ts:19:12)
    at EuropeanController.<anonymous> (/Users/snowball/git/bluelinky/src/controllers/european.controller.ts:256:13)
    at Generator.throw (<anonymous>)
    at rejected (/Users/snowball/git/bluelinky/src/controllers/european.controller.ts:6:65)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
```

andig, 2023-09-13:
Dann lässt sich leider nix machen. Falls Bluelinky eine ylösung findet können wir die übernehmen.

snowball77, 2023-09-13:
Es gibt hier exakt den gleichen Fehler. Dort half wohl, sich nochmal neu anzumelden und das Passwort zu resetten. Werde ich testen: https://github.com/Hacksore/bluelinky/issues/248#issuecomment-1625186390

snowball77, 2023-09-14:
I tried the hint by @SuperDa75 and it works for me now too. Thank you all!