evcc-io/evcc #9838: HomeCharge Wallbox benötigt ChargingScheduleAllowedChargingRateUnit "W"

tvand, 2023-09-11:
### Discussed in https://github.com/evcc-io/evcc/discussions/9678

<div type='discussions-op-text'>

<sup>Originally posted by **tvand** September  2, 2023</sup>
Ich versuche eine Verbindung mit einer HomeCharge Wallbox herzustellen (https://homecharge.de) Die unterstützt angeblich OCPP 1.6. Ich habe die Wallbox bisher erfolgreich mit SteVe betrieben (https://github.com/steve-community/steve) 

Der Ladecontroller stammt von EFR und ist dieser hier: https://www.efr.de/products/smart-energy-geraete/steuerbox/ladecontroller-secc-01/

Ich habe jetzt folgende Konfiguration probiert:
```
chargers:
- type: template
  template: ocpp
  stationid: Wall-E
  connector: 1
  connecttimeout: 5m
  timeout: 2m
  getconfiguration: false
  bootnotification: true
  name: wall_e
```

Das bindet erstmal die Wallbox ohne Fehler ein. Das Laden habe ich manuell gestartet, weiß nicht, ob das richtig war. Mit der Kommunikation hapert es aber noch:
```
[ocpp  ] TRACE 2023/09/09 11:38:39 sent JSON message to Wall-E: [2,\"1746723526\",\"SetChargingProfile\",{\"connectorId\":1,\"csChargingProfiles\":{\"chargingProfileId\":1,\"stackLevel\":0,\"chargingProfilePurpose\":\"TxProfile\",\"chargingProfileKind\":\"Relative\",\"chargingSchedule\":{\"chargingRateUnit\":\"A\",\"chargingSchedulePeriod\":[{\"startPeriod\":0,\"limit\":6}]}}}]
[ocpp  ] TRACE 2023/09/09 11:38:39 received JSON message from Wall-E: [3,\"1746723526\",{\"status\":\"Rejected\"}]
[lp-1  ] ERROR 2023/09/09 11:38:39 max charge current 6A: set charging profile: Rejected
```

Könnte das daran liegen, dass die Wallbox laut GetConfiguration nur W unterstützt?
```
{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":"true","value":"W"}
```
</div>

tvand, 2023-09-11:
Ich habe da mal einen Vorschlag gemacht, wie man das vielleicht hinzufügen könnte:
https://github.com/tvand/evcc/commit/13e4955f02786436d463a10d0060bcfd090badf6

Was meint Ihr dazu?

andig, 2023-09-12:
Das ist ja super unglücklich, v.a. da die Box das Fahrzeug ja nur in A steuern kann. Klappt es mit dem PR bei Dir? Dann spricht m.E. nichts dagegen, ich würde noch etwas anhübschen. Glückwunsch dass du dich so schnell in den Ocpp Code rein finden konntest!

tvand, 2023-09-13:
Sorry, der Test wird noch ein paar Tage dauern, ich kämpfe noch mit dem Build.

tvand, 2023-09-13:
Aktueller Stand: https://github.com/tvand/evcc/commit/995b6b89fb470fcea8eae7c5c8633b36b4e545a1
Offenbar ist phases aber immer 0. Wo kommt das her?

```
[ocpp  ] TRACE 2023/09/13 15:15:37 sent JSON message to Wall-E: [2,"366492940","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"W","chargingSchedulePeriod":[{"startPeriod":0,"limit":4140}]}}}]
[ocpp  ] TRACE 2023/09/13 15:15:37 received JSON message from Wall-E: [3,"366492940",{"status":"Accepted"}]
[ocpp  ] TRACE 2023/09/13 15:15:38 received JSON message from Wall-E: [2,"5927","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging"}]
```

tvand, 2023-09-14:
Ok, was haben wir gelernt:
- Meine Box kann minimal 4140W (3-phasig, 6A)
- Ich könnte das zwar auch begrenzen, müsste dafür aber die Box jedesmal neu booten. Nun ja.
- MeterInterval ist zwar laut Konfiguration unterstützt, gibt aber NotSupported zurück. Standard ist 300.
- Damit funktioniert minpv-Laden dreiphasig.

Box-OCPP-Log:
```
Sep 14 11:39:25 Connected to 'ws://uther.home.lan:8887/Wall-E'.
Sep 14 11:39:26 Status: 14 Preparing
Sep 14 11:39:26 Power set to maximum value 11000 W
Sep 14 11:40:11 current profiles: TxProfile(id 1) 
Sep 14 11:40:13 Status: 20 Charging
Sep 14 11:40:14 Power set to 8792 W
Sep 14 11:40:48 current profiles: TxProfile(id 1) 
Sep 14 11:40:55 Power set to 4140 W
Sep 14 12:52:29 Status: 30 Finishing
Sep 14 12:52:29 current profiles:
```

Soll ich einen PR erstellen?

andig, 2023-09-14:
Gerne 👍🏻

tvand, 2023-09-14:
Done: https://github.com/evcc-io/evcc/pull/9883

tvand, 2023-09-14:
Zusatzinformation, falls Du ein Template erstellen willst: Die Box ist wahrscheinlich in verschiedenen OEM-Versionen auf dem Markt bzw. der Ladecontroller wird in verschiedenen Boxen verwendet. Beispiele:
- https://homecharge.de
- https://www.efr.de/products/smart-energy-geraete/auto-ladebox/iasb-privatkunden/#product-detail0
- https://www.freital.de/media/custom/3303_2621_1.PDF?1672827307

andig, 2023-09-16:
Mir ist immer noch nicht klar wie das funktionieren soll. Woher weiss die Box, mir wievielen Phasen das Fahrzeug lädt? Welche Leistung soll da sinnvollerweise gesetzt werden wenn das Auto nur 1p kann? Fängt die Box bei 1,4kW überhaupt an zu laden?

tvand, 2023-09-16:
Ich sage mal, die Box kann nur mit drei Phasen laden. ich habe ihr über SteVe mal ein Profil mit einer Phase und 1340 W geschickt und da sagt sie in ihrem Log:

`Sep 13 12:58:31 Requested power 1380 W below minimum of 4140 W, setting to 0 W`

andig, 2023-09-16:
Dann hast du bei einem Auto das 1p lädt direkt verloren bzw müsste die Leistung entsprechend des Autos höher gesetzt sein.

tvand, 2023-09-17:
Nein, das denke ich nicht. Wenn man aber der ChargingSchedulePeriod außer der Leistung auch die gewünschten Phasen mitgeben würde, hätte eine OCPP-Box eigentlich alle Informationen, die sie braucht. Deshalb setze ich ja auch die Anzahl der Phasen.

Nach meinem Verständnis der [OCPP-Doku](https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf) 3.12 steuerst Du mit dem Smart Charging Profile nur das Angebot der Box ("vom Netz her gesehen"). Was das Auto dann davon abnimmt, liegt ja in seiner Entscheidung. Die Box kann die tatsächlich abgenommene Leistung immer noch über ihren Zähler messen.

Im von Dir angesprochenen Fall würde die Box ein Minimum von 4140 W dreiphasig setzen, das Auto aber nur eine Phase benutzen. Das kriegt man über das Meter der Box heraus und kann dann entsprechend reagieren. Das ist aber beim Setzen des Profils mit A genauso. Der einzige Unterschied besteht m.E. darin, wie die Box das interpretiert.

Der Use-Case ist meiner Meinung nach auch die Verwaltung von mehreren Chargepoints und die Lastenverteilung zwischen diesen. (Ich habe diese Boxen schon im Dutzend auf Hotelparkplätzen gesehen, da kann ich mir so eine Leistungssteuerung durchaus vorstellen).

andig, 2023-09-29:
> Im von Dir angesprochenen Fall würde die Box ein Minimum von 4140 W dreiphasig setzen, das Auto aber nur eine Phase benutzen. Das kriegt man über das Meter der Box heraus und kann dann entsprechend [...]

Der Punkt hat sich ja jetzt in der Implementierung geklärt ;)

closed 2023-09-29 by andig

premultiply, 2026-01-28:
Den Hersteller für eine standardkonforme Firmware ansprechen.
Wir unterstützen nur OCPP 1.6j, keine Fantasieversionen.

> Da wurden die notwendigen Einstellungen
> getconfiguration: false
> chargingrateunit: W
> aus dem Template gebügelt.

Zurecht, da es sie nicht mehr gibt, da sie automatisch erkannt werden bzw. oder obsolet sind.

PrestigeWeltweit, 2026-01-28:
Danke für die vielen und schnellen Antworten. 

Aber die Aussage stimmt wohl, die Boxen werden ohne Update mit evcc nicht funktionieren.

Ich habe einen sehr heißen Draht zum Entwickler der Wallbox und schaue mal, was sich machen lässt.
Ich halte euch auf den Laufenden.