evcc-io/evcc #9844: 1 Phasen Strombegrenzung!

stadlaa, 2023-09-11:
**Is your feature request related to a problem? Please describe.**
I have a System that can charge up to 32 Ampere on 3 phases. 
But in 1Phase-Mode my fuse is only able to handle 16 Ampere, othervise the unbalanced load  on the neutral-conductor gets to big.

**Describe the solution you'd like**
I'd like a Parameter to configure maxCurrent in 1Phase-Mode seperately from maxCurrent in 3Phase-Mode

**Describe alternatives you've considered**
I will do so with an external python script for now but it is a very unwhole solution due to bad visualization in the evcc GUI where the high Power in 1Phase-Mode is still shown in the Phase-Menu

andig, 2023-09-12:
Fully understand. Unfortunately this is currently not possible. I‘m really sorry but for time being wontfix due to technical reasons.

/cc @premultiply

closed 2023-09-12 by andig

premultiply, 2023-09-12:
Yes, looks simple but would require to add a lot more complex code and conditions. Not now. Maybe at a later time.

savus4, 2023-10-21:
@andig @premultiply 
Maybe here my question is fitting: So I suspect that wanting separate **minimum** current values for 1 and 3 phases is also a won't fix at the moment, or is it different? 
My problem is that our Renault Twingo doesn't want to charge correctly below 8 Ampere when on 3 phases. On 1 phase though, it would be nice to being able to charge with 6 ampere, for days with not a lot of PV available.

premultiply, 2023-10-22:
Its the same case.
Just set it to 8A.
No other solution yet, sorry.