evcc-io/evcc #9847: Hyundai Bluelink: login failed: deviceid not found

gaffa, 2023-09-11:
### Describe the bug

Adding a Bluelink car (2020 Hyundai Kona) does not work.

### Steps to reproduce

1. Configure vehicle
2. Start evcc
3. Look at logs

### Configuration details

```yaml
vehicles:
- type: template
  template: hyundai 
  title: Kona  
  user: h..@....net  
  password: hyundaiaccountpw  
  vin: KMH...775  
  capacity: 39.2  
  language: de  
  name: Kona
```

### Log details

```text
[main  ] ERROR 2023/09/11 20:34:05 creating vehicle Kona failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: deviceid not found
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': login failed: deviceid not found
Capacity: 39.2kWh
Features: [Offline]
```

### What type of operating system are you running?

Windows

### Version

evcc version 0.119.5

gaffa, 2023-09-11:
Further finding: I copied the same evcc.config to a docker container evcc/evcc:latest with 0.119.5 and it is working well. Windows was just for trying the configuration.
Seems platform specific.

andig, 2023-09-12:
We‘ll need a 

    evcc vehicle —log trace

Please add if problem persists. We‘ll reopen issue then.

closed 2023-09-12 by andig

gaffa, 2023-09-12:
Good morning!
After the car has been connected by the docker-hosted evcc it now also works on windows.
Keep this issue closed if you wish so but as the exact same evcc.config did not work on windows but worked on docker/linux I still assume that there is a bug in the windows version of evcc (maybe one of the dependencies behaving differently on windows?).
I do not have the time to now shutdown the docker container, disconnect and reconnect the car and re-analyze as I need to work.
If anyone comes across this and struggles to get it work on windows I recommend trying in a docker container.

SuperDa75, 2023-09-13:
If there are also connection problems in the Bluelink app, then reactivate the connection in the car under “Bluelink”.  After that, evcc also worked for me