evcc-io/evcc #9849: evcc zeigt Erzeugung obwohl es keine geben kann (mit Tibber)

tyset, 2023-09-11:
### Describe the bug

In meiner evcc Installation habe ich nur Tibber und eine Wallbox (connectIQ) konfiguriert. Ich habe keine Solaranlage und kann keinen Strom erzeugen.
Trotzdem hat evcc dieses Wochenende eine Erzeugung von 2,3kw herbei fantasiert. Davor hat es zwar einwandfrei funktioniert, aber das war doch seltsam. Siehe auch der Screenshot.
Zumal dann das Auto nicht mehr lädt, denn es muss (um bei Tibber günstig laden zu können) ja auf PV stehen und bei 2,3kw reicht die Erzeugung dann nicht aus.
[image]

Ich weiß, dass das Verhalten sich vermutlich nicht nachstellen lässt. Trotzdem wollte ich den Fehler melden, da das Verhalten sehr seltsam war. Vielleicht trägt die Meldung ja trotzdem dazu bei solche Situationen besser auszuschließen.

Eine Idee für einen noch besseren Betrieb ohne PV aber variablen Stromtarif:
- Erzeugung wird komplett abgeschaltet
- Schaltfläche "PV" wird umbenannt in "Günstig"
- Schaltfäche "Min+PV" entfällt
Dann würde der Betrieb nur mit Tibber aber ohne PV noch selbsterklärender.

### Steps to reproduce

Ich habe keine Ahnung was genau passiert ist...

### Configuration details

```yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 25s # control cycle interval

log: trace

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  host: 192.168.22.2

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070
  # per default, evcc should be available at http://evcc.local:7070

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: "removed"
    #homeid: "removed" # optional

meters:
- name: tibber
  type: template
  template: tibber-pulse
  usage: grid
  token: removed
  #homeid: removed # Optional 

chargers:
- type: connectiq
  uri:  192.168.22.5:4444               #### Bitte hier die IP Adresse der Wallbox(192.168.178.34) (mit connectIQ/chargeIQ) eintragen ####
  name: Wallbox
  
loadpoints:
- title: connectIQ
  charger: Wallbox
  mode: now
  mincurrent: 6                                    #### Bitte hier den min. Ladestrom eintragen ####
  maxcurrent: 16                                   #### Bitte hier den max. Ladestrom eintragen ####
  phases: 3                                        #### Bitte hier die Anzahl der Phasen eintragen (1 oder 3) ####
  
site:
- title: Zuhause                                   #### Bitte hier den Namen der Station eintragen ####
  meters:
    grid: tibber
```

### Log details

```text
Konnte die Situation nicht nachstellen
```

### What type of operating system are you running?

Docker container

### Version

0.118.1

VolkerK62, 2023-09-12:
Wie kannst du mit 11kW laden und dabei nur 8,8kW aus dem Netz beziehen?
Da muss einer der gemessenen Werte falsch sein.
Hat deine Wallbox eine Messeinrichtung?

andig, 2023-09-12:
Ohne Logfile leider nichs zu machen. Gerne kommentieren wenn es nachstellbar ist.

andig, 2023-09-12:
Ich mache mal wieder auf um über den ohne PV use case weiter zu diskutieren.

VolkerK62, 2023-09-12:
>Zumal dann das Auto nicht mehr lädt, denn es muss (um bei Tibber günstig laden zu können) ja auf PV stehen und bei 2,3kw reicht die Erzeugung dann nicht aus.

Du nutzt doch `smartCostLimit`, da spielt die Erzeugung überhaupt keine Rolle.

tyset, 2023-09-12:
@VolkerK62 ja, sollte man meinen. Aber nachdem evcc behauptet hat ich würde was erzeugen, hat er in PV nicht mehr geladen auch als Strom billig war. Anders kann ich mir das nicht erklären. Zu der Frage wie evcc auf die Idee kommt dass ich Strom erzeuge: gute Frage. Die Wallbox hat einen MID-Zähler, aber kann das ausschlaggebend sein? Weil ich mich sehr gewundert habe habe ich ja das Ticket eröffnet, erklären kann ich mir das verhalten nicht und ich hatte die Hoffnung ihr habt eine Idee.

@andig ich kann für das UI Thema auch ein separates Issue aufmachen falls dir das lieber ist.

VolkerK62, 2023-09-12:
Nichtladen, obwohl billig und ein zu geringer?? angezeigter Netzbezug könnte auch auf ein Problem in der Kommunikation mit Tibber hinweisen. 
Aber, ohne Log ist es nur Raten.

andig, 2023-09-12:
Dann machen wir hier mal zu. Ohne Logfile ist das Kaffeesatzleserei.

closed 2023-09-12 by andig