evcc-io/evcc #9852: The comments are a bit different from the code

912929782, 2023-09-12:
evcc/core/site.go ->sitePower(...)  Comment display：  [minus a residual margin  ]
//   - the net power exported by the site minus a residual margin  
//     (negative values mean grid: export, battery: charging
//   - if battery buffer can be used for charging

But:
evcc/core/helper.go:41 ->sitePower(...)   code display:      return grid + battery + residual

##Problem:
Can someone tell me why comments say minus residual, but code says plus residual，Is something wrong？

andig, 2023-09-12:
/cc @premultiply

premultiply, 2023-09-12:
It is just a question of the implicit sign of each value.

closed 2023-09-12 by andig

912929782, 2023-09-12:
@premultiply   Could you please check again whether this residualpower is positive or negative？

premultiply, 2023-09-12:
`residualpower` is positive for export (common usecase) and negative for import margin.

912929782, 2023-09-12:
How can you set it to a negative value in the program？How can the reserved capacity be matched with a negative value, and what is the corresponding actual scenario @premultiply

premultiply, 2023-09-12:
Just add a "-" in front of the numeric value?

`residualpower: 100` -> `residualpower: -100`