evcc-io/evcc #9860: SoC failed for Cupra

martin-zh, 2023-09-12:
### Describe the bug

SoC not working for Cupra - not shure how long. Tested via openWB 1.9.303 stable

### Steps to reproduce

Request soc

### Configuration details

```yaml
Via openwb
```

### Log details

```text
PID: 1260: Lp2: Error from EVCC: 2023/09/12 17:08:50 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/xxx/vehicles/xxx/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving (LV0)
2023-09-12 17:08:44: PID: 1260: Lp2: Requesting SoC (LV0)
```

### What type of operating system are you running?

Linux

andig, 2023-09-12:
Funktioniert wunderbar. Port 53 deutet darauf hin, dass bei Dir DNS kaputt ist. Kein evcc Thema.

IAmKonni, 2023-09-13:
Ich habe exakt den gleichen Fehler und Log Output seit dem 13.09. vormittags in meiner openWB (gleiche Version).

Wäre ja schon seltsam, das zwei Leute parallel DNS Probleme hätten. Zumal ich das vermutlich auch ausschließen kann.

Ich tippe eher auf ein Problem bei Cupra/VW. Ich muss Mal schauen, ob ich das Problem auch mit evcc nativ nachstellen kann.

IAmKonni, 2023-09-13:
`$ nslookup ola.prod.code.seat.cloud.vwgroup.com
Server:         192.168.0.10
Address:        192.168.0.10#53

Non-authoritative answer:
ola.prod.code.seat.cloud.vwgroup.com    canonical name = ola.prod.code.seat.cloud.vwgroup.com.edgekey.net.
ola.prod.code.seat.cloud.vwgroup.com.edgekey.net        canonical name = e64724.dsca.akamaiedge.net.
Name:   e64724.dsca.akamaiedge.net
Address: 104.126.37.176
Name:   e64724.dsca.akamaiedge.net
Address: 104.126.37.146
Name:   e64724.dsca.akamaiedge.net
Address: 2a02:26f0:ab00::214:8e32
Name:   e64724.dsca.akamaiedge.net
Address: 2a02:26f0:ab00::214:8e18
`

DNS Lookup sieht auch okay aus soweit.

martin-zh, 2023-09-13:
Inzwischen ist der Fehler anders:
2023-09-13 20:20:34: PID: 16053: Lp2: Error from EVCC: 2023/09/13 20:20:34 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/xxx/vehicles/xxxx/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout (LV0)
2023-09-13 20:20:34: PID

andig, 2023-09-13:
Dennoch DNS denn der Fehler passiert beim Lookup.

martin-zh, 2023-09-13:
Sehr seltsam. Ist es denn sicher, dass keine Änderungen seitens VW/Cupra gemacht wurden? Funktioniert die Abfrage aktuell wirklich noch?

andig, 2023-09-13:
> Funktioniert wunderbar.

Ja... ich habe aber auch nicht Dein DNS ;)

DrUdoZucker, 2023-09-14:
Ich bin nun ein weiteres "Opfer" dieses Problems und es wäre schön, wenn EVCC sich mit openWB in Verbindung setzen könnte, um das Problem aus der Welt zu schaffen.

Meine Fehlermeldung ist - siehe auch mein Beitrag im openWB Forum

Hardware: 2 x openWB series2 standard+x1, konfiguriert als Master + Ladepunkt
Software: openWB Version: 1.9.271

Fehlermeldung EV SoC Log:
PID: 20870: Lp1: Error from EVCC: 2023/09/14 10:58:42 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/[Nummer von mir entfernt]/vehicles/[VIN von mir entfernt]/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving (LV0)

Habe Nummern für diesen Beitrag aus string entfernt

Neustart der openWB brachte keine Änderung

martin-zh, 2023-09-14:
Könnte es mit IPV6 zu tun haben?

DrUdoZucker, 2023-09-14:
Wie kann ich denn dies feststellen oder verifizieren? Oder anders gefragt, wie kann ich als Anwender darauf Einfluss nehmen?

premultiply, 2023-09-14:
Klar kann es das. Das hängt aber an deiner lokalen Konstellation oder Konfiguration.
Oder wie Andi schon schrieb: Lokales DNS-Problem.

DrUdoZucker, 2023-09-14:
Sorry, ich verstehe dies nicht. Ich meine, was soll ich als Anwender tun? Habe folgendes gerade ohne Erfolg versucht: in der AVM 7490 FritzBox unter IPv6 die "IPv6- Unterstützung" deaktiviert - war auf aktiv zuvor. Ausserdem findet jeder meiner PCs im Browser alle Webseiten, diese müssen somit Zugriff auf einen DNS Server haben. Was muss ich denn genau prüfen, wenn es klar ist, was das Problem ist. Danke für jede Hilfe.

DrUdoZucker, 2023-09-14:
siehe openWB Forum, um openWBs Support Antwort zu dieser Frage zu lesen
[](https://openwb.de/forum/viewtopic.php?t=3134)

andig, 2023-09-14:
Mal zurück zum Thema. Aus dem Kleingedruckten:

> Tested via openWB 1.9.303 stable

Es geht also um den Aufruf der evcc cloud aus dem openWB soc client. Das DNS Problem besteht damit im evcc Cloudservice. *Damit* lässt sich jetzt arbeiten.

andig, 2023-09-14:
Depends on https://community.fly.io/t/dns-resolution-fails-inside-app/15393

DrUdoZucker, 2023-09-14:
Hi AndiG, könntest du mir bitte sagen, was dein Kommentar für mich genau bedeutet. Muss man zwingend auf 1.9.303 updaten? Oder sagtest du, man werde sich dem Problem jetzt bei EVCC annehmen?

martin-zh, 2023-09-14:
Andy ist dran. Du brauchst vorerst nichts an der openWB zu ändern

martin-zh, 2023-09-15:
Dann sind wir jetzt quitt ;-)

SoC Abfrage scheint aktuell wieder zu funktionieren.

andig, 2023-09-15:
Mit dem Kommentar kann ich ehrlich gesagt wenig anfangen- denn erstmal muss bei so einem Problem ja die Ursache identifiziert werden! Nur weil etwas auf der OpenWB nicht funktioniert heißt das nicht, dass es ein Problem von evcc ist. Du glaubst gar nicht, wie oft lokale PiHole Installationen der Grund für nicht funktionierendes DNS waren...

closed 2023-09-15 by andig

DrUdoZucker, 2023-09-15:
Andi, ich verstehe sehr gut, was du meinst. Ich habe dies selbst erlebt, jedoch denke ich, wenn man Aussagen wie "Funktioniert wunderbar. Port 53 deutet darauf hin, dass bei Dir DNS kaputt ist. Kein evcc Thema." weniger streng, oder sagen wir dogmatisch machte, hätte es weniger Enttäuschung und sicherlich nicht meinen Kommentar gegeben. Dies als konstruktiven Beitrag, ich bin froh, dass es EVCC und deine Arbeit gibt!

mairj2345, 2023-09-30:
Bei mir ist der Fehler seit ein paar Tagen wieder da, nachdem zuvor die soc-Abfrage über evcc modul der openwb seit dem ersten Auftritt vor ein paar Wochen wieder einwandfrei funktionierte.
Was war denn damals die Lösung, dass es wieder funktionierte?

Kikikanone, 2023-10-28:
Hello.
Leider bin ich in dem Thema null fit, ich weiß nur, ich hab so ein Abo für einen token von @andig und den hab ich in Open WB hinterlegt und es hat die letzten Wochen alles prima geklappt und seit gestern oder so wird der soc meines VW EUps nicht mehr korrekt angezeigt in openwb, in der VW App aber schon. Demnach kann ich aktuell leider auch kein PV laden. Muss ich irgendwas tun? Update oder ähnliches? Wenn ja, bitte ich um eine Beschreibung "für Dummies" :) VG Kilian

andig, 2023-10-28:
>  Demnach kann ich aktuell leider auch kein PV laden

1. Dein soc hat überhaupt nichts mit dem PV-Laden zu tun.
2. Wenn die VW Server nicht antworten kann nur VW das Problem lösen- da helfen keine Updates von uns.