evcc-io/evcc #9861: Pracht Alpha XT Doppellader Modbus Steuerung Katastrophal

oMtQB4, 2023-09-12:
### Discussed in https://github.com/evcc-io/evcc/discussions/9839

<div type='discussions-op-text'>

<sup>Originally posted by **eismarkt** September 11, 2023</sup>
benutz noch wer ne Pracht Alpha XT Doppellader? 
Bei mir reagiert die Modbus Steuerung nur wieder willig auf EVCC.
Wenn die Box neu gestartet wird läuft erst mal alles wie gewohnt. 
Nach ein paar Stunden aber fangen die Probleme an. 
PV Überschuss vorhanden EVCC regelt und sendet Laden starten.
in der Software der Box stehen auch 6 Ampere Ladestrom am Ladepunkt und gesamt Leistung steht auch auf 6 Ampere.
Fahrzeug wird aber nicht geladen. Erst wenn ich die Gesamtleistung auf über 16 AMp stelle fängt sie an mit 6 Ampere zu laden.?
Habe schon meherer PNi´s hier liegen aber keins Funktioniert so richtig. 
Hat jemand ne Lösung dazu?
 </div>

eismarkt, 2023-09-13:
was möchtest du wissen?

andig, 2023-09-13:
Eine konkrete Fehlerbeschreibung wäre toll.

eismarkt, 2023-09-13:
steht doch oben! Das Problem besteht wohl mit den neuen Hauptplatinen das dort die einzelnen Ladepunkte über MODBUS nur angesteuert werden wenn beim Gesamtstrom mehr als 16 Ampere eingestellt werden. Das tut EVCC aber nicht sondern stellt immer nur so viel Gesamtstrom ein wie auch beim Ladepunkt . Dadurch startet das Laden nicht.

andig, 2023-09-13:
Ah. Du möchtest also Gesamtstrom=16A, je Ladepunkt weniger? Mir war nicht bewusst, dass wir Gesamtstrom überhaupt anfassen?

oMtQB4, 2023-09-13:
> Ah. Du möchtest also Gesamtstrom=16A, je Ladepunkt weniger? Mir war nicht bewusst, dass wir Gesamtstrom überhaupt anfassen?

Das ist ja der Workaround #3738, der mal implementiert wurde. Den soll man in dem Pull Request, den ich gemacht habe, deaktivieren können.

eismarkt, 2023-09-13:
Ich habe vor kurzem die gesamte Elektronik in der Box neu bekommen und seid dem ist das Verhalten so. Es scheint wohl sehr wahrscheinlich an der neuen Steuerplatine Version D:16 zu liegen, ich habe heute mal das PNI v1 ph1 sowie das RFID Modul v2 eingebaut und das Problem besteht immer noch.  
aktuell ist jetzt wieder Steuerplatine Version D16 , PNI 2.0.0/h1/p1 und RFID Modul v2.0 D.3 verbaut alles das neu zugeschickte. Auf der alten Steuerplatine steht leider keine Version drauf. 
aber es scheint ja das man über modbus die Firmware Versionen auslesen kann.

premultiply, 2023-09-13:
Könntest du ggf. mal mittels `evcc charger --diagnose` die diversen Firmwareversionen der Platinen auslesen und dann sagen welche Kombi funktioniert und welche nicht?

eismarkt, 2023-09-13:
das ist Stand aktuell:
Diagnostic dump:
        Total Current Limit:    0
        Conn 1 Current Limit:   0
        RFID cards learned:     0
        Firmware Version 1:     3
        Firmware Version 2:     16
        Enclosure Temperature:  189
wallbox5
--------
Charge status: A
Enabled:       false

Diagnostic dump:
        Total Current Limit:    0
        Conn 2 Current Limit:   0
        RFID cards learned:     0
        Firmware Version 1:     3
        Firmware Version 2:     16
        Enclosure Temperature:  189