evcc-io/evcc #9871: evcc nur mit variablem Stromtarif (z.B. Tibber)

tyset, 2023-09-13:
**Is your feature request related to a problem? Please describe.**
Wenn man evcc ohne PV verwendete sondern nur mit variablem Stromtarif (z.B. Tibber), sind die Menüs nicht mehr selbsterklärend. Bspw. muss man den Modus "PV" auswählen, um mit günstigem Strom zu laden. Das ist nicht intuitiv und erklärungsbedürftig, insbesondere für neue Nutzer oder Familienmitglieder die das selten benutzen.

**Describe the solution you'd like**
Wenn nur ein variabler Stromtarif konfiguriert ist, aber keine PV, dann wird:
-     Erzeugung und entsprechende Anzeigen werden komplett abgeschaltet
-     Schaltfläche "PV" wird umbenannt in "Günstig"
-     Schaltfläche "Min+PV" entfällt
 
Dadurch würde der Betrieb nur mit Tibber aber ohne PV noch selbsterklärender.

**Describe alternatives you've considered**
Keine alternative Lösung bekannt.

andig, 2023-09-13:
Das Kriterium wäre:

- keine PV
- dynamischer Tarif- als `Grid` oder `Planner`

naltatis, 2023-09-14:
Ja, das betrifft auch noch ein paar andere Stelle/Texte in der UI. Sollten wir aber machen.

Raudi1, 2023-09-16:
Ich benutze EVCC auch ohne PV mit Tibber.
Die Anzeigen vom Sonnenanteil sind ohne PV auch hinfällig. Da gibt es auch Bugs, dass teilweise ein paar Wh Sonnenanteil angezeigt werden, was ja bei mir nicht stimmen kann.
Im selben Zug könnte man auch die Preisgrenzeneinstellung offensichtlicher machen. Momentan sieht man die eingestellte Preisgrenze nur nachdem man auf den aktuellen Preis bei Netzbezug klickt. Das ist einerseits nicht intuitiv zu finden für neue Nutzer und andererseits wäre es wünschenswert direkt in der Übersicht zu sehen was man eingestellt hat. Aktuell sehe ich den derzeitigen Preis gleich 3 mal, aber den eingestellten nur im Untermenü.

Raudi1, 2023-10-28:
Mir ist gerade noch eine weitere mögliche Verbesserung bezüglich Tibber eingefallen:
Es wäre schön, wenn es irgendeine Möglichkeit gäbe die Preise von morgen aktualisieren zu lassen. 
Ich sehe z.B. immer Samstags direkt um 13 Uhr nach wie die Preise von morgen sind, da es zumindest im Sommer am Wochenende häufig Samstags um 13  Uhr am günstigsten war. Dann kann ich direkt sehen, ob es Sinn macht direkt zu laden, oder erst einen Tag später. Leider kann ich dann aber noch nicht die Ladeplanung bei EVCC direkt einstellen, da dort die Tibber Preise erst später automatisch aktualisiert werden.
Meiner Meinung nach sollte es entweder im Einstellmenü der Preisgrenze einen Knopf zum aktualisieren geben, oder dies automatisch beim Aufrufen des Menüs geschehen.

gredie, 2023-11-06:
Ich möchte demnächst EVCC mit Awattar ohne PV benutzen. Dabei aber auch nach dem Anstecken sofort auf einen Minimum-SOC z.b. 20% laden und den Rest dann günstig. Dafür würde sich anbieten. "Min+PV" als "Min+günstig" weiter zu verwenden. Dann könnte man  im Notfall sicher sein das Auto mit einem Minimum-SOC zu haben und den Ladevorgang abzubrechen.

premultiply, 2023-11-07:
Tut es das nicht schon?

gredie, 2023-11-07:
Ja das tut es derzeit. Aber tyset hat im ersten Kommentar ja vorgeschlagen "Min+PV" wegzulassen. Ich dagegen würde den Button behalten und ohne install. PV evtl "Min+günstig" zu benennen

VolkerK62, 2023-12-10:
@gredie 
> Dabei aber auch nach dem Anstecken sofort auf einen Minimum-SOC z.b. 20% laden und den Rest dann günstig.

Genau das passiert im Modus "PV" in Verbindung mit einem (am Fahrzeug eingestellten) MinSoc.

Modus "Min+PV" hingegen, lädt sofort mit minimaler Ladeleistung, auch über einen eingestellten MinSoc hinaus.

closed 2024-06-22 by andig