evcc-io/evcc #9874: Vehicle shows as disconnected

toleabivol, 2023-09-14:
### Describe the bug

I added a vehicle, Nissan Leaf ZE1 2018, via evcc cli and checked it in the yaml file - looks good. 
When I run `./evcc --log trace` I can see the call to the vehicle API and the 200 response.
When I run `./evcc vehicle` after a few seconds I see the real car results 
But in the UI it shows as disconneceted.

[image]

Note: I did not setup the charger/wallbox yet, and just added a random IP for it, so I ignore those errors in the log.

### Steps to reproduce

1. Add vehicle
2. start evcc 
3. Check in the UI that vehicle shows as disconnected

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 30s # control cycle interval

mqtt:
  broker: 127.0.0.1:1883

meters:
- type: custom
  name: grid
  power:
    source: mqtt
    topic: deye/ac/total_power
  energy:
    source: mqtt
    topic: deye/total_energy
  currents:
    - source: mqtt
      topic: deye/ac/l1/current
    - source: mqtt
      topic: deye/ac/l2/current
    - source: mqtt
      topic: deye/ac/l3/current

- type: custom
  name: pv1
  power:
    source: mqtt
    topic: deye/dc/pv1/power
- type: custom
  name: pv2
  power:
    source: mqtt
    topic: deye/dc/pv2/power

- type: custom
  name: battery
  soc:
    source: mqtt
    topic: deye/battery/soc
  power:
    source: mqtt
    topic: deye/battery/power
  capacity: 9.6

chargers:
- type: template
  template: go-e-v3
  host: 192.168.2.2
  name: wallbox

vehicles:
- type: template
  template: carwings
  title: Nissan Leaf 2018
  icon: car
  user: ***
  password: ***
  vin: ***
  capacity: 40
  phases: 1
  cache: 15m
  mode: pv
  minSoc: 20
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 32
  name: nissan

loadpoints:
- title: Garage
  charger: wallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true
  vehicle: nissan

site:
  title: My home
  meters:
    grid: grid
    pv:
      - pv1
      - pv2
    battery:
      - battery
  residualPower: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3246 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: ***
```

### Log details

```text
./evcc --log trace
[main  ] INFO 2023/09/14 12:22:21 evcc 0.119.5 (42e6f54e)
[main  ] INFO 2023/09/14 12:22:21 using config file: /home/tolea/evcc/evcc.yaml
[main  ] INFO 2023/09/14 12:22:21 starting ui and api at :7070
[db    ] INFO 2023/09/14 12:22:21 using sqlite database: /home/tolea/.evcc/evcc.db
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT * FROM `settings` 17 <nil>
[mqtt  ] INFO 2023/09/14 12:22:21 connecting evcc-784334122 at tcp://127.0.0.1:1883
[mqtt  ] DEBUG 2023/09/14 12:22:21 tcp://127.0.0.1:1883 connected
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2023/09/14 12:22:21 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[go-e  ] TRACE 2023/09/14 12:22:21 GET http://192.168.2.2/api/status?filter=alw
[db    ] TRACE 2023/09/14 12:22:26 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[carwings] TRACE 2023/09/14 12:22:26 POST https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/InitialApp_v2.php
[carwings] TRACE 2023/09/14 12:22:27 initial_app_str=***
--
{"status":200,"message":"success","baseprm":"88dSp7wWnV3bvv9Z88zEwg"}
[carwings] TRACE 2023/09/14 12:22:27 POST https://gdcportalgw.its-mo.com/api_v200413_NE/gdc/UserLoginRequest.php
[carwings] TRACE 2023/09/14 12:22:33 Password=***&RegionCode=NE&UserId=***&initial_app_str=***
--
{"status":200,"sessionId":"***","VehicleInfoList":{"VehicleInfo":[{"nickname":"Leaf1***","telematicsEnabled":"true","vin":"***"}],"vehicleInfo":[{"nickname":"Leaf1***","telematicsEnabled":"true","vin":"***","custom_sessionid":"***"}]},"vehicle":{"profile":{"vin":"***","gdcUserId":"","gdcPassword":"","encAuthToken":"***","dcmId":"***","nickname":"Leaf1***","modelyear":"18"}},"EncAuthToken":"***","CustomerInfo":{"UserId":"***","Language":"de-DE","Timezone":"Europe\/Berlin","RegionCode":"NE","OwnerId":"***","EMailAddress":"***","Nickname":"Leaf1tole","Country":"DE","VehicleImage":"\/content\/language\/default\/images\/img\/ph_car.jpg","UserVehicleBoundDurationSec":"946771200","VehicleInfo":{"VIN":"***","DCMID":"**","SIMID":"***","NAVIID":"***","EncryptedNAVIID":"***","MSN":"***","LastVehicleLoginTime":"","UserVehicleBoundTime":"2023-09-13T08:48:05Z","LastDCMUseTime":"","NonaviFlg":"false","CarName":"LEAF","CarImage":"carimg7.png"}},"UserInfoRevisionNo":"1","ngTapUpdatebtn":"300000","timeoutUpdateAnime":"300000","resultKey":"***"}
[go-e  ] TRACE 2023/09/14 12:22:33 GET http://192.168.2.2/status
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/dc/pv1/power: '1861.0'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/dc/pv2/power: '2306.0'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/total_energy: '2312.8'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/battery/power: '-3521.0'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/battery/soc: '67.0'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/ac/total_power: '9.0'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/ac/l1/current: '0.7'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/ac/l2/current: '0.1'
[mqtt  ] TRACE 2023/09/14 12:22:36 recv deye/ac/l3/cur
```

### What type of operating system are you running?

Linux

### Version

0.119.5

StefanSchoof, 2023-09-14:
Disconnect means, that your car is not connected to the wallbox and when your charger is not configured this is expected.

StefanSchoof, 2023-09-14:
@naltatis This misunderstanding I seen several times. Any good idea how to improve the UX here?

toleabivol, 2023-09-14:
I see - ok , I was expecting this kind of response . Then it is ok to close this unless you want to use it for tracking UX improvement.

StefanSchoof, 2023-09-14:
The default poll mode is charging. So the soc is not polled when disconnected: https://docs.evcc.io/en/docs/reference/configuration/loadpoints/#pollmode

andig, 2023-09-14:
Der Verbindungsstatus im UI kommt von der Wallbox, nicht vom Auto...

closed 2023-09-14 by andig