evcc-io/evcc #9886: falsche min/max. Anzeige in den Loadpoint Einstellungen bei 3phasig

StevieC121176, 2023-09-15:
### Describe the bug

PR #9728 sollte:
> * fixed phase-based min/max power estimation based on `phasesConfigured` and `phasesActive` fixes [Experimental: max power falsch #6178](https://github.com/evcc-io/evcc/issues/6178)
> 
beheben. Das funktioniert bei mir nur wenn die Wallbox 1 Phasig eingestellt ist (min 1,4kw und max 3,7kw)
Bei 3 Phasig wird min 1,4kw und max. 11 kw angezeigt (sh. Foto) Richtig wäre aber min 4,1 kw und max 11 kw

[image]

[image]

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

`0.119.5 (11819648)`

naltatis, 2023-09-15:
Magst du mal die Antwort von /api/state zu dem Zeitpunkt posten. Spannend ist wie die "phase*" Werte an dem Ladepunkt aussehen.

StevieC121176, 2023-09-15:
Auto ist noch angeschlossen, Wallbox läuft 3phasig .` /api/state ` liefert folgendes (ich denke das ist der relevante Auszug)

`"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":7849000000000,"chargePower":0,"chargeRemainingDuration":0,"chargeRemainingEnergy":0,"chargeTotalImport":3910.4**,"chargeVoltages":[238.1,238,235.9]**,"chargedEnergy":12000,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":true,"connectedDuration":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"maxCurrent":16,"minCurrent":6,"minSoc":10,"mode":"off","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":null,"phasesEnabled":3,"planActive":false,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":0.41761378320752457,"sessionEnergy":12000,"sessionPrice":0.9876829233057874,"sessionPricePerKWh":0.08230691027548229,"sessionSolarPercentage":99.79808534507751,"targetEnergy":0,"targetSoc":100,"targetTime":"0001-01-`

StevieC121176, 2023-09-15:
Ich habe das Auto jetzt mal 1phasig angeschlossen, `/api/state/` liefert: 

`"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeRemainingDuration":0,"chargeRemainingEnergy":0,"chargeTotalImport":3910.4,"chargeVoltages":[238.7,0,0],"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":true,"connectedDuration":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"maxCurrent":16,"minCurrent":6,"minSoc":10,"mode":"off","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planActive":false,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"targetEnergy":0,"targetSoc":100,"targetTime":"0001-01-`

naltatis, 2023-09-15:
Argh. Das Thema ist immer etwas komplizierter als man denkt.

PR #9895 (2023-09-16): Fix phase dependent min/max power estimate
fixes #9886

fixed 1p
[image]

configurable: auto
[image]

configurable: 1p
[image]

configurable: 3p
[image]
closed 2023-09-16 by andig