evcc-io/evcc #9894: Battery Settings UI and evcc.yaml out of sync, WARN message in log

markusjnagel, 2023-09-15:
### Describe the bug

Site configuration in evcc.yaml and what is shown in the UI (after restart, no settings changed via the UI) seem to be out of sync.
Additionally, there is a (IMO) false warning in the log, unless I have misread/misunderstood the warning.

Site settings in evcc.yaml are:
```
site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
  prioritySoc: 45  #bis 45% hat die Hausbatterie bei Überschuss Vorrang
  bufferSoc: 50 #Batterie darf im Modus PV bis auf 50% entladen werden, wenn nicht genug Überschuss für Mindes>
  bufferStartSoc: 60 #Im Modus PV startet die Ladung auch ohne genug Überschuss, sobald die Batterie mindesten>

```

The UI shows different values (`prioritySoc` is 45, not 40, `bufferSoc` is 50, not 45, `bufferStartSoc`... _could_ be right, but looks like 10% above `bufferSoc`, so at 55, not 60). 

[image]

The logs also show the WARN message 
```
Sep 15 18:30:19 evcc evcc[11374]: [site  ] WARN 2023/09/15 18:30:19 bufferSoc must be larger than prioritySoc
```
which it is, if you look at the site settings.

Not a big issue, but a bit misleading. I will check tomorrow which values actually take effect.
I have set the site to debug, but it doesn't show which values are used, only that (correctly) currently the battery has priority. 
```
Sep 15 18:27:32 evcc evcc[11306]: [site  ] DEBUG 2023/09/15 18:27:32 giving priority to battery charging at soc: 39%

```

### Steps to reproduce

1) configure evcc.yaml with the following site settings:

```
  residualPower: 100
  prioritySoc: 45  #bis 45% hat die Hausbatterie bei Überschuss Vorrang
  bufferSoc: 50 #Batterie darf im Modus PV bis auf 50% entladen werden, wenn nicht genug Überschuss für Mindestladeleistung
  bufferStartSoc: 60 #Im Modus PV startet die Ladung auch ohne genug Überschuss, sobald die Batterie mindestens 60% hat (also bei 61%)

```

2) Restart evcc
```
sudo systemctl restart evcc
```

3) Check the battery settings in the UI:

[image]

### Configuration details

```yaml
# open evcc at http://evcc.local
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: info
[... 316 lines cut ...]
Sep 15 18:35:04 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:04 battery power: 2324W
Sep 15 18:35:04 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:04 grid power: 24W
Sep 15 18:35:04 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:04 site power: 2448W
Sep 15 18:35:04 evcc evcc[11374]: [lp-1  ] DEBUG 2023/09/15 18:35:04 charge currents: [0 0 0]A
Sep 15 18:35:04 evcc evcc[11374]: [lp-1  ] DEBUG 2023/09/15 18:35:04 charger status: A
Sep 15 18:35:19 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:19 ----
Sep 15 18:35:19 evcc evcc[11374]: [lp-1  ] DEBUG 2023/09/15 18:35:19 charge power: 0W
Sep 15 18:35:19 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:19 pv power: 177W
Sep 15 18:35:19 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:19 battery soc: 37%
Sep 15 18:35:19 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:19 battery power: -115W
Sep 15 18:35:20 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:20 grid power: 2362W
Sep 15 18:35:20 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:20 giving priority to battery charging at soc: 37%
Sep 15 18:35:20 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:20 site power: 2462W
Sep 15 18:35:20 evcc evcc[11374]: [lp-1  ] DEBUG 2023/09/15 18:35:20 charge currents: [0 0 0]A
Sep 15 18:35:20 evcc evcc[11374]: [lp-1  ] DEBUG 2023/09/15 18:35:20 charger status: A
Sep 15 18:35:34 evcc evcc[11374]: [site  ] DEBUG 2023/09/15 18:35:34 ----
```

### What type of operating system are you running?

Linux

### Version

0.119.5

naltatis, 2023-09-15:
Hi @markusjnagel,

Die Werte `prioritySoc`, `bufferSoc` und `bufferStartSoc` können inzwischen über die UI editiert werden. Das ist auch der präferierte Weg. Sobald du die Werte einmal in der UI geändert hast, werden diese Werte genutzt (Datenbank).

Aus Kompatiblitätsgründen haben wir die Optionen noch nicht aus der yaml entfernt. Ich verstehe aber, dass das gerade noch ein irreführender Zustand ist. Das Verhalten sollten wir in der Doku klarer beschreiben.

//cc @VolkerK62

naltatis, 2023-09-15:
Duplicate #9245

closed 2023-09-15 by naltatis

markusjnagel, 2023-09-15:
Oh wow.... DANKE EUCH :+1: :bow: für die schnelle Rückmeldung, das erklärt es natürlich...

markusjnagel, 2023-09-15:
oh,... @naltatis ... bevor ich jetzt ein neues Issue aufmache, bei dem das Problem aber vor dem Rechner sitzt... :wink: Jetzt wollte ich die Werte auf die von mir eigentlich in der YAML gesetzten ändern... aber höhere als die gesetzten sind ausgegraut. 
Nimmt er evtl. für die Entscheidung was "auszugrauen" ist, die Werte aus der YAML? 

[image]

markusjnagel, 2023-09-15:
... da scheint sich noch was zu beißen, wenn man die Werte in der YAML hat.
Habe alle auskommentiert und jetzt konnte ich es über das UI so setzen, wie ich es haben wollte. 
Nochmals vielen Dank für den Hinweis (persistiert in der DB). Und überhaupt vielen vielen Dank für evcc :heart_decoration: , Deinen/Euren Einsatz :muscle: und natürlich die Sticker :laughing: 

[image]

naltatis, 2023-09-15:
Das Ausgrauen liegt daran, dass der untere Wert nie größer sein darf als der obere. Wenn du den oberen vorher bewegst, hat der untere auch mehr Luft. Das ist von der UX noch nicht optimal. Aber für jetzt sollte das erstmal funktionieren :D

markusjnagel, 2023-09-15:
> Das Ausgrauen liegt daran, dass der untere Wert nie größer sein darf als der obere. Wenn du den oberen vorher bewegst, hat der untere auch mehr Luft. Das ist von der UX noch nicht optimal. Aber für jetzt sollte das erstmal funktionieren :D

Ich wollte Dir grad schreiben, dass ich das versucht hatte (zuerst den oberen), aber bei genauer Betrachtung des Screenshots bin ich nicht mehr so sicher..... wahrscheinlich ist es so, wie Du sagst. Aber egal wie - über das UI (bei mir jetzt mit auskommentierten Werten in der yaml) funktioniert es so wie es soll :smiley: 
Danke dafür. :+1:  :+1:  :+1: