evcc-io/evcc #9906: Falsche Ladewerte in Verbindung mit Wallbox Pulsar Plus

SKB-CGN, 2023-09-16:
### Describe the bug

Hi,
ich habe heute beobachtet, das sich die live Werte in EVCC von der in der Wallbox unterscheiden.
[image]
[image]

### Steps to reproduce

1. Ldevorgang via EVCC starten
2. Prüfung der Werte über die Wallbox und via EVCC Anzeige

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 7e8c13be6242ad0c1ea1115d90caef9e16045f66d4fdb8fae26c22b050e4deda

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.2.5  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.2.5  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.2.5  
  port: 8080  
  name: battery3

chargers:
- type: template
  template: pulsarplus 
  name: wallbox5

vehicles:
- type: template
  template: hyundai 
  title: Ioniq  
  user: stephan@kreyenborg.koeln  
  password: '!Kirchweihe0?'  
  capacity: 9  
  language: de  
  name: ev4
  phases: 1
  maxcurrent: 16

loadpoints:
- title: Einfahrt
  charger: wallbox5
[... 231 lines cut ...]
[lp-1  ] DEBUG 2023/09/16 15:13:12 guard timer inactive
[lp-1  ] DEBUG 2023/09/16 15:13:12 charger disable
[lp-1  ] DEBUG 2023/09/16 15:13:12 wake-up timer: stop
[site  ] DEBUG 2023/09/16 15:13:17 ----
[lp-1  ] DEBUG 2023/09/16 15:13:17 charge power: 0W
[site  ] DEBUG 2023/09/16 15:13:17 pv power: 1033W
[site  ] DEBUG 2023/09/16 15:13:17 battery soc: 37%
[site  ] DEBUG 2023/09/16 15:13:17 battery power: -636W
[site  ] DEBUG 2023/09/16 15:13:17 grid power: -3383W
[site  ] DEBUG 2023/09/16 15:13:17 site power: -4019W
[lp-1  ] DEBUG 2023/09/16 15:13:17 charger status: B
[lp-1  ] INFO 2023/09/16 15:13:17 stop charging <-
[lp-1  ] DEBUG 2023/09/16 15:13:18 soc estimated: 86.56% (vehicle: 86.00%)
[lp-1  ] DEBUG 2023/09/16 15:13:18 vehicle soc: 87%
[lp-1  ] DEBUG 2023/09/16 15:13:18 vehicle soc limit: 0%
[lp-1  ] DEBUG 2023/09/16 15:13:18 vehicle range: 0km
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.0

andig, 2023-09-16:
Auf welchen Timestamp beziehen sich die Screenshots? Was ist richtig und was ist falsch?

SKB-CGN, 2023-09-16:
Die Tamestamps beziehen sich auf den Punkt, wo der Ladevorgang dann im Gange war.

Die Zeit im Screenshot ist etwas verschoben.

andig, 2023-09-16:
Um welchen Timestampnim Log geht es?

VolkerK62, 2023-09-16:
```
[lp-1  ] INFO 2023/09/16 15:10:57   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/16 15:10:57   meters:      charge ✗
```
Die Wallbox scheint zwar eine Messeinrichtung zu haben, aber nur für `energy` nicht für `power`.
Deshalb wird die Leistung in evcc anhand der freigegebenen Stromstärke (16A) angezeigt (16A x 230V = 3680W)
`[lp-1  ] DEBUG 2023/09/16 15:13:12 charge power: 3680W`

SKB-CGN, 2023-09-16:
@VolkerK62 wieso zeigt dann die App die richtige Leistung an?

andig, 2023-09-16:
Was heisst scheint nicht genau? Bitte mal ocpp auf trace stellen, dann sehen wir auch was los ist.

andig, 2023-09-17:
Lass uns gerne wieder auf machen wenn wir trace log und Screenshots vom gleichen Zeitpunkt haben.

closed 2023-09-17 by andig

SKB-CGN, 2023-09-17:
In meinen Augen ist das sehr unprofessionell, Issues zu schließen, solange Informationen fehlen bzw. das Problem besteht.
Denn, es ist solange ein Issue, bis es behoben wurde - sei es vom User oder Developer. Es aber einfach zu schließen, zeugt eher von der eigenen Selbstsicherheit! Enttäuschend!

andig, 2023-09-17:
> Lass uns gerne wieder auf machen wenn wir trace log und Screenshots vom gleichen Zeitpunkt haben.

Da es an dieser Stelle nicht für uns zu gibt als zu warten so lange geschlossen. Ein Logfile würde helfen. Meckern verschwendet nur unsere Zeit und wird das Problem des fehlenden Logfiles nicht lösen.

andig, 2023-09-17:
Wenn Du soviel Mühe wie für diese Kommentare in das Logfile stecken würdest hätten wir ggf. schon eine Lösung. Lass uns bitte aufhören, hier weiter Zeit zu verschwenden.

SKB-CGN, 2023-09-17:
Ich habe selten einen so überheblichen 'Internet-Entwickler', wie Dich gesehen.

Lass das Issue geschlossen - ich nutze andere Daten, um die Ladeleistung **richtig** anzuzeigen.

Tatsächlich hatte ich mir ein Sponsoring überlegt - dies investiere ich nun Anders.

Unglaublich peinlich und überheblich!