evcc-io/evcc #9907: ESPHome DSMR fatal error

olafz, 2023-09-16:
### Describe the bug

When using a ESPHome grid meter of type slimmelezer (https://docs.evcc.io/en/docs/devices/meters#esphome-dsmr), EVCC crashes during startup.

### Steps to reproduce

Set the configuration as shown below and restart EVCC. It will throw the fatal error as shown below.

### Configuration details

```yaml
The relevant part of the configuration is below:

meters:
  - name: grid
    type: template
    template: slimmelezer
    usage: grid
    host: a.b.c.d
```

### Log details

```text
(...)

[main  ] FATAL 2023/09/16 15:08:51 cannot create meter 'grid': cannot create meter 'custom': 1 error(s) decoding:

* 'Power' expected a map, got 'slice'

(...)
```
```

### What type of operating system are you running?

Docker container

### Version

0.120.0

olafz, 2023-09-16:
A PR to fix this issue is here: https://github.com/evcc-io/evcc/pull/9908

andig, 2023-09-16:
Configfehler.

closed 2023-09-16 by andig