evcc-io/evcc #9910: CO2 limit in smart grid loading is shown as price limit in planer

docolli, 2023-09-16:
### Describe the bug

When for example Grünstromindex is used to get CO2 values and a CO2 limit is set in the smart grid loading window, this value (multiplied by 100) is then shown in the planner as price limit in the planer.

[image]
[image]

[image]
[image]

### Steps to reproduce

1. Go to "Smartes Netzladen"
2. Set CO2 limit to a numerical value
3. Close window
4. Open "PLAN"
5. See Text "Die eingestellte Preisgrenze von xxx,x ct wird in diesem Zeitraum ignoriert." with the selected CO2 value multiplied by 100.
...

### Configuration details

```yaml
most likely not needed
```

### Log details

```text
most likely not needed
```

### What type of operating system are you running?

Linux

### Version

0.120.0

naltatis, 2023-09-16:
Magst du einmal den tariff Block aus deiner config posten. Mir ist noch nicht klar wie das passiert.

docolli, 2023-09-16:
Klar, hier der Block:

```
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.5286 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.066891 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: 88447 # PLZ mit führender Null mit in "" setzen
```

naltatis, 2023-09-17:
War ein ganz dummer Fehler. Fehlendes return statement.

PR #9918 (2023-09-17): Charging plan: fix co2 notice
fixes #9910
closed 2023-09-17 by andig