evcc-io/evcc #9913: evcc release: 1.120.0 - phoenix-em-eth

zwiebelxxl, 2023-09-16:
### Describe the bug

Hallo,

nach update von 1.119.5 auf 1.120.0 kann ich evcc nicht mehr starten.

was wurde geändert?

Viele Grüße!

### Steps to reproduce

restart evcc nach update

### Configuration details

```yaml
chargers:
  - name: phoenix # wallbe
    type: template
    template: phoenix-em-eth
    host: 192.168.10.35

hab auch schon erweitert mit:

chargers:
  - name: phoenix # wallbe
    type: template
    template: phoenix-em-eth
    host: 192.168.10.35
    modbus: tcpip
    id: 255
    port: 502
```

### Log details

```text
Sep 16 19:55:14 as2 systemd[1]: Started evcc.
Sep 16 19:55:14 as2 evcc[519046]: [main  ] INFO 2023/09/16 19:55:14 evcc 0.120.0
Sep 16 19:55:14 as2 evcc[519046]: [main  ] INFO 2023/09/16 19:55:14 using config file: /opt/evcc/evcc.yaml
Sep 16 19:55:17 as2 evcc[519046]: [main  ] FATAL 2023/09/16 19:55:17 cannot create charger 'phoenix': cannot create charger 'template': cannot create charger 'phoenix-em-eth': 1 error(s) decoding:
Sep 16 19:55:17 as2 evcc[519046]: * '' has invalid keys: rtu
```

### What type of operating system are you running?

Linux

andig, 2023-09-16:
@premultiply falsches Template?

premultiply, 2023-09-16:
Ich verstehe nicht woher diese Fehlermeldung kommen soll.

zwiebelxxl, 2023-09-16:
Wenn ich die Wallbox neu konfiguriere dann kommt das gleiche

--------------------------
./evcc configure
...
 Wähle eine der folgenden Gerätekategorien aus: Wallbox

- Konfiguration Wallbox

 Wähle eine Wallbox: Phoenix Contact EM-CP-PP-ETH (Modbus TCP)

Führe folgende Einstellungen durch:

 Modbus ID 255
 IP-Adresse oder Hostname 192.168.10.35
 Port 502

Teste die Konfiguration von Phoenix Contact EM-CP-PP-ETH (Modbus TCP) ...
   Fehler: cannot create charger 'phoenix-em-eth': 1 error(s) decoding:

* '' has invalid keys: rtu

andig, 2023-09-16:
@premultiply template mit der Modbus Änderung verwürfelt? Rtu statt rs485?!

Hofyyy, 2023-09-16:
Ich weiss zwar nicht ob es beim Einkreisen hilft. Aber mein Update auf die Version lief mit der Wallbe Pro (modbus tcp) ohne probleme. ich musste auch nicht in die config.

premultiply, 2023-09-16:
Eigentlich nicht: https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/phoenix-em-eth.yaml

andig, 2023-09-17:
Fehler gefunden:

```
Auf geht's:

 In welchem Modus soll die Konfiguration durchgeführt werden? Standard Modus (So einfach und schnell wie möglich)

- Ein Gerät konfigurieren

Wähle eine der folgenden Gerätekategorien aus:

- Konfiguration Wallbox

 Wähle eine Wallbox: Phoenix Contact EM-CP-PP-ETH (Modbus TCP)

Führe folgende Einstellungen durch:

 Modbus ID 255
 IP-Adresse oder Hostname foo
 Port 502

Teste die Konfiguration von Phoenix Contact EM-CP-PP-ETH (Modbus TCP) ...
   Fehler: cannot create charger 'phoenix-em-eth': 1 error(s) decoding:

* '' has invalid keys: rtu
```

@premultiply das alte Template hatte noch `rtu` für RTU over TCP als Paramter.

@zwiebelxxl bitte den Parameter aus Deiner Config einfach raus löschen- der sollte eh auf "false" stehen?

Ich wäre dafür an dieser Stelle keinen Bugfix zu machen- die Lösung ist trivial und der Parameter ohnehin unnütz.

premultiply, 2023-09-17:
Spannend. Der Controller unterstützt überhaupt kein Modbus RTU.

zwiebelxxl, 2023-09-17:
ich habe kein parameter rtu

so schaut es jetzt (unten) aus und evcc startet nicht.
```
chargers:
  - name: phoenix # wallbe
    type: template
    template: phoenix-em-eth
    host: 192.168.10.35
```

andig, 2023-09-17:
@premultiply dann braucht es doch einen Fix. Ursache: Du hast vom einfachen `host` Parameter auf `modbus` umgestellt. In go gibts aber kein `modbus.TCPSettings` sondern einfach nur `host`. Das clashed. Entweder die Änderung zurück drehen (das muss dann auch überall im modbus Branch passieren) oder in go konsequent auf `modbus.TCPSettings` umbauen.

andig, 2023-09-17:
Das Problem gibts in allen Templates: https://github.com/evcc-io/evcc/pull/9729/files#diff-22dbd6727892ea6f7ff4403ed9f2e2a3630f498e33dce114a324d5f5aa54c850L10-L17

maikheinrich, 2023-09-17:
Hallo zwiebelxxl

ich habe ebenfalls den Phoenix Controller im Einsatz und gestern auf die 120 upgedatet. Hier läuft alles wunderbar. Hier ein Auszug der Config...

# charger definitions
chargers:
- name: evcc_1
  type: phoenix-em-eth
  uri: 192.168.20.151:502
  meter:
    power: true
    energy: true
    currents: true

Liebe Grüße

Maik

andig, 2023-09-17:
Ja, das passt. Der Unterschied ist, dass Du `type` direkt nutzt (keine Übersetzung), das bei @zwiebelxxl aber noch durch ein Template läuft das da ein `rtu: false` rein bastelt.

zwiebelxxl, 2023-09-17:
Danke @maikheinrich und @andig so gehts jetzt!
```
chargers:
  - name: phoenix # wallbe
    type: phoenix-em-eth
    uri: 192.168.10.35
```

premultiply, 2023-09-17:
Aaarg. Das ist aber echt hinterhältig. 😕

Bitte dann temporär die Direktdefinition verwenden.
Das muss ich mir genauer ansehen. Komme da aber erst heute Abend dazu.

closed 2023-09-17 by andig

Sami0178, 2023-10-09:
Hallo, 
wie sieht nun die korrekte Config für den phoenix-em-eth aus? (Aufruf wie in Doku funktioniert bei mir leider nicht)
Wenn ich aber nur den type ohne template nutze, gehts, aber es werden keine Zählerwerte (start/ende) mehr gelesen. ?
chargers:
  - name: phoenix
    type: phoenix-em-eth
    uri: 192.168.2.12:502

andig, 2023-10-09:
Bitte neues Issue für „werden nicht gelesen“. Evtl geht Modbus hier nur registerweise?