evcc-io/evcc #9917: MaxCurrent for Vehicle

kscholty, 2023-09-17:
I'm ve got three different cars charging at a 32A charger with phase switching. 

If I set the charger's Maxcurrent to 32A in the config evcc will push the current up to that value before switching to 3 phases.
Unfortunately, not all of the cars support that much current on one phase, while others do.
Furthermore, 32A@1p stresses the components way more than 11A@3p.

I propose to add a `MaxCurrent` value to vehicles as well that is then taken into account for phase switching, similar to the `phases` parameter we already got.

Alternatively we could monitor the consumed power and switch as soon as the car does not consume the provided current. This requires an attached meter, however.
Of course, we can combine the approaches.

I've set my charger to 23 A so that evcc will switch earlier, but that's no solution for cars that can charge only 1 Phase, but with 32A (as. e.g. the MG4 Standard).

If you like, I can propose an implementation.

naltatis, 2023-09-17:
Setting maxcurrent per vehicle is already possible right now. But when I unterstand correctly you want different min/maxcurrent based on the number of phases right?

https://docs.evcc.io/en/docs/reference/configuration/vehicles/#onidentify

kscholty, 2023-09-17:
My bad. I simply didn't understand how this "on Identify" works. Thanks for the hint.

closed 2023-09-17 by kscholty

andig, 2023-09-17:
> But when I unterstand correctly you want different min/maxcurrent based on the number of phases, right?

That's currently not possible- technical restrictions, sorry...

premultiply, 2023-09-17:
> Alternatively we could monitor the consumed power and switch as soon as the car does not consume the provided current.

It may already work like this.