evcc-io/evcc #9924: SMA EV Charger 22: AC-Strombegrenzung kleiner als Minimaler Ladestrom

kafetzke, 2023-09-17:
### Describe the bug

Moin zusammen,

ich nutze einen SMA EV Charger 22, um damit u.a. einen Renault ZOE Ph2 zu laden. Wie sicherlich bekannt, ist der ZOE aufgrund seines eigenwilligen 22kW-Laders ziemlich "mäkelig" was das Thema minimaler Ladestrom angeht.

Eine brauchbare Lösung wäre, den minimalen Ladestrom auf 8A anzuheben. Das klappt dann auch, solange ausreichend PV-Strom vorhanden ist.

ABER:
Leider setzt EVCC den Parameter "AC-Strombegrenzung" regelmäßig auf unter 8A (und damit unterhalb den minimalen Ladestrom. In Folge dessen läd das Fahrzeug nicht.

API-Values aus Sicht des SMA EV Charger 22:
- Minimaler Ladestrom: Parameter.Chrg.AMinCha
- AC-Strombegrenzung: Parameter.Inverter.AcALim

Bitte begrenzt daher die AC-Strombegrenzung nach unten anhand des eingestellten Minimalen Ladestroms (AC-Strombegrenzung >= Minimaler Ladestrom).

Viele Grüße

### Steps to reproduce

Zum Nachstellen wird weniger als 2kW PV-Überschuss benötigt. Darüber hinaus klappt es unregelmäßig dann doch (da die AC-Strombegrenzung >=8 Ampere gesetzt wird

- wähle Fahrzeug "Renault ZOE" aus
- Wähle Modus Min-PV aus

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: debug
levels:
  cache: error

# unique installation id
plant: 07f50f8afca18eaffe9094548d9f6305044510b07883f07b330282e8fe8437ba

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.0.43
  interface: ens192
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.0.191
  password: xxxxxxxxxxx
  name: pv2
- type: template
  template: tasmota
  usage: pv
  host: 192.168.0.222 # IP-Adresse oder Hostname
  user: # Standard-User ist admin (Optional)
  password: # Passwort des Benutzerkontos
  name: pv3
[... 491 lines cut ...]
[lp-1  ] DEBUG 2023/09/17 12:47:26 set min current: 6
[site  ] DEBUG 2023/09/17 12:47:27 ----
[lp-1  ] DEBUG 2023/09/17 12:47:27 charge power: 0W
[site  ] DEBUG 2023/09/17 12:47:27 pv power: 3504W
[site  ] DEBUG 2023/09/17 12:47:27 grid power: -2807W
[site  ] DEBUG 2023/09/17 12:47:27 grid powers: [-741 -832 -1234]W
[site  ] DEBUG 2023/09/17 12:47:27 grid currents: [-3.33 -3.66 -5.32]A
[site  ] DEBUG 2023/09/17 12:47:27 site power: -2807W
[lp-1  ] DEBUG 2023/09/17 12:47:27 charge currents: [-0 -0 -0]A
[lp-1  ] DEBUG 2023/09/17 12:47:27 charger status: B
[lp-1  ] DEBUG 2023/09/17 12:47:27 pv charge current: 4.07A = 0A + 4.07A (-2807W @ 3p)
[lp-1  ] DEBUG 2023/09/17 12:47:28 set min current: 6
[lp-1  ] DEBUG 2023/09/17 12:47:33 set min current: 6
[lp-1  ] DEBUG 2023/09/17 12:47:34 set min current: 6
[lp-1  ] DEBUG 2023/09/17 12:47:36 set min current: 6
[site  ] DEBUG 2023/09/17 12:47:37 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.0

andig, 2023-09-17:
Bei welchem Timestamp siehst du den Fehler?

kafetzke, 2023-09-17:
Den Fehler sehe ich direkt auf der Wallbox, dort sind die Werte für Minimaler Ladestrom und AC-Strombegrenzung im Klartext zu sehen. Zudem zeigt mir dies auch das Log der Wallbox sehr deutlich an. Leider komme ich da nur als Screenshot dran, aber immerhin - ich habe dir die drei letzten "Vorkommnisse" markiert. In allen Fällen war der minimale Ladestrom auf 8 Ampere gesetzt und die AC-Strombegrenzung wurde durch den evcc-User auf der Wallbox auf Werte unter 8 Ampere gestellt.

[image]

Ich habe das Debugging gemäß der Anleitung auch erst gestartet, nachdem ich den EVCC-Dienst gestoppt habe. Daher wäre ich jetzt sowieso überrascht, wenn man dies im Log sehen kann.

Gibt es ein "aufzeichnendes" Logfile, dass ich bereitstellen kann, dass also während des Betriebes geschrieben wird?

andig, 2023-09-17:
Das 

    set min current: 6

kommt aus dem API. Irgendwas- oder irgendwer- sorgt also dafür, dass es bewusst aufgerufen wird. Der Ladepunkt selbst würde das sonst nicht machen. 

Wir brauchen ein Logfile das zeigt, dass und warum der Ladepunkt dem Charger ein Limit <8A gibt. In dem Logfile muss sichergestellt sein, dass niemand über das UI ein niedrigeres Limit vorgibt.

Ich mache hier zu bis wir ein passendes Logfile haben.

closed 2023-09-17 by andig

andig, 2023-09-17:
> Gibt es ein "aufzeichnendes" Logfile, dass ich bereitstellen kann, dass also während des Betriebes geschrieben wird?

Ja, siehe Doku

kafetzke, 2023-09-17:
OK, damit konnte ich das Log sinnvoller erzeugen. Ich habe das Verhalten direkt nochmal reproduziert, diesmal mit aktivem Log, in dem das Ganze auch zu sehen ist:

Wenn ich das Logfile richtig interpretiere, sind die markierten Zeilen relevant. Was mich allerdings etwas irritiert: im Logfile wird von "set min current: 6" gesprochen. Dies widerspricht sowohl dem evcc.yaml als auch der tatsächlichen Konfiguration auf der Wallbox. Dort werden die 8A als minimalen Ladestrom korrekt gesetzt.

root@evcc:~# evcc --log debug
[main  ] INFO 2023/09/17 13:50:13 evcc 0.120.0
[main  ] INFO 2023/09/17 13:50:13 using config file: /root/evcc.yaml
[main  ] INFO 2023/09/17 13:50:13 starting ui and api at :80
[db    ] INFO 2023/09/17 13:50:14 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/09/17 13:50:17 phases not configured, assuming 3p
[semp  ] WARN 2023/09/17 13:50:17 SEMP_BASE_URL unspecified, using http://192.168.0.244:80 instead
[site  ] INFO 2023/09/17 13:50:17 site config:
[site  ] INFO 2023/09/17 13:50:17   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/09/17 13:50:17     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/09/17 13:50:17     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/09/17 13:50:17     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/17 13:50:17     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/17 13:50:17   vehicles:
[site  ] INFO 2023/09/17 13:50:17     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/09/17 13:50:17 loadpoint 1:
[lp-1  ] INFO 2023/09/17 13:50:17   mode:        pv
[lp-1  ] INFO 2023/09/17 13:50:17   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/17 13:50:17   meters:      charge ✓
[lp-1  ] INFO 2023/09/17 13:50:17     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/09/17 13:50:17 phase timer inactive
[lp-1  ] DEBUG 2023/09/17 13:50:17 pv timer inactive
[lp-1  ] DEBUG 2023/09/17 13:50:17 guard timer inactive
[site  ] WARN 2023/09/17 13:50:18 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/17 13:50:18 ----
[lp-1  ] DEBUG 2023/09/17 13:50:18 charge power: 0W
[site  ] DEBUG 2023/09/17 13:50:18 pv power: 1703W
[site  ] DEBUG 2023/09/17 13:50:18 grid power: -949W
[site  ] DEBUG 2023/09/17 13:50:18 grid powers: [-42 -396 -512]W
[site  ] DEBUG 2023/09/17 13:50:18 grid currents: [-0.889 -1.83 -2.25]A
[site  ] DEBUG 2023/09/17 13:50:18 site power: -949W
[lp-1  ] DEBUG 2023/09/17 13:50:18 charge currents: [-0 -0 -0]A
[lp-1  ] DEBUG 2023/09/17 13:50:18 charger status: B
[lp-1  ] INFO 2023/09/17 13:50:18 car connected
[lp-1  ] DEBUG 2023/09/17 13:50:18 vehicle api refresh
[lp-1  ] DEBUG 2023/09/17 13:50:18 pv timer elapse
[lp-1  ] DEBUG 2023/09/17 13:50:18 pv timer inactive
[lp-1  ] DEBUG 2023/09/17 13:50:18 charger: guard elapse
[lp-1  ] DEBUG 2023/09/17 13:50:18 guard timer inactive
[main  ] DEBUG 2023/09/17 13:50:18 vehicle status: B (Renault ZOE)
[lp-1  ] INFO 2023/09/17 13:50:18 vehicle updated: unknown -> Renault ZOE
[lp-1  ] DEBUG 2023/09/17 13:50:18 set min current: 8
[lp-1  ] DEBUG 2023/09/17 13:50:18 set target soc: 80
[lp-1  ] DEBUG 2023/09/17 13:50:18 vehicle status: B
[lp-1  ] DEBUG 2023/09/17 13:50:18 vehicle soc: 82%
[lp-1  ] DEBUG 2023/09/17 13:50:18 vehicle range: 246km
[lp-1  ] DEBUG 2023/09/17 13:50:19 targetSoc reached: 82.0% > 80%
[site  ] DEBUG 2023/09/17 13:50:19 ----
[lp-1  ] DEBUG 2023/09/17 13:50:19 charge power: 0W
[site  ] DEBUG 2023/09/17 13:50:19 pv power: 1703W
[site  ] DEBUG 2023/09/17 13:50:19 grid power: -949W
[site  ] DEBUG 2023/09/17 13:50:19 grid powers: [-42 -396 -512]W
[site  ] DEBUG 2023/09/17 13:50:19 grid currents: [-0.889 -1.83 -2.25]A
[site  ] DEBUG 2023/09/17 13:50:19 site power: -949W
[lp-1  ] DEBUG 2023/09/17 13:50:19 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:19 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:19 vehicle odometer: 59825km
[lp-1  ] DEBUG 2023/09/17 13:50:19 charge currents: [-0 -0 -0]A
[lp-1  ] DEBUG 2023/09/17 13:50:19 charger status: B
[lp-1  ] DEBUG 2023/09/17 13:50:19 targetSoc reached: 82.0% > 80%
[lp-1  ] DEBUG 2023/09/17 13:50:21 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:21 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:23 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:24 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:25 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:26 set min current: 6
[... 32 lines cut ...]
[site  ] DEBUG 2023/09/17 13:50:34 grid powers: [-34 -402 -521]W
[site  ] DEBUG 2023/09/17 13:50:34 grid currents: [-0.866 -1.88 -2.29]A
[site  ] DEBUG 2023/09/17 13:50:34 site power: -957W
[lp-1  ] DEBUG 2023/09/17 13:50:34 charge currents: [-0 -0 -0]A
[lp-1  ] DEBUG 2023/09/17 13:50:34 charger status: B
[lp-1  ] DEBUG 2023/09/17 13:50:34 pv charge current: 1.39A = 0A + 1.39A (-957W @ 3p)
**[lp-1  ] DEBUG 2023/09/17 13:50:34 max charge current: 6A**
[lp-1  ] DEBUG 2023/09/17 13:50:34 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:34 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:34 charger enable
[lp-1  ] DEBUG 2023/09/17 13:50:34 wake-up timer: start
[lp-1  ] DEBUG 2023/09/17 13:50:36 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:36 set min current: 6
[site  ] DEBUG 2023/09/17 13:50:38 ----
[lp-1  ] DEBUG 2023/09/17 13:50:38 charge power: 0W
[site  ] DEBUG 2023/09/17 13:50:38 pv power: 1727W
[... cut ...]
[site  ] DEBUG 2023/09/17 13:50:38 grid powers: [-47 -406 -541]W
[site  ] DEBUG 2023/09/17 13:50:38 grid currents: [-0.915 -1.89 -2.37]A
[site  ] DEBUG 2023/09/17 13:50:38 site power: -995W
[lp-1  ] DEBUG 2023/09/17 13:50:38 charge currents: [-0 -0 -0]A
[lp-1  ] DEBUG 2023/09/17 13:50:38 charger status: B
[lp-1  ] DEBUG 2023/09/17 13:50:38 pv charge current: 1.44A = 0A + 1.44A (-995W @ 3p)
[lp-1  ] DEBUG 2023/09/17 13:50:38 set min current: 6
[lp-1  ] DEBUG 2023/09/17 13:50:38 set min current: 6

andig, 2023-09-17:
> Was mich allerdings etwas irritiert: im Logfile wird von "set min current: 6" gesprochen. Dies widerspricht sowohl dem evcc.yaml als auch der tatsächlichen Konfiguration auf der Wallbox. Dort werden die 8A als minimalen Ladestrom korrekt gesetzt.

Dein Ladepunkt erlaubt 6A. Der Charger hat die Option in evcc gar nicht.